Koenkk/zigbee2mqtt #3621: WebSocket support

vanviegen, 2020-05-27:
While writing my own web client for z2m, the thought struck me that MQTT is actually pretty inconvenient for that. So why not just cut out the middleman? :-) It would be great if instead of `z2m -> mosquitto -> my custom server --> my custom client`, it would be possible to do a straight `z2m --> my custom client`.

While doing some initial work on a [patch](https://github.com/vanviegen/zigbee2mqtt/commit/fc4905ccb5fa28abcedf95b28d25e2049dcd3e5a) (which I foolishly failed to base on `dev`, I'm realizing just now), it seemed that there would be a lot of common ground with the new API (#3281). I'm thinking there could be one set of request/response commands, which we would be able to expose over MQTT as well as WebSocket (as well as HTTP, as well as ...?).

The mentioned patch is just meant as a talking point. As of now, it fails a bunch of tests and is not all that complete. But before I dive in further, I'm looking to see if there's any interest in pursuing this direction... ?

So here's the architecture as used by the patch:

There's a bunch of `Commands` classes, that contain the commands such as `rename`, `setState`, `group/add`, `group/addDevice`, etc. The commands are just methods in these classes, prefixed by a dollar sign, to indicate that they are meant for public consumption.

The new requestResponse extension relays MQTT messages coming in on `z2m/bridge/request/#` to the Command classes, and sends the return value back over `z2m/bridge/response/#`. (This class is fully 100% guaranteed *untested* -- it's just there to get the idea across.) 

The new webSocket extension (which needs to be enabled in the configuration.yaml) sets up a WebSocket server, and does something similar as the requestResponse extension to have the Commands classes handle incoming commands. Besides that, it pushes out a (what will hopefully be a) coherent and always up-to-date overview of the devices, their supported keys and their current states.

The implementations of the Commands methods themselves are borrowed and adapted from their legacy API counterparts. The legacy implementations now use the Commands methods, to prevent code duplication.

Again: all of this is currently just an early work-in-progress, I'm interested to hear any thoughts about the desirability of an architecture like this, and the idea to support protocols other than MQTT (and WebSocket in particular).

Koenkk, 2020-05-28:
I've also been thinking about this but couldn't find a good use case for this yet. In your case the web client can also directly connect to MQTT (afaik https://www.npmjs.com/package/mqtt will also run in the browser).

sjorge, 2020-05-28:
mosquitto a popular mqtt broker also supporst websockets: http://www.steves-internet-guide.com/mqtt-websockets/

vanviegen, 2020-05-28:
Ah, I didn't know MQTT-over-WebSocket was a thing! That *does* make things a bit simpler...

Still, the state exposed over MQTT seems very much targeted towards Home Assistant, making it pretty cumbersome to work with. The WebSocket implementation could provide a nice and clean API. (And yes, of course it would also be possible to optionally expose this same API over MQTT as well.)

The amount of code involved seems relatively small, given a suitable architecture.

Koenkk, 2020-05-28:
> Still, the state exposed over MQTT seems very much targeted towards Home Assistant

In what sense? We provide 2 possible outputs, JSON and attribute based (see `experimental`: https://www.zigbee2mqtt.io/information/configuration.html#configuration). I feel both are very generic.

Also how will the WebSocket api solve that? (I don't see that this should change the structure of the messages send)

vanviegen, 2020-05-31:
Ah yes, I probably mischaracterized the z2m API, it is pretty generic. Sorry. The thing is that I also had to work with the homeassistant topics (which makes things a bit baroque), as the z2m topics don't appear to expose any information about what type a device is and what attributes it can handle. Or is there a way? If not, it may be a quick win to publish a list of supported attributes as part of the (legacy) devices/get request.

Also, I'm wondering why the list of devices is not streamed in the same way that the state for a device is? Same for groups. That would prevent the client having to manually track changes based on events, which seems to be error prone.

And indeed, like I said earlier, none of this is WebSocket-specific. This would just be part of a/the new API, that could be exposed over MQTT as well. I think very little protocol-specific code would be needed. As reliability is important to me, I would be very happy not to have to deal with the extra moving parts of MQTT-over-WebSocket and Mosquitto with a WebSocket plugin. I don't like middlemen. :-)

sjorge, 2020-05-31:
I would love for some abstract device hints to be available outside of the homeassitant stuff.

Some high lever things like:
- switch
- bulb
- thermostat
- sensor

Combined with some feature flags like:
- bulb
  - brightness
  - colorxy
  - color_temp
- thermostat
  - local_temperature
  - occupied_temperature
  - uncoccupied_temperature
- sensor
  - lux
  - temperature
  - humidity
  - motion

Then something like that’s should be easy to adapt to things like HA, homekit, homey, ...

Koenkk, 2020-06-01:
That is definitely something that can be added to the new api indeed.

vanviegen, 2020-06-21:
FYI, I'm continuing work on my stab at this... https://github.com/vanviegen/zigbee2mqtt/
I'll reach out again when/if it reaches a state that I'm happy enough with.

hidaris, 2020-06-22:
May not be relevant, but have you guys considered adding [web of things](https://en.wikipedia.org/wiki/Web_of_Things) support? currently I implement it externally, see [webthing-zigbee](https://github.com/hidaris/webthing-zigbee)

sjorge, 2020-06-22:
Part of this (making it easy to figure out what features a device has) and https://github.com/Koenkk/zigbee2mqtt/issues/3297 should make it easier to integrate with other things like homekit, homie,...

Might be nice to eventually move all of them (including homeassistant) outside of z2m, but we're not there yet. A combo of COVID, new apartment,.. has prevented me from persuing #3297 more of now.

But it would be nice to have some other integrations work easily, but having them all in the core would (IMHO) put much more demand on Koen and other developers helping with z2m. As mentioned above spending some of that time and energy to make integration 3rd party systems be it via mqtt, websockets, z2m extensions,... easier.

closed 2020-08-01 by stale[bot]

LorbusChris, 2021-12-21:
Could this be reopened?

onedr0p, 2022-01-03:
I agree, `zwavejs2mqtt` has web socket integration which cuts out having a MQTT server for integration and IMO is much more stable.