Koenkk/zigbee2mqtt #3623: Error network map after pairing GL-C-008-2ID

nukusinji, 2020-05-28:
# Bug Report
## What happened
Error network map after device(GL-C-008-2ID) pairing.

zigbee2mqtt:info  2020-05-28 09:52:52: Network scan finished
zigbee2mqtt:error 2020-05-28 09:52:52: Failed to call 'NetworkMap' 'onMQTTMessage' (TypeError: Cannot read property 'map' of undefined
    at fingerprintMatch (/app/node_modules/zigbee-herdsman-converters/index.js:75:42)
    at Object.findByDevice (/app/node_modules/zigbee-herdsman-converters/index.js:50:21)
    at /app/lib/extension/networkMap.js:74:61
    at Array.forEach (<anonymous>)
    at NetworkMap.graphviz (/app/lib/extension/networkMap.js:60:24)
    at NetworkMap.onMQTTMessage (/app/lib/extension/networkMap.js:44:65)
    at async Controller.callExtensionMethod (/app/lib/controller.js:352:21))

## What did you expect to happen

## How to reproduce it (minimal and precise)
It is work when I remove GL-C-008-2ID device.

## Debug Info
Zigbee2mqtt version: last-dev
Adapter hardware: CC1352P-2
Adapter firmware version: 20200419

Koenkk, 2020-05-28:
Can you share your `data/database.db` entry of your GL-C-008-2ID?

nukusinji, 2020-05-28:
{"id":14,"type":"Router","ieeeAddr":"0x00124b001fb4db2e","nwkAddr":11415,"manufId":0,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GL-C-007","epList":[11,13,15],"endpoints":{"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"GL-C-007","manufacturerName":"GLEDOPTO","powerSource":1,"zclVersion":1,"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20180920","swBuildId":"2.0.3"}}},"binds":[]},"13":{"profId":49246,"epId":13,"devId":57694,"inClusterList":[4096],"outClusterList":[4096],"clusters":{},"binds":[]},"15":{"profId":49246,"epId":15,"devId":544,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{},"binds":[]}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20180920","swBuildId":"2.0.3","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1590676631369}

nukusinji, 2020-05-28:
> Can you share your `data/database.db` entry of your GL-C-008-2ID?

GL-C-008-2ID shown GL-C-007
{"id":14,"type":"Router","ieeeAddr":"0x00124b001fb4db2e","nwkAddr":11415,"manufId":0,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GL-C-007","epList":[11,13,15],"endpoints":{"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"GL-C-007","manufacturerName":"GLEDOPTO","powerSource":1,"zclVersion":1,"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20180920","swBuildId":"2.0.3"}}},"binds":[]},"13":{"profId":49246,"epId":13,"devId":57694,"inClusterList":[4096],"outClusterList":[4096],"clusters":{},"binds":[]},"15":{"profId":49246,"epId":15,"devId":544,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{},"binds":[]}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20180920","swBuildId":"2.0.3","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1590676631369}

Claude2666, 2020-05-29:
I have the same problem since I upgraded to 0.13.1. It was working fine in 0.13

Koenkk, 2020-05-30:
Can you replace `/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/index.js` with https://gist.github.com/Koenkk/fff03c42fea348e20e62c09bc9040094 and post the logging after this error appears?

nukusinji, 2020-05-30:
> Can you replace `/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/index.js` with https://gist.github.com/Koenkk/fff03c42fea348e20e62c09bc9040094 and post the logging after this error appears?

It's same..
Error is not appear when map is 'raw'. just occur on graphviz.

  error 2020-05-30 22:14:46: Failed to call 'NetworkMap' 'onMQTTMessage' (TypeError: Cannot read property 'map' of undefined                                                                      
      at fingerprintMatch (/app/node_modules/zigbee-herdsman-converters/index.js:76:42)                                                                                                                       
      at Object.findByDevice (/app/node_modules/zigbee-herdsman-converters/index.js:50:21)                                                                                                                    
      at /app/lib/extension/networkMap.js:74:61                                                                                                                                                               
      at Array.forEach (<anonymous>)                                                                                                                                                                          
      at NetworkMap.graphviz (/app/lib/extension/networkMap.js:60:24)                                                                                                                                         
      at NetworkMap.onMQTTMessage (/app/lib/extension/networkMap.js:44:65)                                                                                                                                    
      at async Controller.callExtensionMethod (/app/lib/controller.js:352:21))

Koenkk, 2020-05-30:
I would expect some logging above it this error now. (coming from the `console.log`), can you share that part?

nukusinji, 2020-05-30:
> I would expect some logging above it this error now. (coming from the `console.log`), can you share that part?

It's log from start scan to finish scan

    debug 2020-05-30 23:04:26: Received MQTT message on 'Ryan_home/zigbee2mqtt/bridge/networkmap' with data 'graphviz'
    info  2020-05-30 23:04:26: Starting network scan (includeRoutes 'false')
    debug 2020-05-30 23:04:26: LQI succeeded for 'Coordinator'
    debug 2020-05-30 23:04:27: LQI succeeded for 'living_room_curtain'
    error 2020-05-30 23:04:27: Failed to execute LQI for 'living_room_curtain_light'
    debug 2020-05-30 23:04:27: LQI succeeded for 'router'
    debug 2020-05-30 23:04:30: Received Zigbee message from 'main_room_inner_curtain', type 'attributeReport', cluster 'genBasic', data '{"65281":{"3":37,"5":25,"7":"0x0000000000000000","8":4384,"9":1281,"100":100}}' from endpoint 1 with groupID 0
    debug 2020-05-30 23:04:36: Received Zigbee message from 'teac_turntable_plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
    info  2020-05-30 23:04:36: MQTT publish: topic 'Ryan_home/zigbee2mqtt/teac_turntable_plug', payload '{"power":0,"current":0.01,"voltage":218.6,"linkquality":45,"state":"OFF"}'
    error 2020-05-30 23:04:37: Failed to execute LQI for 'laundry_light'
    debug 2020-05-30 23:04:38: LQI succeeded for 'teac_turntable_plug'
    debug 2020-05-30 23:04:41: Received Zigbee message from 'teac_turntable_plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":5}' from endpoint 1 with groupID 0
    info  2020-05-30 23:04:41: MQTT publish: topic 'Ryan_home/zigbee2mqtt/teac_turntable_plug', payload '{"power":0,"current":0.01,"voltage":218.6,"linkquality":48,"state":"OFF"}'
    debug 2020-05-30 23:04:47: Received Zigbee message from 'computer_room_outlet', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,0]}' from endpoint 1 with groupID 0
    info  2020-05-30 23:04:47: MQTT publish: topic 'Ryan_home/zigbee2mqtt/computer_room_outlet', payload '{"power":0,"linkquality":42,"energy":0,"state":"OFF"}'
    error 2020-05-30 23:04:48: Failed to execute LQI for 'main_room_curtain'
    debug 2020-05-30 23:04:50: Received Zigbee message from 'living_room_curtain', type 'attributeReport', cluster 'genBasic', data '{"65506":21,"appVersion":82}' from endpoint 1 with groupID 0
    debug 2020-05-30 23:04:53: Received Zigbee message from 'computer_room_outlet', type 'attributeReport', cluster 'seMetering', data '{"meteringDeviceType":0,"unitOfMeasure":0}' from endpoint 1 with groupID 0
    debug 2020-05-30 23:04:54: Received Zigbee message from 'computer_room_outlet', type 'attributeReport', cluster 'seMetering', data '{"divisor":1000,"multiplier":1}' from endpoint 1 with groupID 0
    debug 2020-05-30 23:04:54: Received Zigbee message from 'teac_turntable_plug', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsVoltage":2188}' from endpoint 1 with groupID 0
    info  2020-05-30 23:04:54: MQTT publish: topic 'Ryan_home/zigbee2mqtt/teac_turntable_plug', payload '{"power":0,"current":0.01,"voltage":218.8,"linkquality":45,"state":"OFF"}'
    error 2020-05-30 23:04:58: Failed to execute LQI for 'main_room_inner_curtain'
    debug 2020-05-30 23:05:17: Received Zigbee message from 'router', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 8 with groupID 0
    info  2020-05-30 23:05:17: MQTT publish: topic 'Ryan_home/zigbee2mqtt/router', payload '{"led":false,"linkquality":63}'
    error 2020-05-30 23:05:27: Failed to execute LQI for 'kitchen_raspi_plug'
    debug 2020-05-30 23:05:38: Received Zigbee message from 'main_room_sub', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
    info  2020-05-30 23:05:38: MQTT publish: topic 'Ryan_home/zigbee2mqtt/main_room_sub', payload '{"state_top":"OFF","linkquality":63,"state_bottom":"OFF","state_undefined":"OFF"}'
    debug 2020-05-30 23:05:38: Received Zigbee message from 'main_room_curtain', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 57041
    debug 2020-05-30 23:05:38: Skipping re-transmitted Xiaomi message
    debug 2020-05-30 23:05:41: Received Zigbee message from 'computer_room_outlet', type 'attributeReport', cluster 'seMetering', data '{"meteringDeviceType":0,"unitOfMeasure":0}' from endpoint 1 with groupID 0
    debug 2020-05-30 23:05:42: Received Zigbee message from 'computer_room_outlet', type 'attributeReport', cluster 'seMetering', data '{"divisor":1000,"multiplier":1}' from endpoint 1 with groupID 0
    error 2020-05-30 23:05:50: Failed to execute LQI for 'computer_room_outlet'
    info  2020-05-30 23:05:50: Network scan finished
    error 2020-05-30 23:05:50: Failed to call 'NetworkMap' 'onMQTTMessage' (TypeError: Cannot read property 'map' of undefined
        at fingerprintMatch (/app/node_modules/zigbee-herdsman-converters/index.js:76:42)
        at Object.findByDevice (/app/node_modules/zigbee-herdsman-converters/index.js:50:21)
        at /app/lib/extension/networkMap.js:74:61
        at Array.forEach (<anonymous>)
        at NetworkMap.graphviz (/app/lib/extension/networkMap.js:60:24)
        at NetworkMap.onMQTTMessage (/app/lib/extension/networkMap.js:44:65)
        at async Controller.callExtensionMethod (/app/lib/controller.js:352:21))
    debug 2020-05-30 23:05:56: Saving state to file /app/data/state.json

Koenkk, 2020-05-30:
Can you try replacing `/opt/zigbee2mqtt/lib/extension/networkMap.js` with https://gist.github.com/Koenkk/0dc1ab6544def1039abeb566ff6fed20 and provide the logging again?

nukusinji, 2020-05-30:
> Can you try replacing `/opt/zigbee2mqtt/lib/extension/networkMap.js` with https://gist.github.com/Koenkk/0dc1ab6544def1039abeb566ff6fed20 and provide the logging again?

    info  2020-05-30 23:42:57: Starting network scan (includeRoutes 'false')
    error 2020-05-30 23:42:57: Failed to execute LQI for 'living_room_curtain_light'
    error 2020-05-30 23:43:17: Failed to execute LQI for 'laundry_light'
    info  2020-05-30 23:43:17: MQTT publish: topic 'Ryan_home/zigbee2mqtt/router', payload '{"led":false,"linkquality":90}'
    info  2020-05-30 23:43:22: MQTT publish: topic 'Ryan_home/zigbee2mqtt/aqara_double_sw_main', payload '{"battery":100,"voltage":3005,"linkquality":90}'
    error 2020-05-30 23:43:27: Failed to execute LQI for 'main_room_curtain'
    error 2020-05-30 23:43:37: Failed to execute LQI for 'main_room_inner_curtain'
    info  2020-05-30 23:43:44: MQTT publish: topic 'Ryan_home/zigbee2mqtt/aqara_double_sw_main_inner', payload '{"battery":100,"voltage":3015,"linkquality":93}'
    error 2020-05-30 23:43:47: Failed to execute LQI for 'kitchen_raspi_plug'
    error 2020-05-30 23:44:00: Failed to execute LQI for 'computer_room_outlet'
    info  2020-05-30 23:44:00: Network scan finished
    error 2020-05-30 23:44:00: Failed to call 'NetworkMap' 'onMQTTMessage' (TypeError: Cannot read property 'map' of undefined
        at fingerprintMatch (/app/node_modules/zigbee-herdsman-converters/index.js:76:42)
        at Object.findByDevice (/app/node_modules/zigbee-herdsman-converters/index.js:50:21)
        at /app/lib/extension/networkMap.js:75:61
        at Array.forEach (<anonymous>)
        at NetworkMap.graphviz (/app/lib/extension/networkMap.js:60:24)
        at NetworkMap.onMQTTMessage (/app/lib/extension/networkMap.js:44:65)
        at async Controller.callExtensionMethod (/app/lib/controller.js:352:21))

Koenkk, 2020-05-30:
Can you change it to `logger.info('THIS IS DEBUG LOGGING' + JSON.stringify(device));` and post the log?

nukusinji, 2020-05-30:
> Can you change it to `logger.info('THIS IS DEBUG LOGGING' + JSON.stringify(device));` and post the log?

How can I set logger,infor(DEBUG + JSON)??
I can set DEBUG level.

        info  2020-05-30 23:42:57: Starting network scan (includeRoutes 'false')
        error 2020-05-30 23:42:57: Failed to execute LQI for 'living_room_curtain_light'
        error 2020-05-30 23:43:17: Failed to execute LQI for 'laundry_light'
        info  2020-05-30 23:43:17: MQTT publish: topic 'Ryan_home/zigbee2mqtt/router', payload '{"led":false,"linkquality":90}'
        info  2020-05-30 23:43:22: MQTT publish: topic 'Ryan_home/zigbee2mqtt/aqara_double_sw_main', payload '{"battery":100,"voltage":3005,"linkquality":90}'
        error 2020-05-30 23:43:27: Failed to execute LQI for 'main_room_curtain'
        error 2020-05-30 23:43:37: Failed to execute LQI for 'main_room_inner_curtain'
        info  2020-05-30 23:43:44: MQTT publish: topic 'Ryan_home/zigbee2mqtt/aqara_double_sw_main_inner', payload '{"battery":100,"voltage":3015,"linkquality":93}'
        error 2020-05-30 23:43:47: Failed to execute LQI for 'kitchen_raspi_plug'
        error 2020-05-30 23:44:00: Failed to execute LQI for 'computer_room_outlet'
        info  2020-05-30 23:44:00: Network scan finished
        error 2020-05-30 23:44:00: Failed to call 'NetworkMap' 'onMQTTMessage' (TypeError: Cannot read property 'map' of undefined
            at fingerprintMatch (/app/node_modules/zigbee-herdsman-converters/index.js:76:42)
            at Object.findByDevice (/app/node_modules/zigbee-herdsman-converters/index.js:50:21)
            at /app/lib/extension/networkMap.js:75:61
            at Array.forEach (<anonymous>)
            at NetworkMap.graphviz (/app/lib/extension/networkMap.js:60:24)
            at NetworkMap.onMQTTMessage (/app/lib/extension/networkMap.js:44:65)
            at async Controller.callExtensionMethod (/app/lib/controller.js:352:21))

Koenkk, 2020-05-30:
can you try replacing the `networkmap.js` file with https://gist.github.com/Koenkk/0dc1ab6544def1039abeb566ff6fed20 ?

nukusinji, 2020-05-30:
> can you try replacing the `networkmap.js` file with https://gist.github.com/Koenkk/0dc1ab6544def1039abeb566ff6fed20 ?

Here is logs.

    debug 2020-05-31 01:17:30: Received MQTT message on 'Ryan_home/zigbee2mqtt/bridge/networkmap' with data 'graphviz'
    info  2020-05-31 01:17:30: Starting network scan (includeRoutes 'false')
    debug 2020-05-31 01:17:31: LQI succeeded for 'Coordinator'
    debug 2020-05-31 01:17:31: LQI succeeded for 'living_room_curtain'
    error 2020-05-31 01:17:31: Failed to execute LQI for 'living_room_curtain_light'
    debug 2020-05-31 01:17:32: LQI succeeded for 'router'
    debug 2020-05-31 01:17:36: Received Zigbee message from 'computer_room_outlet', type 'attributeReport', cluster 'seMetering', data '{"instantaneousDemand":0}' from endpoint 1 with groupID 0
    info  2020-05-31 01:17:36: MQTT publish: topic 'Ryan_home/zigbee2mqtt/computer_room_outlet', payload '{"power":0,"linkquality":51,"energy":0,"state":"OFF"}'
    debug 2020-05-31 01:17:41: Received Zigbee message from 'computer_room_outlet', type 'attributeReport', cluster 'seMetering', data '{"meteringDeviceType":0,"unitOfMeasure":0}' from endpoint 1 with groupID 0
    error 2020-05-31 01:17:42: Failed to execute LQI for 'laundry_light'
    debug 2020-05-31 01:17:42: Received Zigbee message from 'computer_room_outlet', type 'attributeReport', cluster 'seMetering', data '{"divisor":1000,"multiplier":1}' from endpoint 1 with groupID 0
    debug 2020-05-31 01:17:42: LQI succeeded for 'teac_turntable_plug'
    debug 2020-05-31 01:17:47: Received Zigbee message from 'bathroom', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 2 with groupID 0
    info  2020-05-31 01:17:47: MQTT publish: topic 'Ryan_home/zigbee2mqtt/bathroom', payload '{"state_bottom":"OFF","linkquality":90,"state_center":"OFF","state_top":"OFF"}'
    debug 2020-05-31 01:17:48: Received Zigbee message from 'bathroom', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 3 with groupID 0
    info  2020-05-31 01:17:48: MQTT publish: topic 'Ryan_home/zigbee2mqtt/bathroom', payload '{"state_bottom":"OFF","linkquality":90,"state_center":"OFF","state_top":"OFF"}'
    error 2020-05-31 01:17:52: Failed to execute LQI for 'main_room_curtain'
    debug 2020-05-31 01:17:57: Received Zigbee message from 'living_room_curtain', type 'attributeReport', cluster 'genBasic', data '{"65506":21,"appVersion":82}' from endpoint 1 with groupID 0
    debug 2020-05-31 01:17:59: Received Zigbee message from 'main_room_curtain', type 'attributeReport', cluster 'genBasic', data '{"65281":{"3":34,"5":55,"7":"0x0000000000000000","8":4384,"9":1282,"100":100}}' from endpoint 1 with groupID 0
    error 2020-05-31 01:18:02: Failed to execute LQI for 'main_room_inner_curtain'
    error 2020-05-31 01:18:16: Failed to execute LQI for 'kitchen_raspi_plug'
    debug 2020-05-31 01:18:17: Received Zigbee message from 'router', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 8 with groupID 0
    info  2020-05-31 01:18:17: MQTT publish: topic 'Ryan_home/zigbee2mqtt/router', payload '{"led":false,"linkquality":63}'
    debug 2020-05-31 01:18:29: Received Zigbee message from 'computer_room_outlet', type 'attributeReport', cluster 'seMetering', data '{"meteringDeviceType":0,"unitOfMeasure":0}' from endpoint 1 with groupID 0
    debug 2020-05-31 01:18:30: Received Zigbee message from 'computer_room_outlet', type 'attributeReport', cluster 'seMetering', data '{"divisor":1000,"multiplier":1}' from endpoint 1 with groupID 0
    debug 2020-05-31 01:18:38: Received Zigbee message from 'bathroom_tuya_weather', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2420}' from endpoint 1 with groupID 0
    info  2020-05-31 01:18:38: MQTT publish: topic 'Ryan_home/zigbee2mqtt/bathroom_tuya_weather', payload '{"battery":10,"voltage":2400,"linkquality":54,"temperature":24.2,"humidity":73.8}'
    debug 2020-05-31 01:18:39: Received Zigbee message from 'bathroom_tuya_weather', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":6900}' from endpoint 1 with groupID 0
    info  2020-05-31 01:18:39: MQTT publish: topic 'Ryan_home/zigbee2mqtt/bathroom_tuya_weather', payload '{"battery":10,"voltage":2400,"linkquality":57,"temperature":24.2,"humidity":69}'
    error 2020-05-31 01:18:39: Failed to execute LQI for 'computer_room_outlet'
    info  2020-05-31 01:18:39: Network scan finished
    info  2020-05-31 01:18:39: THIS IS DEBUG LOGGING{"ieeeAddr":"0x00158d0003faa7b1","friendlyName":"pantry_aqara_weather","type":"EndDevice","networkAddress":14336,"manufacturerName":"LUMI","modelID":"lumi.weather","lastSeen":1590854266351}
    info  2020-05-31 01:18:39: THIS IS DEBUG LOGGING{"ieeeAddr":"0xec1bbdfffe8d058f","friendlyName":"living_room_curtain","type":"Router","networkAddress":51693,"manufacturerName":"_TZE200_5zbp6j0u","modelID":"TS0601","failed":[],"lastSeen":1590855477604}
    info  2020-05-31 01:18:39: THIS IS DEBUG LOGGING{"ieeeAddr":"0x00124b001fb4db2e","friendlyName":"living_room_curtain_light","type":"Router","networkAddress":16503,"manufacturerName":"GLEDOPTO","modelID":"GL-C-007","failed":["lqi"],"lastSeen":1590855035293}
    error 2020-05-31 01:18:39: Failed to call 'NetworkMap' 'onMQTTMessage' (TypeError: Cannot read property 'map' of undefined
        at fingerprintMatch (/app/node_modules/zigbee-herdsman-converters/index.js:76:42)
        at Object.findByDevice (/app/node_modules/zigbee-herdsman-converters/index.js:50:21)
        at /app/lib/extension/networkMap.js:75:61
        at Array.forEach (<anonymous>)
        at NetworkMap.graphviz (/app/lib/extension/networkMap.js:60:24)
        at NetworkMap.onMQTTMessage (/app/lib/extension/networkMap.js:44:65)
        at async Controller.callExtensionMethod (/app/lib/controller.js:352:21))

Koenkk, 2020-05-30:
Found the issue, should be fixed in latest dev branch now. Please confirm.

nukusinji, 2020-05-30:
> Found the issue, should be fixed in latest dev branch now. Please confirm.

Okey. Thanks.
But still latest dev is Latest commit e4c3ffe.
 I will check when update.

nukusinji, 2020-05-31:
> Found the issue, should be fixed in latest dev branch now. Please confirm.

I checked it works good.

Thanks~

Koenkk, 2020-06-01:
Cool! thanks for confirming.

closed 2020-06-01 by Koenkk