Koenkk/zigbee2mqtt #3625: MQTT availability messages sent before MQTT connection at startup are lost

crowbarz, 2020-05-28:
# Bug Report
## What happened
After restarting zigbee2mqtt, some of my devices occasionally show as unavailable on Home Assistant. However, they are still sending Zigbee reports which are then also sent via MQTT, and they can still be controlled via zigbee2mqtt by sending commands via MQTT. It is a different set of devices every restart.

## What did you expect to happen
These devices seem to be online to zigbee2mqtt, but is not considered `online` by Home Assistant due to the `zigbee2mqtt/DEVICE/availability` MQTT message being lost. This has been confirmed by sniffing MQTT traffic with MQTT Explorer.

## How to reproduce it (minimal and precise)
Analysis of the logs identified that the MQTT availability messages for the affected devices were attempted to be sent _before_ MQTT was connected. (see lines in the abridged log below marked `>>`, in this instance the `online` MQTT message is dropped `Kitchen Floor Right` and not resent after MQTT is connected, but the `online` message _is_ sent for `Kitchen Floor Left` and all other devices).

```
2020-05-28 00:11:46: Starting zigbee2mqtt versio
n 1.13.1 (commit #3046680)
...
2020-05-28 00:11:48: Setup reporting for '0x0017880100dbd6aa' - 11 - genOnOff
2020-05-28 00:11:48: Setup reporting for '0x0017880100dfff5a' - 11 - genOnOff
>>2020-05-28 00:11:56: Received Zigbee message from 'Kitchen Floor Right', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":32}' from endpoint 1 with groupID 0
>>2020-05-28 00:11:56: Not connected to MQTT server!
>>2020-05-28 00:11:56: Cannot send message: topic: 'zigbee/Kitchen Floor Right/availability', payload: 'online
...
2020-05-28 00:12:11: Connecting to MQTT server at mqtts://mqtt.xxx:8883
2020-05-28 00:12:11: Connected to MQTT server
...
2020-05-28 00:12:12: MQTT publish: topic 'zigbee/Hallway Ground/availability', payload 'online'
2020-05-28 00:12:12: MQTT publish: topic 'zigbee/Kitchen Floor Left/availability', payload 'online'
```
The availability messages (as well as resulting reporting messages) appear to be triggered by a Zigbee message being received by the controller _before_ MQTT is connected. The logs show that availability messages for all devices are normally sent _after_ MQTT is connected. When that happens, any devices for which availability messages have already been sent (but dropped because MQTT is not connected) are skipped, so they remain `offline`.

Suggested possible solutions to this issue:
1. cache the MQTT messages that are sent prior to MQTT connection (and whilst MQTT is otherwise disconnected), and resend them after (re-)connection, and/or:
1. resend availability for the device along with all other devices after MQTT is connected. The [code seems to me to try to do this already](https://github.com/Koenkk/zigbee2mqtt/blame/master/lib/extension/availability.js#L61), but the logs don't show that they are being sent.

## Debug Info
Zigbee2mqtt version: 1.13.1
Adapter hardware: CC2531
Adapter firmware version: 20190619

Koenkk, 2020-05-28:
Should be fixed in latest dev branch (edge for hassio). Could you confirm this?

crowbarz, 2020-05-29:
Looks good now - thanks for fixing!

closed 2020-05-29 by crowbarz