Koenkk/zigbee2mqtt #3627: No matching payload found for entity

Bastiencc, 2020-05-28:
# Bug Report
## What happened
I get lot of this in log :
``` No matching payload found for entity: salon_switch_mur_update_available with state topic: zigbee2mqtt/salon_switch_mur. Payload: , with value template Template("{{ value_json.update_available}}") ```

## What did you expect to happen
What do I need to do ? thanks

## Debug Info
Zigbee2mqtt version: 1.13.1
Adapter hardware: CC2531
Adapter firmware version: latest

Koenkk, 2020-05-28:
For what device is this? 2 possibilities:
- The device still needs to check for an update (after that `update_available` will be published) (can take a few hours)
- We don't support OTA updates for this device (we need to remove `update_available` sensor for this device).

Bastiencc, 2020-05-28:
It's a Philips Hue Dimmer Switch

Bastiencc, 2020-05-29:
I get this everytime I restart HA

Koenkk, 2020-05-29:
Can you check manually for an update? (https://www.zigbee2mqtt.io/information/ota_updates.html#check-if-an-update-is-available)

Make sure to wakeup the device right before executing this.

Bastiencc, 2020-05-30:
I tried, but I get this everytime , I missed something ?
```
May 30 14:20:56 raspberrypi npm[15950]: zigbee2mqtt:info  2020-05-30 14:20:56: MQTT publish: topic 'zigbee2mqtt/salon_switch_canape', payload '{"battery":89,"linkquality":42,"counter":1,"brightness":255,"update_available":true,"action":"on-press","duration":0}'
May 30 14:20:56 raspberrypi npm[15950]: zigbee2mqtt:info  2020-05-30 14:20:56: MQTT publish: topic 'zigbee2mqtt/salon_switch_canape', payload '{"battery":89,"linkquality":42,"counter":1,"brightness":255,"update_available":true,"action":""}'
May 30 14:20:56 raspberrypi npm[15950]: zigbee2mqtt:info  2020-05-30 14:20:56: MQTT publish: topic 'zigbee2mqtt/salon_switch_canape/action', payload 'on-press'

May 30 14:21:00 raspberrypi npm[15950]: zigbee2mqtt:info  2020-05-30 14:21:00: Checking if update available for 'salon_switch_canape'
May 30 14:21:00 raspberrypi npm[15950]: zigbee2mqtt:info  2020-05-30 14:21:00: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"ota_update","message":"Checking if update available for 'salon_switch_canape'","meta":{"status":"checking_if_available","device":"salon_switch_canape"}}'
May 30 14:21:30 raspberrypi npm[15950]: zigbee2mqtt:error 2020-05-30 14:21:30: Failed to check if update available for 'salon_switch_canape' (Device didn't respond to OTA request)
May 30 14:21:30 raspberrypi npm[15950]: zigbee2mqtt:info  2020-05-30 14:21:30: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"ota_update","message":"Failed to check if update available for 'salon_switch_canape' (Device didn't respond to OTA request)","meta":{"status":"check_failed","device":"salon_switch_canape"}}'
```

Koenkk, 2020-05-30:
The payload has `"update_available":true`, do you still get this home assistant warning for this device?

Also can you provide the zigbee2mqtt debug log of at the moment these errors occur?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

Bastiencc, 2020-05-30:
Yes I still get this at each HA restart.
I do not see something special in Z2M log when HA restart : 
``` 
May 30 16:54:53 raspberrypi npm[3369]: zigbee2mqtt:debug 2020-05-30 16:54:53: Received MQTT message on 'hass/status' with data 'online'

May 30 16:55:02 raspberrypi npm[3369]: zigbee2mqtt:debug 2020-05-30 16:55:02: Received Zigbee message from 'chambre_1_reveil', type 'attributeReport', cluster 'genBasic', data '{"65281":{"3":41,"5":49,"7":"0x0000000000000000","8":279,"9":1794,"10":0,"100":0,"101":66,"102":472}}' from endpoint 1 with groupID 0
May 30 16:55:02 raspberrypi npm[3369]: zigbee2mqtt:info  2020-05-30 16:55:02: MQTT publish: topic 'zigbee2mqtt/chambre_1_reveil', payload '{"state":"OFF","brightness":66,"color_temp":472,"linkquality":34}'

May 30 16:55:23 raspberrypi npm[3369]: zigbee2mqtt:info  2020-05-30 16:55:23: MQTT publish: topic 'zigbee2mqtt/cuisine_switch_mur', payload '{"battery":90.5,"linkquality":39,"counter":1,"brightness":255,"update_available":true}'
May 30 16:55:23 raspberrypi npm[3369]: zigbee2mqtt:info  2020-05-30 16:55:23: MQTT publish: topic 'zigbee2mqtt/salon_switch_mur', payload '{"battery":99.5,"linkquality":55,"counter":2,"brightness":255,"update_available":true}'
May 30 16:55:23 raspberrypi npm[3369]: zigbee2mqtt:info  2020-05-30 16:55:23: MQTT publish: topic 'zigbee2mqtt/salon_move', payload '{"battery":100,"voltage":3045,"illuminance":425,"linkquality":55,"occupancy":false,"illuminance_lux":425}'
May 30 16:55:23 raspberrypi npm[3369]: zigbee2mqtt:info  2020-05-30 16:55:23: MQTT publish: topic 'zigbee2mqtt/cuisine_plafond', payload '{"state":"OFF","brightness":254,"color_temp":338,"linkquality":57,"color":{"x":0.4421,"y":0.4103}}'
May 30 16:55:23 raspberrypi npm[3369]: zigbee2mqtt:info  2020-05-30 16:55:23: MQTT publish: topic 'zigbee2mqtt/cuisine_water_leak', payload '{"battery":100,"voltage":3025,"linkquality":57,"water_leak":false}'
May 30 16:55:23 raspberrypi npm[3369]: zigbee2mqtt:info  2020-05-30 16:55:23: MQTT publish: topic 'zigbee2mqtt/chambre_1_plafond', payload '{"state":"OFF","linkquality":57,"brightness":254,"color_temp":338,"color":{"x":0.4421,"y":0.4103}}'
May 30 16:55:23 raspberrypi npm[3369]: zigbee2mqtt:info  2020-05-30 16:55:23: MQTT publish: topic 'zigbee2mqtt/chambre_1_switch_mur', payload '{"battery":90.5,"linkquality":55,"counter":1,"brightness":255,"update_available":true}'
May 30 16:55:23 raspberrypi npm[3369]: zigbee2mqtt:info  2020-05-30 16:55:23: MQTT publish: topic 'zigbee2mqtt/pluie_capteur', payload '{"battery":100,"voltage":3015,"linkquality":42,"water_leak":false}'
May 30 16:55:23 raspberrypi npm[3369]: zigbee2mqtt:info  2020-05-30 16:55:23: MQTT publish: topic 'zigbee2mqtt/chambre_1_switch_chevet', payload '{"battery":65,"linkquality":55,"counter":2,"brightness":255,"update_available":true}'
May 30 16:55:23 raspberrypi npm[3369]: zigbee2mqtt:info  2020-05-30 16:55:23: MQTT publish: topic 'zigbee2mqtt/cigare_capteur', payload '{"temperature":25.07,"linkquality":47,"humidity":66.78,"pressure":993.2,"battery":80,"voltage":2965}'
May 30 16:55:23 raspberrypi npm[3369]: zigbee2mqtt:info  2020-05-30 16:55:23: MQTT publish: topic 'zigbee2mqtt/aqara_cube', payload '{"linkquality":99,"side":3,"battery":100,"voltage":3005,"angle":78.59}'
May 30 16:55:23 raspberrypi npm[3369]: zigbee2mqtt:info  2020-05-30 16:55:23: MQTT publish: topic 'zigbee2mqtt/chambre_1_switch_chevet_2', payload '{"counter":2,"linkquality":31,"battery":100,"brightness":255,"update_available":true}'
May 30 16:55:23 raspberrypi npm[3369]: zigbee2mqtt:info  2020-05-30 16:55:23: MQTT publish: topic 'zigbee2mqtt/wc_move', payload '{"illuminance":141,"linkquality":55,"occupancy":false,"battery":86,"voltage":2975,"illuminance_lux":141}'
May 30 16:55:23 raspberrypi npm[3369]: zigbee2mqtt:info  2020-05-30 16:55:23: MQTT publish: topic 'zigbee2mqtt/salon_switch_canape', payload '{"battery":71,"linkquality":55,"counter":1,"brightness":255,"update_available":true}'
May 30 16:55:23 raspberrypi npm[3369]: zigbee2mqtt:info  2020-05-30 16:55:23: MQTT publish: topic 'zigbee2mqtt/wc_plafond', payload '{"state":"OFF","linkquality":57,"brightness":121,"color_temp":338,"color":{"x":0.4421,"y":0.4103}}'
May 30 16:55:23 raspberrypi npm[3369]: zigbee2mqtt:info  2020-05-30 16:55:23: MQTT publish: topic 'zigbee2mqtt/chambre_1_reveil', payload '{"state":"OFF","brightness":66,"color_temp":472,"linkquality":34}'
May 30 16:55:23 raspberrypi npm[3369]: zigbee2mqtt:info  2020-05-30 16:55:23: MQTT publish: topic 'zigbee2mqtt/chambre_1_capteur', payload '{"battery":91,"voltage":2985,"temperature":24.45,"humidity":37.08,"pressure":992.8,"linkquality":57}'
May 30 16:55:23 raspberrypi npm[3369]: zigbee2mqtt:info  2020-05-30 16:55:23: MQTT publish: topic 'zigbee2mqtt/cave_2_capteur', payload '{"battery":68,"voltage":2945,"temperature":12.42,"humidity":63.54,"linkquality":55}'
May 30 16:55:23 raspberrypi npm[3369]: zigbee2mqtt:info  2020-05-30 16:55:23: MQTT publish: topic 'zigbee2mqtt/cave_1_capteur', payload '{"battery":62,"voltage":2935,"temperature":12.85,"humidity":52.84,"linkquality":55}'
May 30 16:56:01 raspberrypi npm[3369]: zigbee2mqtt:debug 2020-05-30 16:56:01: Received Zigbee message from 'wc_plafond', type 'attributeReport', cluster 'genBasic', data '{"65281":{"3":47,"5":462,"7":"0x0000000000000000","8":279,"9":1795,"10":44426,"100":0,"101":121,"102":338}}' from endpoint 1 with groupID 0
[... cut ...]
- No matching payload found for entity: chambre_1_switch_mur_update_available with state topic: zigbee2mqtt/chambre_1_switch_mur. Payload: , with value template Template("{{ value_json.update_available}}")
- No matching payload found for entity: chambre_1_switch_chevet_update_available with state topic: zigbee2mqtt/chambre_1_switch_chevet. Payload: , with value template Template("{{ value_json.update_available}}")
- No matching payload found for entity: salon_switch_canape_update_available with state topic: zigbee2mqtt/salon_switch_canape. Payload: , with value template Template("{{ value_json.update_available}}")
```

Koenkk, 2020-05-30:
I'm starting to wonder if this is a HA bug. All the device have the `update_available` property. The error message from HA is also very strange: 
```
- No matching payload found for entity: salon_switch_canape_update_available with state topic: zigbee2mqtt/salon_switch_canape. Payload: , with value template Template("{{ value_json.update_available}}")

```

The payload is empty?

```
Payload: , 
```

One thing you could try is:
- stop zigbee2mqtt and home assistant
- Clear retained MQTT messages https://community.openhab.org/t/clearing-mqtt-retained-messages/58221
- Use an MQTT client and subscribe to `zigbee2mqtt/salon_switch_canape`
- Start zigbee2mqtt
- Start home assistant
- Does the error still appear? If thats the case, what are all received messages in the MQTT client?

Bastiencc, 2020-05-30:
I cleared all retained MQTT messages, and now everything is ok 👍 
Thanks for you help and have a nice weekend !

closed 2020-05-30 by Bastiencc