Koenkk/zigbee2mqtt #3628: Ikea blinds pairing

brucey531, 2020-05-28:
Afternoon

I am a little confused by your pairing advise for the Ikea smart blinds here:
https://www.zigbee2mqtt.io/devices/E1757.html

Can you confirm, when you put the blinds into pairing mode, do will they loose connection to the signal repeater and ikea button. You say, make sure repeater is connected first then seems to apply you have to pair the repeater again but its not particularly clear. 

Pairing mode in the ikea instructions says to hold for 2 secs. 5 seconds appears to be a reset of the device so may be worth updating your instructions.

Many thanks

DatafoxChris, 2020-06-06:
Just played around with two Ikea Fyrtur Blinds. They come with a repeater and a button which have been pre-paired with the Blinds. When i joined the repeater to z2m the button didn't control the Blinds anymore. I have also joined the Blinds and the Button to z2m, and used Domoticz to make the button control the Blinds by using a small lua script. I tried to pair the button (using mqtt commands) with the blinds, but couldn't get that to work.

sjorge, 2020-06-06:
You should be able to bind the button to the blinds in z2m:
https://www.zigbee2mqtt.io/information/binding.html

DatafoxChris, 2020-06-06:
I have created one group for one blind and i have two blinds, and i have one button per blind. Created the groups using the Domoticz plugin called Zigbee2MQTT Python Plugin. They are called 'AnkeBlinds1' and 'AnkeBlinds2'. Now i have to do the pairing by sending MQTT commands through MQTT-Explorer:

Topic: zigbee2mqtt/bridge/bind/[AnkeBlinds1]
RAW Payload: [Ikea OpenClose - Anke 1]

The response is:
Failed to call 'Bind' 'onMQTTMessage' (AssertionError [ERR_ASSERTION]: Source undefined or not a device

Do you know why the binding is not working ?

sjorge, 2020-06-06:
It's the other way around, you bind the group to the device.

DatafoxChris, 2020-06-06:
I tried both ways, both result in the same error...

DatafoxChris, 2020-06-11:
I just didn’t need the brackets [] 🙀

brucey531, 2020-06-12:
If you have both repeater and button linked to MQTT, why not just create an automation within HA so the button continues to work.

DatafoxChris, 2020-06-12:
Thanks for responding !   

Yes, i did exactly that. Although i can now directly pair the buttons to the correct fyrturs successfully.

brucey531, 2020-06-12:
Thats good news, So to confirm, you managed to re-pair the button directly back to the blind but leave the repeater connected to Zigbee2Mqtt?

DatafoxChris, 2020-06-12:
I tested the direct-pairing with another switch and a light bulb, and it worked. I noticed that i should not add the square brackets shown in the examples to encapsulate the friendly names in the topic and in the payload. That's why at first i didn't succeed.

closed 2020-07-26 by stale[bot]

rchiileea, 2023-09-27:
Just bumping this as its still a issue and dont want it auto closed