Koenkk/zigbee2mqtt #3631: Cant Start zigbee2mqtt

makeitready, 2020-05-29:
Hi, 

At first, i want you to know that i want to integrade my aquara sensors to my openhab2 system.
Openhab2 runs on a Synology DS116 and the cc2531 stick is plugged into a raspberry pi 3. The mqtt broker (Mosquitto) also runs on the Synology NAS
And i am a beginner in this mqtt topic and programming

So i want to install zigbee2mqtt.
I followed the instructions https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html, but at the command "npm start" it fails. (of course i changed the ip of the mqtt broker in the config file to the Synologys IP

here is what i get: https://pastebin.com/SAXiLqfa

Can somebody pleeeeeeaaaase help me out?
Do i have to set up the mosquitto broker? 
also i wonder why the bright green LED of the Stick only is on then i plug it in, after a few seconds it turns off.

Thanks!

BrunoZumba, 2020-05-29:
Let me try to help.

I dont know much about zigbee2mqtt, i'm just a regular user. I don't use OpenHab either.
But is everything correct with the MQTT section on yout zigbee2mqtt configuration.yaml?
Is the "base_topic" set?

Can you paste the configuration.yaml here (changing/hiding your IP, user, passwords and stuff)?

rvt, 2020-05-29:
what version of node are you using?

makeitready, 2020-05-29:
Hi and thanks for the reply,

the node version is v12.17.0  and the config file is here. https://pastebin.com/rv8zYCyB

soostamas89, 2020-05-31:
Hello for every body!
Majbe not the same, but similar problem.
I use RPI  3 b+ and CC2531 sniffer for zigbee2mqtt. 
I use the following installation guide:
https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html
After the installation, a can start the communication, and I can pair some devices (Xiami...), BUT if i turn off->on (so restart) the raspberry, I can not start zigbee2mqtt any more.
here is the log:
zigbee2mqtt:info  2020-05-31 23:01:25: Starting zigbee-herdsman...
zigbee2mqtt:error 2020-05-31 23:01:26: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-05-31 23:01:26: Failed to start zigbee
zigbee2mqtt:error 2020-05-31 23:01:26: Exiting...
zigbee2mqtt:error 2020-05-31 23:01:26: Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'
    at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:113:32)
    at Generator.next (<anonymous>)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:8:71
    at new Promise (<anonymous>)
    at __awaiter (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:4:12)
    at SerialPort.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:111:49)
    at SerialPort._error (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:198:14)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:242:12
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.13.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.13.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

Thank you in advance if somebody can help!
Tamas

makeitready, 2020-06-01:
Hi there,
i tried to run it again, and i figured out that i couldn't connect to my mqtt broker via desktop app. I solved this Problem and now i can connect to mosquitto. 

But still, rpm start does not work.

The error code is still the same as at first the pastebin

now i can offer you the log file: https://pastebin.com/GY1tPACV

Please give me some help ! :)

Thanks :D 👯

Koenkk, 2020-06-01:
@soostamas89 your adapter is being hijacked by another process (`Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'`). You can find out which by executing `ls -l /proc/[0-9]/fd/ |grep /dev/PORT_PATH`

@makeitready this log is not useful as it's the npm log, not the zigbee2mqtt one, try executing `npm start` in `/opt/zigbee2mqtt` manually

makeitready, 2020-06-01:
Hi @Koenkk 

How do i do that? 😅 
Thanks!

soostamas89, 2020-06-07:
Hello @Koenkk !
Thank you very mucth, and sorry for the delayed answer from my side.
It is working fine. :)
The main problem was, that the Zigbee2MQTT service is already run, and I wanted to started again ...I found the next one : 
systemctl status zigbee2mqtt.service
And after that everything was clear. :)
Thank you again! 
Tamas

soostamas89, 2020-06-07:
Sorry, the next one :
sudo journalctl -u zigbee2mqtt.service -f
:D

closed 2020-07-26 by stale[bot]