Koenkk/zigbee2mqtt #3632: Zigbee2MQTT stops working after a while.....

njordan77, 2020-05-29:
Using latest version 1.13.1 on a Raspi with the 2531 USB Dongle and mostly Ikea stuff attached. ~20 items. and a repeater.

Never had any failure since 1.8.x but since recent updatges ~11/12 its being problematic that suddenly without anyone at home no more zigbee items are recognized by the system....Its all still running as a service (active)....the log file has following entry - not sure if this could have anything to do with the problem.

`error 2020-05-29 00:23:02: Failed to call 'OTAUpdate' 'onZigbeeEvent' (Error: CommandResponse 0x14b457fffe686b52/1 genOta.queryNextImageResponse({"status":149}, {"timeout":10000,"manufacturerCode":null,"disableDefaultResponse":true}) fa$
    at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:369:23)
    at Generator.throw (<anonymous>)
    at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))`

Koenkk, 2020-05-29:
Can you share some more debug logging? Also what firmware are you running?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

njordan77, 2020-05-29:
I have the June'19 firmware on the stick. Guess this is the latest stable that most people will have in use.... I will turn on logging of debug messages, but it may take a few days as last time it went stable for about 3-5 days. Kind Regards Norbert

Koenkk, 2020-05-30:
Both the default and source routing firmware were released in june. In case of stability issues please try the source routing firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

njordan77, 2020-05-30:
Thanks ..... can do this, but if any communication is completely stopped i do not think its a stability issue - more looks like something bigger.

From the log i have the default version...is the source_routing generally a perferred firmware compared to default ??

`info  2020-05-29 08:50:56: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'`

Koenkk, 2020-05-30:
@njordan77 I mean the stability of the adapter itself. The source routing firmware has much more memory available than the default firmware. When the adapter runs out of memory it will crash so communication completely stops. Therefore I would recommend the source routing firmware to see if that solves your problem.

njordan77, 2020-06-01:
hm, now read @firmware subpage the following:

>>>Note that the source routing firmware only supports 5 direct children, therefore you need to have routers in range of the coordinator.

problem/situation is that i mostly have passive ikea switches/motion detectors - and a single repeater. I definitely have more than 5 direct children. Does this then make any sense?
In addition i have ~20 items at all, so not close to the limit.

closed 2020-07-26 by stale[bot]