Koenkk/zigbee2mqtt #3634: FR - Have "update_available" set to "progress"

SeLLeRoNe, 2020-05-29:
# Bug Report
## What happened
I am creating some automations and UI in Home Assistant to show when an update for a zigbee device is available, that part it is easy but I noticed that the update_available remains to "true" even during the update, which might cause problem if an update is started twice (I would asusme it cause problem, not really sure of that)

## What did you expect to happen
It would be nice if the update_available is set to "updating" or anything rather than true/false during the update, so that the system "know" that it doesn't have to trigger the update again.

## How to reproduce it (minimal and precise)
Update any Zigbee device that has an available update.

## Debug Info
Zigbee2mqtt version: 1.31.1
Adapter hardware: CC2531
Adapter firmware version: Can't remember and not sure how to check, but should be this: https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.0.x/bin/CC2531_20190425.zip

sjorge, 2020-05-29:
AFAIK it won't cause trouble because it sees it is already updating.
You can also grab the progress log messages to figure out if one is started, failed or complete. I for got the exact topic but it is spammy so if you're testing it should be obvious which topics you need to subscribe to.

SeLLeRoNe, 2020-05-30:
Ah, I didn't know it was posting the update status in MQTT; I'll look into that, thanks.

Good also to know that if it is already updating it will not start a new update task :)

SeLLeRoNe, 2020-06-04:
Thanks for the advice :) I'll keep that in mind.
To avoid that having this FR applied would help a lot :D

SeLLeRoNe, 2020-07-15:
Just a side note on this, I started two different updates at the same time and both were processed without problem, I thought there was a check to see if one was already in progress,

sjorge, 2020-07-15:
I don’t think there is, I have done 4 bulbs at once with the source routing firmware

~ sjorge

Koenkk, 2020-07-15:
@SeLLeRoNe do you mean updating 2 different devices or the same device twice?

SeLLeRoNe, 2020-07-15:
2 different :) sorry I wasn't clear enough

Koenkk, 2020-07-15:
That is indeed perfectly fine (although be aware to not update too much devices at a time as it will have impact on your network performance: https://www.zigbee2mqtt.io/information/ota_updates.html#update-to-latest-firmware)

SeLLeRoNe, 2020-07-15:
Thanks for the info :)
 An "updating" status would still be very helpful without having to do much work or log analysing to automate everything

Koenkk, 2020-07-16:
@SeLLeRoNe since the new api (#3281) won't publish any progress updates (to `zigbee2mqtt/bridge/log`) between a OTA update request and reply, I was thinking about adding a property `update_progress` to the device topic (e.g. `zigbee2mqtt/FRIENDLY_NAME`). This property will only be present when a device is being updated.

Would that also solve this issue? Since the existence or absence of `update_progress` indicates if a device is being updated.

sjorge, 2020-07-16:
Can’t we reuse the update_available for this? Having it set to ‘inprogress’ when an update is running seems to still be fine given the name.

~ sjorge

Koenkk, 2020-07-16:
I forgot to mention that this `update_progress` will contain the progress of the update in percents (e.g. `"update_progress": 20`).

SeLLeRoNe, 2020-07-16:
That should work, I can put a check on that to define if it is running or not and drive the rest with that, but rather than having 2 properties for the update maybe it's better to have 1 or have them as a "group"?

For instance, rather than update_available and update_progress, we can either have:
update: true/false/percentage
OR an array:
update.state: available/updating/idle
update.progress: progress percentage

It's just an idea, I am sure you'll come up with probably something better to have things organized :)

Koenkk, 2020-07-16:
That's even better added it to TODO of https://github.com/Koenkk/zigbee2mqtt/issues/3281#issue-593587723

Koenkk, 2020-08-02:
This has been implemented for the new api now (https://github.com/Koenkk/zigbee2mqtt/issues/3281). Docs can be found here: https://github.com/Koenkk/zigbee2mqtt.io/blob/new_api/docs/information/ota_updates.md

Closing this now, in case of any issues please report in #3281

closed 2020-08-02 by Koenkk

SeLLeRoNe, 2020-08-03:
Dumb question, is this already on the dev branch?
Because I just updated to latest dev and I can't see the new structure in MQTT (I still see the old one), should I be doing something different?

Thanks

Koenkk, 2020-08-03:
@SeLLeRoNe yes it's in the dev branch, but until the new api is officially released (#3281) you need to add the following to your `configuration.yaml`.

```yaml
experimental:
  new_api: true
```

SeLLeRoNe, 2020-08-03:
Thanks, I'll try and report on the other thread then :)