Koenkk/zigbee2mqtt #3637: Unsupported device with Zigbee model 'undefined'

tuga49, 2020-05-29:
Hi, on my log, sometimes appear this message:

`May 29 20:56:18 server npm[7806]: zigbee2mqtt:warn  2020-05-29 20:56:18: Received message from unsupported device with Zigbee model 'undefined'
May 29 20:56:18 server npm[7806]: zigbee2mqtt:warn  2020-05-29 20:56:18: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
`

I have a CC2531 as a Coordinator
I have a CC2530 + CC2591 as a Router

I started to see those messages since I connected my CC2530 + CC2591 to the zigbee network. I have sensors connected to the Router and they work. But I see this message sometimes.
The firmware I have is: router-cc2530-cc2591-std.hex. I also tried to flash firmware again, but it continues to happen. I also seen to plug and unplug 3 times to reset the connection. But it didn't help. Anyone have an idea or could help me? thanks

closed 2020-07-26 by stale[bot]

junky89, 2020-09-14:
Hi, I have the same problem. I'm running HassOS 4.13 with:

- zzh! CC2652R Multiprotocol RF Stick as a Coordinator
- CC2530 + CC2591 as a Router

I tried with "zigbee2mqtt" and "zigbee2mqtt-edge", both the same results. Log's keep displaying:

`Zigbee2MQTT:warn  2020-09-14 22:52:16: Received message from unsupported device with Zigbee model 'undefined'
Zigbee2MQTT:warn  2020-09-14 22:52:16: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.`

After a CC2530 reset (plug-unplug x3) it displays: 
`Zigbee2MQTT:info  2020-09-14 22:51:16: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x00124b001d3acbf7"},"type":"device_announced"}'` 

and after, the same as above.

tonyldo, 2020-12-23:
Same here:
Device: CC2530+CC2591
Firmware: router-cc2530-cc2591-std.hex
Device type: Router
Support status: Unsupported
Error:
Received message from unsupported device with Zigbee model 'undefined' and manufacturer name 'undefined'

majuss, 2021-02-16:
Exaclty the same on my end, can't use the router.

scollk, 2021-05-06:
I am having the same issue - tried reflashing the device multiple times with the various firmware options, including : router-cc2530-cc2591-std.hex

Are there any further diagnostics we can provide? As far as I’m aware this should be a supported device/other people have them working with zigbee2mqtt

Can we request reopen?

Vebryn, 2021-05-22:
Same error with some devices using CC2652R
Device is properly detected and paired with Zigpy.