Koenkk/zigbee2mqtt #3638: Feature request - Adapter/Network telemetry

to4ko, 2020-05-29:
# Feature request
## What happened
It would be very usefull to have some kind of telemetry for zigbee stuff - network details(num of routers, end devices, direcr childrens for adapter etc) as well as adapter info like free/available memory. Would be very usefull while trying to troubleshoot issues. Especially in case of big networks. 

## What did you expect to happen
Have details regarding my network

## Debug Info
Zigbee2mqtt version: 1.13.1

Spirituss, 2020-05-29:
I also have a lot of devices (more than 100 zigbee) and would like to monitor the quality and architecture of the zigbee network from HA.

Koenkk, 2020-05-30:
I think a lot of information can already be retrieved from the network map (https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgenetworkmap)

to4ko, 2020-06-01:
not actually....will keep looking further. 

but i still pretty sure that it might be very useful to have this kind of information published in specific topic (especially info regarding zigbee adapter - very useful on big networks)

Spirituss, 2020-07-20:
I also have a lot of devices and need in more powerful tool since there are many "bugs" within big network which require detailed analytics.

closed 2020-08-27 by github-actions[bot]