Koenkk/zigbee2mqtt #3640: Vimar 14592.0 Support

matteopiccioni, 2020-05-30:
Hi,
first of all sorry if my english will not be so fine
There is in Italy a Vimar product that is Zigbee3.0 standard (Alexa compatible) and Bluetooth compatible.
Its a wireless connected electronic switch mechanism, all should works without needs to add new wiring, is enough to replace the old Vimar component.
https://www.vimar.com/en/int/catalog/product/index/code/14592.0

The Vimar 14592.0 is in bluetooth mode by default (and in this way you need Vimar HUB),
using Vimar View Wireless app I was able to switch in Zigbee mode (and you should use Alexa)

After I was able to pair it with my cc2531
The successful pairing log was:
```
info  2020-05-28 22:36:28: Successfully interviewed '0x0b.......b9', device has successfully been paired
warn  2020-05-28 22:36:28: Device '0x0b......b9' with Zigbee model '2_Way_Switch_v1.0' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

```
Following this guide:
https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
I my /opt/zigbee2mqtt/data/configuration.yaml I add section: 
```
advanced:
  log_level: debug
```
and in node_modules/zigbee-herdsman-converters/devices.js file I add my new device:
```
    {
        zigbeeModel: ['2_Way_Switch_v1.0'],
        model: '14592.0',
        vendor: 'Vimar',
        description: 'Deviatore Vimar',
        supports: 'on/off',
        fromZigbee: [],
        toZigbee: [],
    },
```

My problem is:
when I try to trigger some actions on my devices, device is not reporting anything

Following the guide I understand that I should add a _configure:_ section 
So I try to copy some of the examples
but when I restart the services nothing happens while I try to log some actions:
```
    {
        zigbeeModel: ['2_Way_Switch_v1.0'],
        model: '14592.0',
        vendor: 'Vimar',
        description: 'Deviatore Vimar',
        supports: 'on/off',
        fromZigbee: [],
        toZigbee: [],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.endpoints[1];
            await bind(endpoint, coordinatorEndpoint, ['genOnOff']);
        },    
    },
```

or I have errors wile zigbee2mqtt.services starts (I try to copy the BTicino ones)

```
   {
        zigbeeModel: ['2_Way_Switch_v1.0'],
        model: '14592.0',
        vendor: 'Vimar',
        description: 'Deviatore Vimar',
        supports: 'on/off',
        fromZigbee: [],
        toZigbee: [],
        meta: {configureKey: 2},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['genIdentify', 'genOnOff', 'genBinaryInput']);
        },    
    },

error 2020-05-30 10:29:11: Failed to configure '0x0b.....b9', attempt 1 (TypeError: Cannot read property 'bind' of undefined
    at bind (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices.js:40:24)
    at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices.js:10296:19)
    at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.js:104:45)
    at Configure.onZigbeeStarted (/opt/zigbee2mqtt/lib/extension/configure.js:72:28)
    at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:350:44))

```

In my 
 /opt/zigbee2mqtt/data/configuration.yaml I can see in device section my Vimar device:

```
devices:
  '0x0b......b9':
    friendly_name: '0x0b.....b9'
```

In my home assistant web page,
into Configuration -> Integrations section, I see the MQTT big icon that shows:
 - 2 devices (my xiaomi and ikea ones)
 - 8 entities (not my Vimar inside)

What should I do?

Thanks for your help 
Matteo

Koenkk, 2020-05-30:
Can you share the `data/database.db` entry of this device?

matteopiccioni, 2020-05-30:
Sure, I will do as soon as I will go back to my house
where should I find this db file? Should I hide or remove some personal data?

Koenkk, 2020-05-30:
It doesn't contain any personal data, you can find it under `data/database.db`, just only post the line of that device.

matteopiccioni, 2020-05-30:
hello,
here to info about the device:

```
{
  "id": 3,
  "type": "Router",
  "ieeeAddr": "0x0b........b9",
  "nwkAddr": 58274,
  "manufId": 4660,
  "manufName": "Vimar",
  "powerSource": "Mains (single phase)",
  "modelId": "2_Way_Switch_v1.0",
  "epList": [
    10,
    242
  ],
  "endpoints": {
    "10": {
      "profId": 260,
      "epId": 10,
      "devId": 0,
      "inClusterList": [
        6,
        0,
        3,
        7
      ],
      "outClusterList": [
        5,
        4,
        3
      ],
      "clusters": {
        "genBasic": {
          "attributes": {
            "modelId": "2_Way_Switch_v1.0",
            "manufacturerName": "Vimar",
            "powerSource": 1,
            "zclVersion": 2,
            "appVersion": 1,
            "stackVersion": 10,
            "hwVersion": 11,
            "dateCode": "20200101",
            "swBuildId": ""
          }
        }
      },
      "binds": []
    },
    "242": {
      "profId": 41440,
      "epId": 242,
      "devId": 97,
      "inClusterList": [],
      "outClusterList": [
        33
      ],
      "clusters": {},
      "binds": []
    }
  },
  "appVersion": 1,
  "stackVersion": 10,
  "hwVersion": 11,
  "dateCode": "20200101",
  "swBuildId": "",
  "zclVersion": 2,
  "interviewCompleted": true,
  "meta": {
    "configured": 1
  },
  "lastSeen": 1590698188943
}
```

Thanks

Koenkk, 2020-06-01:
Try with endpoint 10 instead of 1.

```js
   {
        zigbeeModel: ['2_Way_Switch_v1.0'],
        model: '14592.0',
        vendor: 'Vimar',
        description: 'Deviatore Vimar',
        supports: 'on/off',
        fromZigbee: [],
        toZigbee: [],
        meta: {configureKey: 2},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(10);
            await bind(endpoint, coordinatorEndpoint, ['genIdentify', 'genOnOff', 'genBinaryInput']);
        },    
    },
```

matteopiccioni, 2020-06-01:
It works...Thanks! I am able to turn on and turn off my device sending messages.

In wich way I should add the device into homeassistant.js ?
Should be enough to add the line 
    '14592.0': [cfg.switch],
into mapping section?

Koenkk, 2020-06-01:
yes

matteopiccioni, 2020-06-01:
Yes, I confirm all is working in the right way

Should I submit a pull to the repository?
Or the new device support will be added automatically?

Koenkk, 2020-06-01:
I just added the device, will be supported in the latest dev branch in a few hours.

matteopiccioni, 2020-06-27:
Hello,
I bought one more Vimar 14592.0 device
I was able to pari the device:
```
info  2020-06-25 22:24:40: Device '0x0b010e45f464609b' joined
info  2020-06-25 22:24:40: Starting interview of '0x0b010e45f464609b'
info  2020-06-25 22:24:40: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x0b010e45f464609b"}}'
info  2020-06-25 22:24:40: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x0b010e45f464609b"}}'
info  2020-06-25 22:24:41: MQTT publish: topic 'zigbee2mqtt/Deviatore Vimar', payload '{"state":"OFF","linkquality":23}'
info  2020-06-25 22:24:53: MQTT publish: topic 'zigbee2mqtt/Deviatore Vimar', payload '{"state":"OFF","linkquality":23}'
info  2020-06-25 22:24:54: MQTT publish: topic 'homeassistant/switch/0x0b010e45f464609b/switch/config', payload '{"payload_off":"OFF","payload_on":"ON","value_template":"{{ value_json.state }}","command_topic":"zigbee2mqtt/0x0b010e45f464609b/set","state_topic":"zigbee2mqtt/0x0b010e45f464609b","json_attributes_topic":"zigbee2mqtt/$
info  2020-06-25 22:24:54: MQTT publish: topic 'homeassistant/sensor/0x0b010e45f464609b/linkquality/config', payload '{"icon":"mdi:signal","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x0b010e45f464609b","json_attributes_topic":"zigbee2mqtt/0x0b010e45f464609b","name":"0x0b$
info  2020-06-25 22:25:05: MQTT publish: topic 'zigbee2mqtt/Deviatore Vimar', payload '{"state":"OFF","linkquality":21}'
info  2020-06-25 22:25:08: Successfully interviewed '0x0b010e45f464609b', device has successfully been paired
info  2020-06-25 22:25:08: Device '0x0b010e45f464609b' is supported, identified as: Vimar Deviatore Vimar (14592.0)
info  2020-06-25 22:25:08: Configuring '0x0b010e45f464609b'
info  2020-06-25 22:25:08: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x0b010e45f464609b","model":"14592.0","vendor":"Vimar","description":"Deviatore Vimar","supported":true}}'
info  2020-06-25 22:25:08: Successfully configured '0x0b010e45f464609b'
```
Unfortunately in log I have errors
```
error 2020-06-26 20:55:05: Publish 'set' 'state' to '0x0b010e45f464609b' failed: 'Error: Command 0x0b010e45f464609b/10 genOnOff.on({}, {"timeout":10000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'No network route' (205))'
info  2020-06-26 20:55:05: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to '0x0b010e45f464609b' failed: 'Error: Command 0x0b010e45f464609b/10 genOnOff.on({}, {\"timeout\":10000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: Data$
...
error 2020-06-26 21:14:24: Publish 'set' 'state' to '0x0b010e45f464609b' failed: 'Error: Command 0x0b010e45f464609b/10 genOnOff.on({}, {"timeout":10000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'No network route' (205))'
info  2020-06-26 21:14:24: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to '0x0b010e45f464609b' failed: 'Error: Command 0x0b010e45f464609b/10 genOnOff.on({}, {\"timeout\":10000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: Data$
```

I was able to see it in home assistant zigbee device but If I try to turn on the light I have the following error:
```
debug 2020-06-27 23:45:34: Received MQTT message on 'zigbee2mqtt/0x0b010e45f464609b/set' with data 'ON'
debug 2020-06-27 23:45:34: Publishing 'set' 'state' to '0x0b010e45f464609b'
debug 2020-06-27 23:45:35: Received Zigbee message from 'Deviatore Vimar', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 10 with groupID 0
info  2020-06-27 23:45:35: MQTT publish: topic 'zigbee2mqtt/Deviatore Vimar', payload '{"state":"OFF","linkquality":28}'
error 2020-06-27 23:45:38: Publish 'set' 'state' to '0x0b010e45f464609b' failed: 'Error: Command 0x0b010e45f464609b/10 genOnOff.on({}, {"timeout":10000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'No network route' (205))'
debug 2020-06-27 23:45:38: Error: Command 0x0b010e45f464609b/10 genOnOff.on({}, {"timeout":10000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'No network route' (205))
    at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:349:23)
    at Generator.throw (<anonymous>)
    at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)
info  2020-06-27 23:45:38: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to '0x0b010e45f464609b' failed: 'Error: Command 0x0b010e45f464609b/10 genOnOff.on({}, {\"timeout\":10000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"0x0b010e45f464609b"}}'

```

In my database.db I have this entry:
```
[... cut ...] ":11,"dateCode":"20200101","swBuildId":""}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0008bdb801","endpointID":1}]},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[]}},"appVersion":1,"stackVersion":10,"hwVersion":11,"dateCode":"20200101","swBuildId":"","zclVersion":2,"interviewCompleted":true,"meta":{"configured":3},"lastSeen":1593117732379}
```

What should I do?
Thanks!

Koenkk, 2020-06-28:
Can you try switching to the source routing firmware? https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

matteopiccioni, 2020-06-28:
ok, I will try..unfortunately I need some days to be able to flash the firmware
in the meantime can I try anything else?
Thanks

Koenkk, 2020-06-29:
Sometimes this problem can be solved by replugging or repairing the device.

matteopiccioni, 2020-06-29:
Replug does not works
I try to remove the devices but as I restart zigbee2mqtt service the device appears again
I did 
zigbee2mqtt/bridge/config/remove
with payload 0x0b010e45f464609b
In the log I can see that device is removed successfully but at restart it is still here
I try to remove devices using MQTT Explorer, without luck

How should I remove device?

Koenkk, 2020-06-30:
Try with `zigbee2mqtt/bridge/config/force_remove`

matteopiccioni, 2020-07-01:
force_remove command removed the device, but still it doesn't works
I have just switched to the source routing firmware
I will test as soon as possible

matteopiccioni, 2020-07-01:
Hello,
After force removing the device I am no more able to add It to the network.
I remove power and follow istructions but It does not works
Maybe in the device still thinks to be connected to the zigbee network?
What could I try?
Thanks

matteopiccioni, 2020-07-03:
I was finally able to reset the device and let it join again to the network, with routing firmware

Rarely I am able to turn on the light, and Rarely to turn off command fails, but it fails often

```
error 2020-07-03 21:18:48: Publish 'set' 'state' to 'Luce soggiorno' failed: 'Error: Command 0x0b010e1d73e7bcda/10 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: Timeout - 4115 - 10 - 26 - 6 - 11 after 10000ms)'
info  2020-07-03 21:18:48: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'Luce soggiorno' failed: 'Error: Command 0x0b010e1d73e7bcda/10 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Error: Timeout - 4115 - 10 - 26 - 6 - 11 after 10000ms)'","meta":{"friendly_name":"Luce soggiorno"}}'
```
The device with problem is 'Luce soggiorno'

[image]

```
{"id":5,"type":"Router","ieeeAddr":"0x0b010e1d73e7bcda","nwkAddr":4115,"manufId":4660,"manufName":"Vimar","powerSource":"Mains (single phase)","modelId":"2_Way_Switch_v1.0","epList":[10,242],"endpoints":{"10":{"profId":260,"epId":10,"devId":0,"inClusterList":[6,0,3,7],"outClusterList":[5,4,3],"clusters":{"genBasic":{"attributes":{"modelId":"2_Way_Switch_v1.0","manufacturerName":"Vimar","powerSource":1,"zclVersion":2,"appVersion":1,"stackVersion":10,"hwVersion":11,"dateCode":"20200101","swBuildId":""}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0008bdb801","endpointID":1}]},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[]}},"appVersion":1,"stackVersion":10,"hwVersion":11,"dateCode":"20200101","swBuildId":"","zclVersion":2,"interviewCompleted":true,"meta":{"configured":3},"lastSeen":1593803454081}
{"id":6,"ieeeAddr":"0x0b010e1d73e7bcda","nwkAddr":5272,"epList":[],"endpoints":{},"interviewCompleted":false,"meta":{},"lastSeen":1593193285955}

```

matteopiccioni, 2020-07-03:
oh I see that I have linkquality = 0 for 'Luce soggiorno'

Koenkk, 2020-07-04:
Looks that it has a weak range, try to improve your network range by following: https://www.zigbee2mqtt.io/how_tos/how_to_improve_network_range.html

matteopiccioni, 2020-07-06:
I try to use using an USB extension cable and now the device is working better, 
Thank you

In the next days I will try to add an external antenna to my CC2531,
Thanks again

Koenkk, 2020-07-06:
Great! Assume this can be closed now.

closed 2020-07-06 by Koenkk

legolas0802, 2020-09-20:
yesterday i updated the firmware of the vimar button in question, since that moment it has stopped working. I tried to re-pair but it tells me that the device is not supported:
Zigbee2MQTT:info  2020-09-20 09:05:51: MQTT publish: topic 'zigbee2mqtt/Router C2530', payload '{"led":false,"linkquality":88}'
Zigbee2MQTT:info  2020-09-20 09:06:15: Starting interview of '0xf4ce36bfa1dad3dd'
Zigbee2MQTT:info  2020-09-20 09:06:15: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xf4ce36bfa1dad3dd"},"type":"pairing"}'
Zigbee2MQTT:info  2020-09-20 09:06:15: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xf4ce36bfa1dad3dd"},"type":"device_announced"}'
Zigbee2MQTT:info  2020-09-20 09:06:30: Successfully interviewed '0xf4ce36bfa1dad3dd', device has successfully been paired
Zigbee2MQTT:warn  2020-09-20 09:06:30: Device '0xf4ce36bfa1dad3dd' with Zigbee model 'On_Off_Switch_v1.0' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Zigbee2MQTT:info  2020-09-20 09:06:30: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"0xf4ce36bfa1dad3dd","supported":false},"type":"pairing"}'

Koenkk, 2020-09-20:
Can you try changing the definition in devices.js to:

```js
{
    zigbeeModel: ['2_Way_Switch_v1.0', 'On_Off_Switch_v1.0'],
    model: '14592.0',
    vendor: 'Vimar',
    description: '2-way switch IoT connected mechanism',
    supports: 'on/off',
    fromZigbee: [fz.on_off],
    toZigbee: [tz.on_off],
    meta: {configureKey: 3},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(10);
        await bind(endpoint, coordinatorEndpoint, ['genOnOff']);
    },
},
``` 

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to modify devices.js

legolas0802, 2020-09-20:
> Can you try changing the definition in devices.js to:
>
> ```js
> {
>     zigbeeModel: ['2_Way_Switch_v1.0', 'On_Off_Switch_v1.0'],
>     model: '14592.0',
>     vendor: 'Vimar',
>     description: '2-way switch [...]

Thanks, it works perfectly now

Koenkk, 2020-09-20:
Added.

Changes will be available in the latest dev branch tomrrow (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

legolas0802, 2020-09-20:
thanks, you have been very kind

Mauro Cimino