Koenkk/zigbee2mqtt #3642: New device Zemismart

aureliosaraiva, 2020-05-30:
I am adding support for another Zigbee device from Zemismart.

I managed to create the device correctly and make it recognized by zigbee2mqtt and the home assistant.

However when I try to turn on a lamp by the homeassistant you get this error in the log:

```
I am adding support for another Zigbee device from Zemismart.

I managed to create the device correctly and make it recognized by zigbee2mqtt and the home assistant.

However when I try to turn on a lamp by the homeassistant you get this error in the log:
```

```
zigbee2mqtt:debug 2020-05-30 15:54:59: Received MQTT message on 'zigbee2mqtt/0x00124b001c46142a/top/set' with data 'ON'
zigbee2mqtt:debug 2020-05-30 15:54:59: Publishing 'set' 'state' to '0x00124b001c46142a'
zigbee2mqtt:error 2020-05-30 15:54:59: Publish 'set' 'state' to '0x00124b001c46142a' failed: 'TypeError: Cannot read property 'command' of undefined'
zigbee2mqtt:debug 2020-05-30 15:54:59: TypeError: Cannot read property 'command' of undefined
    at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:122:26)
    at EntityPublish.onMQTTMessage (/app/lib/extension/publish.js:189:52)
    at Controller.callExtensionMethod (/app/lib/controller.js:350:44)
    at Controller.onMQTTMessage (/app/lib/controller.js:244:14)
    at MQTT.emit (events.js:310:20)
    at MQTT.onMessage (/app/lib/mqtt.js:95:14)
    at MqttClient.emit (events.js:310:20)
    at MqttClient._handlePublish (/app/node_modules/mqtt/lib/client.js:1269:12)
    at MqttClient._handlePacket (/app/node_modules/mqtt/lib/client.js:409:12)
    at work (/app/node_modules/mqtt/lib/client.js:320:12)

```

```js
{
        zigbeeModel: ['FB56+ZSW1IKJ2.7'],
        model: 'HGZB-043',
        vendor: 'Zemismart',
        description: 'Smart light switch - 3 gang with neutral wire',
        supports: 'on/off',
        fromZigbee: [fz.ignore_basic_report, fz.on_off],
        toZigbee: [tz.on_off],
        endpoint: (device) => {
            return {'top': 1, 'center': 2, 'bottom': 3};
        },
        meta: {configureKey: 1, multiEndpoint: true},
        configure: async (device, coordinatorEndpoint) => {
            await bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
            await bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
            await bind(device.getEndpoint(3), coordinatorEndpoint, ['genOnOff']);
        },
    },
```
I used this with base, since it is the same, but different model.
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L8418

```js
// homeassistant.js
'HGZB-043': [switchEndpoint('top'), switchEndpoint('center'), switchEndpoint('bottom')],
```

aureliosaraiva, 2020-05-30:
Looking at the `entity` logs arrives as undefined

https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/toZigbee.js#L122

aureliosaraiva, 2020-05-30:
When press button on deviceI received this

`zigbee2mqtt:info  2020-05-30 16:23:14: MQTT publish: topic 'zigbee2mqtt/0x00124b001c46142a', payload '{"state_undefined":"OFF","linkquality":139}'`

Koenkk, 2020-06-01:
Can you share the `database.db` entry of this device?

aureliosaraiva, 2020-06-01:
```
{"id":43,"type":"Router","ieeeAddr":"0x00124b001c46142a","nwkAddr":16071,"manufId":0,"manufName":"Feibit Inc co.  ","powerSource":"Mains (single phase)","modelId":"FB56+ZSW1IKJ2.7","epList":[16,17,18],"endpoints":{"16":{"profId":260,"epId":16,"devId":2,"inClusterList":[0,3,4,5,6],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"FB56+ZSW1IKJ2.7","manufacturerName":"Feibit Inc co.  ","powerSource":1,"zclVersion":1,"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20190601        "}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[]},"17":{"profId":260,"epId":17,"devId":2,"inClusterList":[0,3,4,5,6],"outClusterList":[],"clusters":{"genBasic":{"attributes":{}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[]},"18":{"profId":260,"epId":18,"devId":2,"inClusterList":[0,3,4,5,6],"outClusterList":[],"clusters":{"genBasic":{"attributes":{}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[]}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20190601        ","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1590866654328}
```

Koenkk, 2020-06-02:
Try with

```js
{
        zigbeeModel: ['FB56+ZSW1IKJ2.7'],
        model: 'HGZB-043',
        vendor: 'Zemismart',
        description: 'Smart light switch - 3 gang with neutral wire',
        supports: 'on/off',
        fromZigbee: [fz.ignore_basic_report, fz.on_off],
        toZigbee: [tz.on_off],
        endpoint: (device) => {
            return {'top': 16, 'center': 17, 'bottom': 18};
        },
        meta: {configureKey: 1, multiEndpoint: true},
        configure: async (device, coordinatorEndpoint) => {
            await bind(device.getEndpoint(16), coordinatorEndpoint, ['genOnOff']);
            await bind(device.getEndpoint(17), coordinatorEndpoint, ['genOnOff']);
            await bind(device.getEndpoint(18), coordinatorEndpoint, ['genOnOff']);
        },
    },
```

aureliosaraiva, 2020-06-02:
Now it works, I will test it during this week if everything is ok. I open a Pull Request by adding it over the weekend.

Thank you.

Another thing, explains the difference between using `top: 16` and `top: 1`

Koenkk, 2020-06-02:
good that it works. 

The endpoint() method maps endpoints to friendly names, e.g. endpoint 16 is top. As the device has no endpoint 1, top: 1 won't work. You can find the endpoint number of this device in the database. So for this case `[16,17,18]`.

aureliosaraiva, 2020-06-02:
Got it, thanks for the explanation. I bought some more zigbee devices which, from what I analyzed, is not yet supported. Then when I arrive I will register them in the project too. These instructions help a lot.

Thank you.

aureliosaraiva, 2020-06-05:
I was looking for a device with the same model: https://www.zigbee2mqtt.io/devices/HGZB-043.html

I'm opening a PR adding support for this new one that has a different zigbeeModel. Do I add only zigbeeModel to the list?

closed 2020-07-26 by stale[bot]