Koenkk/zigbee2mqtt #3643: Aqara 2ch relay - click_undefined topic

to4ko, 2020-05-31:
# Bug Report
## What happened
Wrong topic appeared for Aqara 2ch relay. 
Deleting this one helped for some time. Than this topic comes back.

Home Assistant Error:
2020-05-31 10:15:35 ERROR (MainThread) [homeassistant.util.logging] Exception in async_discover when dispatching 'mqtt_discovery_new_device_automation_mqtt': ({'automation_type': 'trigger', 'type': 'click', 'topic': 'zigbee2mqtt/Relay_Fan_B/click', 'device': {'identifiers': ['zigbee2mqtt_0x00158d0001a248e9'], 'name': 'Relay_Fan_B', 'sw_version': 'Zigbee2mqtt 1.13.1', 'model': 'Aqara wireless relay controller (LLKZMK11LM)', 'manufacturer': 'Xiaomi'}, 'platform': 'mqtt'},)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/device_automation.py", line 42, in async_discover
    hass, config, config_entry, discovery_data
  File "/usr/src/homeassistant/homeassistant/components/mqtt/device_trigger.py", line 174, in async_setup_trigger
    config = TRIGGER_DISCOVERY_SCHEMA(config)
  File "/usr/local/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 272, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 594, in validate_dict
    return base_validate(path, iteritems(data), out)
  File "/usr/local/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 432, in validate_mapping
    raise er.MultipleInvalid(errors)
voluptuous.error.MultipleInvalid: required key not provided @ data['subtype']

Topic:
[image]

Payload:
{"automation_type":"trigger","type":"click","topic":"zigbee2mqtt/Relay_Fan_B/click","device":{"identifiers":["zigbee2mqtt_0x00158d0001a248e9"],"name":"Relay_Fan_B","sw_version":"Zigbee2mqtt 1.13.1","model":"Aqara wireless relay controller (LLKZMK11LM)","manufacturer":"Xiaomi"}}

## What did you expect to happen
Do not have this kind of topics
## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2mqtt version: 1.13.1
Adapter hardware: CC2538
Adapter firmware version: {'type': 'zStack30x', 'meta': {'transportrev': 2, 'product': 2, 'majorrel': 2, 'minorrel': 7, 'maintrel': 2, 'revision': 20200211}}

Koenkk, 2020-06-01:
Can you provide the debug logging right before this message is published:

```js
{"automation_type":"trigger","type":"click","topic":"zigbee2mqtt/Relay_Fan_B/click","device":{"identifiers":["zigbee2mqtt_0x00158d0001a248e9"],"name":"Relay_Fan_B","sw_version":"Zigbee2mqtt 1.13.1","model":"Aqara wireless relay controller (LLKZMK11LM)","manufacturer":"Xiaomi"}}
```

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

to4ko, 2020-06-01:
This is it - two cases for two relays

`debug 2020-05-29 16:29:59: Received Zigbee message from 'Relay_Fan_B', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2020-05-29 16:29:59: MQTT publish: topic 'zigbee2mqtt/Relay_Fan_B', payload '{"power":0.05,"consumption":0.24,"temperature":43,"linkquality":118,"state_l1":"OFF","state_l2":"OFF","last_seen":1590758999729}'
info  2020-05-29 16:29:59: MQTT publish: topic 'zigbee2mqtt/Relay_Fan_B', payload '{"power":0.05,"consumption":0.24,"temperature":43,"linkquality":118,"state_l1":"OFF","state_l2":"OFF","last_seen":1590758999729,"click":""}'
info  2020-05-29 16:29:59: MQTT publish: topic 'homeassistant/device_automation/0x00158d0001a248e9/click_undefined/config', payload '{"automation_type":"trigger","type":"click","topic":"zigbee2mqtt/Relay_Fan_B/click","device":{"identifiers":["zigbee2mqtt_0x00158d0001a248e9"],"name":"Relay_Fan_B","sw_version":"Zigbee2mqtt 1.13.1","model":"Aqara wireless relay controller (LLKZMK11LM)","manufacturer":"Xiaomi"}}'
info  2020-05-29 16:29:59: MQTT publish: topic 'zigbee2mqtt/Relay_Fan_B/click', payload 'undefined'
debug 2020-05-29 16:29:59: Received Zigbee message from 'Relay_Valve_B', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2020-05-29 16:29:59: MQTT publish: topic 'zigbee2mqtt/Relay_Valve_B', payload '{"power":0.03,"consumption":0.01,"temperature":46,"linkquality":55,"state_l1":"OFF","state_l2":"OFF","last_seen":1590758999770}'
info  2020-05-29 16:29:59: MQTT publish: topic 'zigbee2mqtt/Relay_Valve_B', payload '{"power":0.03,"consumption":0.01,"temperature":46,"linkquality":55,"state_l1":"OFF","state_l2":"OFF","last_seen":1590758999770,"click":""}'
info  2020-05-29 16:29:59: MQTT publish: topic 'homeassistant/device_automation/0x00158d000366815b/click_undefined/config', payload '{"automation_type":"trigger","type":"click","topic":"zigbee2mqtt/Relay_Valve_B/click","device":{"identifiers":["zigbee2mqtt_0x00158d000366815b"],"name":"Relay_Valve_B","sw_version":"Zigbee2mqtt 1.13.1","model":"Aqara wireless relay controller (LLKZMK11LM)","manufacturer":"Xiaomi"}}'
info  2020-05-29 16:29:59: MQTT publish: topic 'zigbee2mqtt/Relay_Valve_B/click', payload 'undefined'`

`debug 2020-05-29 16:19:58: Received Zigbee message from 'Relay_Fan_B', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2020-05-29 16:19:58: MQTT publish: topic 'zigbee2mqtt/Relay_Fan_B', payload '{"power":0.05,"consumption":0.24,"temperature":42,"linkquality":118,"state_l1":"OFF","state_l2":"OFF","last_seen":1590758398723}'
info  2020-05-29 16:19:58: MQTT publish: topic 'zigbee2mqtt/Relay_Fan_B', payload '{"power":0.05,"consumption":0.24,"temperature":42,"linkquality":118,"state_l1":"OFF","state_l2":"OFF","last_seen":1590758398723,"click":""}'
info  2020-05-29 16:19:58: MQTT publish: topic 'homeassistant/device_automation/0x00158d0001a248e9/click_undefined/config', payload '{"automation_type":"trigger","type":"click","topic":"zigbee2mqtt/Relay_Fan_B/click","device":{"identifiers":["zigbee2mqtt_0x00158d0001a248e9"],"name":"Relay_Fan_B","sw_version":"Zigbee2mqtt 1.13.1","model":"Aqara wireless relay controller (LLKZMK11LM)","manufacturer":"Xiaomi"}}'
info  2020-05-29 16:19:58: MQTT publish: topic 'zigbee2mqtt/Relay_Fan_B/click', payload 'undefined'
debug 2020-05-29 16:19:58: Received Zigbee message from 'Illuminance_MB', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":20087}' from endpoint 1 with groupID 0
info  2020-05-29 16:19:58: MQTT publish: topic 'zigbee2mqtt/Illuminance_MB', payload '{"illuminance":20087,"illuminance_lux":101,"linkquality":97,"battery":100,"voltage":3100,"last_seen":1590758398739}'
debug 2020-05-29 16:19:58: Received Zigbee message from 'Relay_Valve_B', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2020-05-29 16:19:58: MQTT publish: topic 'zigbee2mqtt/Relay_Valve_B', payload '{"power":0.04,"consumption":0.01,"temperature":46,"linkquality":55,"state_l1":"OFF","state_l2":"OFF","last_seen":1590758398763}'
info  2020-05-29 16:19:58: MQTT publish: topic 'zigbee2mqtt/Relay_Valve_B', payload '{"power":0.04,"consumption":0.01,"temperature":46,"linkquality":55,"state_l1":"OFF","state_l2":"OFF","last_seen":1590758398763,"click":""}'
info  2020-05-29 16:19:58: MQTT publish: topic 'homeassistant/device_automation/0x00158d000366815b/click_undefined/config', payload '{"automation_type":"trigger","type":"click","topic":"zigbee2mqtt/Relay_Valve_B/click","device":{"identifiers":["zigbee2mqtt_0x00158d000366815b"],"name":"Relay_Valve_B","sw_version":"Zigbee2mqtt 1.13.1","model":"Aqara wireless relay controller (LLKZMK11LM)","manufacturer":"Xiaomi"}}'
info  2020-05-29 16:19:58: MQTT publish: topic 'zigbee2mqtt/Relay_Valve_B/click', payload 'undefined'`

Koenkk, 2020-06-01:
This issue may already have been fixed in the latest dev branch, can you check?

to4ko, 2020-06-01:
> This issue may already have been fixed in the latest dev branch, can you check?

Assuming that i'm running zigbee2mqtt as HA add-on - should i install dev version and just copy config?

Koenkk, 2020-06-01:
For the HA addon you can use the edge version (which is the zigbee2mqtt dev branch)

to4ko, 2020-06-01:
> For the HA addon you can use the edge version (which is the zigbee2mqtt dev branch)

`zigbee2mqtt:debug 2020-06-01 17:59:28: Received Zigbee message from 'Relay_Valve_B', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-06-01 17:59:28: MQTT publish: topic 'zigbee2mqtt/Relay_Valve_B', payload '{"power":0.01,"consumption":0.01,"temperature":46,"linkquality":57,"state_l1":"OFF","state_l2":"OFF","last_seen":1591023568112}'
zigbee2mqtt:info  2020-06-01 17:59:28: MQTT publish: topic 'zigbee2mqtt/Relay_Valve_B', payload '{"power":0.01,"consumption":0.01,"temperature":46,"linkquality":57,"state_l1":"OFF","state_l2":"OFF","last_seen":1591023568112,"click":""}'
(node:241) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'toString' of undefined
    at HomeAssistant.onPublishEntityState (/app/lib/extension/homeassistant.js:1775:45)
    at EventBus.<anonymous> (/app/lib/extension/homeassistant.js:1704:63)
    at EventBus.emit (events.js:228:7)
    at EventBus.emit (/app/lib/eventBus.js:15:15)
    at Controller.publishEntityState (/app/lib/controller.js:322:23)
(node:241) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 7)`

Koenkk, 2020-06-02:
Fixed the issue, fix is available in the latest dev branch in a few hours.

closed 2020-06-02 by Koenkk