Koenkk/zigbee2mqtt #3644: Home Assistant CC2531 problem with joining new devices

piwoszq, 2020-05-31:
Hello,
I have problem with joining new devices in Home Assistant 0.110.4 and zigbee2mqtt addon 1.13.1.1.

Adapter hardware: CC2531 20190608

I had 2 Xiaomi temperature sensors. Both worked fine, now I have a new Home Assistant installation and I can't add them. CC2531 works on another Home Assistant (my friend's). I copied configuration from his configuration file... still can't add...

Log:
```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
./run.sh: line 17: [Info] Configuration backup found in /share/zigbee2mqtt/.configuration.yaml.bk. Skipping config backup.: No such file or directory
[Info] Socat is DISABLED and not started
2020-05-31T15:51:53: PM2 log: Launching in no daemon mode
2020-05-31T15:51:53: PM2 log: App [npm:0] starting in -fork mode-
2020-05-31T15:51:53: PM2 log: App [npm:0] online
> zigbee2mqtt@1.13.1 start /zigbee2mqtt-1.13.1
> node index.js
zigbee2mqtt:info  2020-05-31 15:51:56: Logging to console and directory: '/share/zigbee2mqtt/log/2020-05-31.15-51-56' filename: log.txt
zigbee2mqtt:debug 2020-05-31 15:51:56: Removing old log directory '/share/zigbee2mqtt/log/2020-05-31.15-30-40'
zigbee2mqtt:debug 2020-05-31 15:51:57: Loaded state from file /share/zigbee2mqtt/state.json
zigbee2mqtt:info  2020-05-31 15:51:57: Starting zigbee2mqtt version 1.13.1 (commit #unknown)
zigbee2mqtt:info  2020-05-31 15:51:57: Starting zigbee-herdsman...
zigbee2mqtt:debug 2020-05-31 15:51:57: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/share/zigbee2mqtt/database.db","databaseBackupPath":"/share/zigbee2mqtt/database.db.backup","backupPath":"/share/zigbee2mqtt/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"},"adapter":{"concurrent":null}}'
zigbee2mqtt:error 2020-05-31 15:53:02: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-05-31 15:53:02: Failed to start zigbee
zigbee2mqtt:error 2020-05-31 15:53:02: Exiting...
zigbee2mqtt:error 2020-05-31 15:53:02: Error: AREQ - ZDO - stateChangeInd after 60000ms
    at Timeout._onTimeout (/zigbee2mqtt-1.13.1/node_modules/zigbee-herdsman/dist/utils/waitress.js:47:35)
    at listOnTimeout (internal/timers.js:531:17)
    at processTimers (internal/timers.js:475:7)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.13.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.13.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm
 ERR! A complete log of this run can be found in:
npm
 ERR!     /root/.npm/_logs/2020-05-31T13_53_02_236Z-debug.log
2020-05-31T15:53:03: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2020-05-31T15:53:03: PM2 log: App [npm:0] starting in -fork mode-
2020-05-31T15:53:03: PM2 log: App [npm:0] online
> zigbee2mqtt@1.13.1 start /zigbee2mqtt-1.13.1
> node index.js
zigbee2mqtt:info  2020-05-31 15:53:05: Logging to console and directory: '/share/zigbee2mqtt/log/2020-05-31.15-53-05' filename: log.txt
zigbee2mqtt:debug 2020-05-31 15:53:05: Removing old log directory '/share/zigbee2mqtt/log/2020-05-31.15-30-46'
zigbee2mqtt:debug 2020-05-31 15:53:06: Loaded state from file /share/zigbee2mqtt/state.json
zigbee2mqtt:info  2020-05-31 15:53:06: Starting zigbee2mqtt version 1.13.1 (commit #unknown)
zigbee2mqtt:info  2020-05-31 15:53:06: Starting zigbee-herdsman...
zigbee2mqtt:debug 2020-05-31 15:53:06: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/share/zigbee2mqtt/database.db","databaseBackupPath":"/share/zigbee2mqtt/database.db.backup","backupPath":"/share/zigbee2mqtt/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"},"adapter":{"concurrent":null}}'
zigbee2mqtt:info  2020-05-31 15:53:08: zigbee-herdsman started
zigbee2mqtt:info  2020-05-31 15:53:08: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'
zigbee2mqtt:debug 2020-05-31 15:53:08: Zigbee network parameters: {"panID":6754,"extendedPanID":"0xdddddddddddddddd","channel":11}
zigbee2mqtt:info  2020-05-31 15:53:08: Currently 0 devices are joined:
zigbee2mqtt:warn  2020-05-31 15:53:08: `permit_join` set to  `true` in configuration.yaml.
zigbee2mqtt:warn  2020-05-31 15:53:08: Allowing new devices to join.
zigbee2mqtt:warn  2020-05-31 15:53:08: Set `permit_join` to `false` once you joined all devices.
zigbee2mqtt:info  2020-05-31 15:53:08: Zigbee: allowing new devices to join.
zigbee2mqtt:info  2020-05-31 15:53:08: Connecting to MQTT server at mqtt://core-mosquitto
zigbee2mqtt:info  2020-05-31 15:53:09: Connected to MQTT server
zigbee2mqtt:info  2020-05-31 15:53:09: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
zigbee2mqtt:info  2020-05-31 15:53:09: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.13.1","commit":"unknown","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"debug","permit_join":true}'
```

Koenkk, 2020-06-01:
Make sure to check: https://www.zigbee2mqtt.io/information/FAQ.html#why-does-my-device-not-or-fail-to-pair

nicandris, 2020-06-21:
geesh... I had an issue with disconnecting devices for a while, i reinstalled zigbee2mqtt 2 weeks ago and it happened again. so I decided to go with a fresh installation and reflashing the cc2531. 

After reflashing I couldn't connect even 1 of my devices. no interview at all. and then i tried the faq...

I can't believe that this was my issue all along. Just a freaking usb extension cable solved it... my pi ended up looking like this but oh well...
[image]

closed 2020-06-21 by Koenkk

mrIngrid, 2020-10-04:
Exactly same problem, only with cc2530+cc2592 
E18-MS1PA1 chip.
With cc2590+cc2691 all is OK.