Koenkk/zigbee2mqtt #3645: Ubisys S2 as binding source

theFork, 2020-05-31:
I would like to bind the second channel of a Ubisys S2 dual switch to a hue bulb. There is nothing attached to the S2 output, I would like to use this only as a switch/remote control. This is what I send via MQTT:                                                           

```
topic: zigbee2mqtt/bridge/bind/eg_bad_spiegelschrank_s2/2
payload: eg_bad_stimmungslicht
```

And this is the result:

```
zigbee2mqtt:error 2020-05-31 15:57:30: Nothing to bind from 'eg_bad_spiegelschrank_s2' to 'eg_bad_stimmungslicht'
zigbee2mqtt:info  2020-05-31 15:57:30: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_bind_failed","message":{"from":"eg_bad_spiegelschrank_s2","to":"eg_bad_stimmungslicht"}}'
```
I've also tried to send  a `/l2` to specify the second endpoint which had the same result.

**Debug Info**
Zigbee2mqtt version:1.13.1.1
Adapter hardware: CC2531
Adapter firmware version: 20190425

theFork, 2020-05-31:
@felixstorm Is this already implemented and should actually work?

felixstorm, 2020-06-02:
@theFork You will probably need to bind from endpoint 4 (i.e. publish to `zigbee2mqtt/bridge/bind/eg_bad_spiegelschrank_s2/4`) as this is the _outgoing_ endpoint for the second _button input_.  
Endpoint 2 only is the _incoming_ endpoint for the second switched _output_ (which you would need to target if you want to control the S2 output from some other ZigBee device).

Looking at the current `devices.js` it _should_ work with this change, but as I do not own a S2, I cannot try myself.

theFork, 2020-06-02:
Tested, works.
Perfect Felix, thanks a lot!

closed 2020-06-02 by theFork