Koenkk/zigbee2mqtt #3647: logging to rsyslog

copystring, 2020-05-31:
Excuse me if this has been asked before (Is this the same? https://github.com/Koenkk/zigbee2mqtt/issues/3080)
I would like to have logging to a remove rsyslog server.
I did fiddle around with rsyslog using the z2m log to send it off to an remove server but I was not happy with it.
For example this does not allow to detect severity of the message (info, error, etc). Instead it is a fixed value like info or whatever is set in the config of rsyslog.
Or does anyone know a better way to do this?

Would it be a lot of work to natively implement this into z2m?

martyzeq, 2020-06-03:
it would be great to have this implemented, many users run z2m on some raspberry with system on sdcard and every log written elsewhere counts, please @Koenkk let our sdcards live longer :)

Koenkk, 2020-06-03:
Zigbee2mqtt uses Winston for logging. I've checked for any rsyslog Winston plugin but they seem to heavily outdated and not maintained anymore (https://github.com/imsky/winston-rsyslog2 / https://github.com/iwater/winston-rsyslog).

There is a well maintained plugin for syslog (https://github.com/winstonjs/winston-syslog) so this would be possible to add.

Would supporting syslog fix this issue?

Koenkk, 2020-06-03:
I did an initial implementation but cannot test it now.

To enable it switch to the latest dev branch https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html

Example configuration

```yaml
advanced:
  log_output: ['console', 'syslog']
  log_syslog:
    host: my_host  # leave out to use default (localhost)
    port: 100 # leave out to use default
```

List of all options can be found here: https://github.com/winstonjs/winston-syslog#usage

copystring, 2020-06-03:
Yeah. Cool. This works.
However the host is always reported as localhost instead of the actual hostname (zigbee2mqtt in my case).
Severity type works fine.
Maybe the facility should be daemon instead of local0?

copystring, 2020-06-04:
For me this is done.
How about you @martyzeq?
Then I would close this issue.

martyzeq, 2020-06-04:
totally awesome 👍 thank you @Koenkk exactly what I was missing

closed 2020-06-04 by copystring

Dylantje, 2020-06-12:
pffff luckily found something about the syslog ..
Lost my system and armbian forum already ended.

The syslog killed my odroid.
I did update domoticz and fast after that z2m .. I did search a long time in my system for a solution. Now I find out that syslog is really something like z2m ...

Only for me the question, i come from a cli update. **Where can i set syslog turn off ...**

@Koenkk . Please make a setting in the config oid and put an explanation at the install wiki ... Days of misery. And my whole, there fore, stable system, bricked. Because I'm not sure what can do a syslog in my odroid. So a lot of misery.

But otherwise very happy with z2m ..

Koenkk, 2020-06-12:
@Dylantje by default syslog logging is not enabled so I'm not sure what happens in your case.

Dylantje, 2020-06-12:
My config:
`homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost'
serial:
  port: /dev/ttyACM1
  disable_led: false
  channel: 11
devices:
  '0x00158d0001a9e2ba':
    friendly_name: 'ZBX: Fietshok pir'
  '0x000d6ffffed65967':
    friendly_name: 'ZBI: Carportcam R'
  '0x00158d00010e4dac':
    friendly_name: 'ZBX: WC Pir'
  '0x00158d0001a4b86d':
    friendly_name: 'ZBXA: Woonkamer'
  '0x00158d00044457a5':
    friendly_name: 'ZBX:  33 BVTHL'
  '0x00158d0001b76f68':
    friendly_name: 'ZBX: Schakelaar over'
  '0x00158d000117b3fd':
    friendly_name: 'ZBX: Ryan Pir'
  '0x00158d000130d60b':
    friendly_name: 'ZBX: Dylan Pir'
  '0x00158d0001ddcd53':
    friendly_name: 'ZBXA: Zolder Overl Pir'
  '0x00158d0001a35d4e':
    friendly_name: 'ZBXA: 22'
  '0x00158d0004102c94':
    friendly_name: 'ZBX: 21 Over'
  '0x00158d000213872d':
    friendly_name: 'ZBXA: Vierkant 1'
  '0x00158d00010dcd46':
    friendly_name: 'ZBX: Rond 1'
  '0xec1bbdfffe47c996':
    friendly_name: 'ZB: Ikea repeater 1 [ Ryan ]'
  '0x00158d0001db35fb':
    friendly_name: 'ZBX: Carport Pir'
  '0x0017880100d7c02a':
    friendly_name: 'ZBHW: Garage R'
  '0x00158d0001f29be2':
    friendly_name: 'ZBX: Bijkeuken Pir'
  '0x00158d0001ddccef':
    friendly_name: 'ZBXA: Keuken Pir'
  '0x00158d00018373ad':
    friendly_name: 'ZBXA: Achterdeur'
  '0x00158d0001a64c71':
    friendly_name: 'ZBXA: Bijkeukendeur'
  '0x00158d0001a8d32e':
    friendly_name: 'ZBX: Gang Pir'
  '0x0017880100ed186c':
    friendly_name: 'ZBHW: Gang R'
  '0x0017880100dbcce7':
    friendly_name: 'ZBHW: Zolder R'
  '0x00178801063405a8':
    friendly_name: 'ZBHR1: Nieuw'
  '0x0017880100e56e0e':
    friendly_name: 'ZBHW: Over'
  '0x0017880100e612b6':
    friendly_name: 'ZBHK: Eettafel 1 R'
  '0x0017880100ffc053':
    friendly_name: 'ZBHK: Eettafel 2 R'
  '0x0017880100fedd0f':
    friendly_name: 'ZBHK: Zitkamer R'
  '0x001788010117afe5':
    friendly_name: 'ZBHG: Kamer TV R'
  '0x0017880100e8dd9d':
    friendly_name: 'ZBHKS: Kachel 1 R'
  '0x0017880100e84bd3':
    friendly_name: 'ZBHKS: Kachel 2 R'
  '0x00158d00014dc9a2':
    friendly_name: 'ZBX: Overloop Pir'
  '0x00158d000117b661':
    friendly_name: 'ZBX: Voordeur'
  '0x00158d000121b9c5':
    friendly_name: 'ZBX: Badkamer T&V'
  '0x00158d00015a9ce3':
    friendly_name: 'ZBX: Garage Pir'
  '0x00158d000149a348':
    friendly_name: 'ZBX: Garagedeur'
  '0x00158d00016da147':
    friendly_name: 'ZBX: Pir over'
  '0x34c69a000bec3df7':
    friendly_name: '0x34c69a000bec3df7'
groups:
  '1':
    friendly_name: 'ZB: Lampen'
    devices:
      - 0x0017880100e612b6/11
      - 0x0017880100ffc053/11
      - 0x0017880100e84bd3/11
      - 0x0017880100fedd0f/11
      - 0x0017880100dbcce7/11
`

[z2m.zip](https://github.com/Koenkk/zigbee2mqtt/files/4773110/z2m.zip)
In the attach my z2m map

martyzeq, 2020-06-12:
It is great to have opportunity to redirect logs from z2m to syslog. You can keep whole syslog in ramdisk or (which is my favourite) have some syslog-ng listening somewhere on your NAS and gathering all logs from your raspberries, oranges, bananas and other fruits ;-) It is nice to have a choice... Many, many choices

Dylantje, 2020-06-13:
I understand... Only my problem is thst the syslog kills my system. it logs to my zram from my odroid, and his in 1 or 2 hour full. And then is my system broke.

I did not config the log. Only is there a way the turn the log OFF. or perhaps otherwise make sure the log  stays very small. [ can i see your config? ]

Dylantje, 2020-06-14:
@Koenkk Please are there opties to turn the syslog off or shrink this log. My ssystem a Odroid N2 with latest armbian is not working any more with z2m.

Koenkk, 2020-06-14:
@Dylantje syslog is turned off by default and has to be explicitly enabled in the config (https://github.com/Koenkk/zigbee2mqtt/issues/3647#issuecomment-638390628). As I don't see this in your config syslog logging of zigbee2mqtt is not enabled. I don't understand how it can end up in your syslog.

Dylantje, 2020-06-14:
@Koenkk my config:
`homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost'
serial:
  port: /dev/ttyACM1
  disable_led: false
  channel: 11
devices:
  '0x00158d0001a9e2ba':
    friendly_name: 'ZBX: Fietshok pir'`

Works for months...
Everything i did was a cli update tot latets z2m

Where can i turn on off the logging?

Or can you give me a example config that is the best way to use?

Koenkk, 2020-06-14:
This config is fine, it doesn't enable syslog.

If you still believe zigbee2mqtt is the issue you can try to remove syslog functionality entirely by changing `/opt/zigbee2mqtt/lib/util/logger.js` and remove the following code:

```js
if (output.includes('syslog')) {
    require('winston-syslog').Syslog;
    const options = {app_name: 'zigbee2mqtt', ...settings.get().advanced.log_syslog};
    transportsToUse.push(new winston.transports.Syslog(options));
}
```

Dylantje, 2020-06-14:
Koen, This part is not in the file you intended.

Turn on z2m is very fast a big syslog.

Koenkk, 2020-06-14:
Can you provide the log when starting zigbee2mqtt?

Dylantje, 2020-06-14:
[syslog.zip](https://github.com/Koenkk/zigbee2mqtt/files/4776620/syslog.zip)

Hope you found a thing...

Koenkk, 2020-06-14:
Can you post your `/opt/zigbee2mqtt/lib/util/logger.js` file here?

Dylantje, 2020-06-14:
[lib.zip](https://github.com/Koenkk/zigbee2mqtt/files/4776754/lib.zip)

Perhaps my whol map?

For your idea..
Worked fine for monds [ Domoticze latest beta ]

This problem i did have with  my "old" system, when i upgrade to z2m 14.

Was thinking armbian, or domoticz errors..
So a lot searching on different forum.... syslog and z2m should be the problem.

I make a lot of changes in a lot places. So my ondriod was instable. [ windows noob so stupid changes, what i have read on forum ]

So i was thinking everything new/clean should be better. :-)

Newest version of odroid/armbian
Newest version of Domoticz
Newest version of z2m 

Only the map data i replaced after the clean install.

Same problem as the old setup :-(

Koenkk, 2020-06-14:
The lines I mentioned in https://github.com/Koenkk/zigbee2mqtt/issues/3647#issuecomment-643773530 are on line 83. If after removing that the problem still occurs, it's for sure not zigbee2mqtt itself logging to syslog.

Dylantje, 2020-06-14:
sorry  did not fin the lines. [ used the search int he document ]

Removed the lines.
Save the file.
start z2m
Reboot the odroid.

Only no differences..
Can i not remove the whole logging or somthing like that?

I use also the z2m plugin from your site..
Perhaps there a setting?
Update that also the latest version?

Can i go back to 1.12 or 1,13 ? How please..

copystring, 2020-07-13:
@Koenkk does it make sense to change messages in syslog to just the actual message instead of the full json?
Right now it looks like this: `{"message":"Failed to ping 'TerassenlichtRechts'","level":"error"}`
Make it look like `Failed to ping 'TerassenlichtRechts'` and it is a lot more readable.

magcode, 2020-07-13:
I also have this issue, when sending log data to the TICK stack. I tried to fiddle around with `json: false` as a config option. But no success. JSON seems to be coming from `winston-syslog`

Koenkk, 2020-07-14:
Indeed seems to be coming from `winston-syslog` and there are no options to control this behaviour (if I read correctly), please open an issue [https://github.com/winstonjs/winston-syslog/issues](https://github.com/winstonjs/winston-syslog/issues) to see if they can consider this.

magcode, 2020-07-15:
I checked again.

It seems we can control it ourselfes.
This is working fine:

```
  require('winston-syslog').Syslog;
  const { format } = require('logform');

  const mostSimpleFormat = format.printf(info => `${info.message}`)

  const winston = require('winston');
  
  options={}
  options.app_name="hello"
  options.host="myserver"
  options.port=6514
  options.protocol="udp4"
  options.type="5424"

  const logger = winston.createLogger({
    levels: winston.config.syslog.levels,
    format: mostSimpleFormat,
    transports: [    
      new winston.transports.Syslog(options)
    ]
  });
logger.info('Clean');
  ```

So the thing is to explictely pass a very simple `format`.

Koenkk, 2020-07-15:
Cool, I missed that. Does this also include the log level itself? Feel free to make a PR .

magcode, 2020-07-15:
I don't think the level is needed. Its already provided to syslog in a structured manner
Here is an example when pushing the sample above via syslog to some logging visualization tool:

```
logger.info('clean');
logger.error('clean');
```

[image]

Alright. Will create a dedicated ticket and my first PR (excited)

Koenkk, 2020-07-15:
In that case forget my comment, looks good :)