Koenkk/zigbee2mqtt #3649: Unsupported Device - IR blaster (zigbee) FNB56-ZIR04FB1.2

chrisjbawden, 2020-06-01:
Looking to add support for a "Nue Zigbee IR Controller" - https://www.amazon.com.au/ZigBee-Controller-Controlling-Appliances-Working/dp/B07DQCC6SN

I understand this devices needs to be added to the following files - 

vi /app/node_modules/zigbee-herdsman-converters/devices.js
vi /app/node_modules/zigbee-herdsman-converters/converters/fromZigbee.js
vi /app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js
vi /app/lib/extension/homeassistant.js

I am just not sure whats needs to be added (lack of technological knowledge)

This is where I am starting - 
note: referencing https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html for instructions

converter:
nue_ircontroller: {
    cluster: '?', #<-------controller?
    type: '?', #<-----controller?
    convert: (model, msg, publish, options) => {
        const temperature = parseFloat(msg.data['measuredValue']) / 100.0;
        return {temperature: temperature};
    },
},

devices:
{
    zigbeeModel: ['ir.blaster'],
    model: 'FNB56-ZIR04FB1.2,
    vendor: 'Nue',
    description: 'Nue Zigbee IR Controller',
    supports: 'IR',
    fromZigbee: [fz.nue_ircontroller], 
    toZigbee: [],
},

Is there a template for an IR I can use for reference?

Environment - 
zigbee2mqtt Home Assistant addon

Device - 
FNB56-ZIR04FB1.2

Logs - 
zigbee2mqtt:warn  2020-06-01 13:09:41: Received message from unsupported device with Zigbee model 'FNB56-ZIR04FB1.2'
zigbee2mqtt:warn  2020-06-01 13:09:41: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

# Bug Report
## What happened
Device paired
Logs indicate device is not supported - instructions provided

## What did you expect to happen
Device to pair
Device to be supported

## How to reproduce it (minimal and precise)
Pair device: FNB56-ZIR04FB1.2

## Debug Info
Zigbee2mqtt version:
Adapter hardware: CC2531
Adapter firmware version:
note: unsure where to retrieve this - please let me know and I will locate it

saury, 2020-06-01:
No IR Blaster among the supported device list so far. 
I suppose it's quite hard for supporting as you need to obtain the contract between this device and its IR bridge.

tricoos, 2020-06-03:
Did you have a look at the URL from the log? https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

chrisjbawden, 2020-06-03:
> Did you have a look at the URL from the log? https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

Yep... thats where I got the following templates - 

converter:
nue_ircontroller: {
cluster: '?', #<-------controller?
type: '?', #<-----controller?
convert: (model, msg, publish, options) => {
const temperature = parseFloat(msg.data['measuredValue']) / 100.0;
return {temperature: temperature};
},
},

devices:
{
zigbeeModel: ['ir.blaster'],
model: 'FNB56-ZIR04FB1.2,
vendor: 'Nue',
description: 'Nue Zigbee IR Controller',
supports: 'IR',
fromZigbee: [fz.nue_ircontroller],
toZigbee: [],
},

Problem is finding something of a guide for an IR type device. I guess for the entry for the converter this is more relevant as it looks like the entry for the device can really be anything you want.

peca89, 2020-06-05:
Since there are no similar devices supported, you probably need to pair it with original supported zigbee bridge, sniff it using zigbe sniffer ([How to sniff Zigbee traffic](https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html) and reverse engineer the protocol. Not easy to do without technical knowledge.

closed 2020-07-26 by stale[bot]