Koenkk/zigbee2mqtt #3650: One of three WSDCGQ11LM is intermittently unavailable

nikolaykasyanov, 2020-06-01:
# Bug Report
## What happened

One of my 3 Aqara temperature, humidity and pressure sensors started going unavailable in Home Assistant all of a sudden. It still reports data but right a sample is received it goes unavailable. Here's how the graph in HASS looks like:

[image]

Here's some logs that look suspicious:

```
zigbee2mqtt_1  | zigbee2mqtt:info  2020-05-29 19:47:19: MQTT publish: topic 'zigbee2mqtt/0x00158d0001fa6f1b', payload '{"battery":100,"voltage":3045,"temperature":22.78,"humidity":47.85,"pressure":1016,"linkquality":23,"last_seen":"2020-05-29T19:47:19+00:00"}'
zigbee2mqtt_1  | zigbee2mqtt:info  2020-05-29 19:47:19: MQTT publish: topic 'zigbee2mqtt/0x00158d0001fa6f1b/availability', payload 'online'
zigbee2mqtt_1  | zigbee2mqtt:debug 2020-05-29 19:48:53: Non-pingable device '0x00158d0001fa6f1b' was last seen '458737.656' seconds ago.
zigbee2mqtt_1  | zigbee2mqtt:info  2020-05-29 19:48:53: MQTT publish: topic 'zigbee2mqtt/0x00158d0001fa6f1b/availability', payload 'offline'
zigbee2mqtt_1  | zigbee2mqtt:debug 2020-05-29 19:53:53: Non-pingable device '0x00158d0001fa6f1b' was last seen '459037.656' seconds ago.
zigbee2mqtt_1  | zigbee2mqtt:debug 2020-05-29 19:58:53: Non-pingable device '0x00158d0001fa6f1b' was last seen '459337.657' seconds ago.
zigbee2mqtt_1  | zigbee2mqtt:debug 2020-05-29 20:03:53: Non-pingable device '0x00158d0001fa6f1b' was last seen '459637.658' seconds ago.
zigbee2mqtt_1  | zigbee2mqtt:debug 2020-05-29 20:08:53: Non-pingable device '0x00158d0001fa6f1b' was last seen '459937.659' seconds ago.
zigbee2mqtt_1  | zigbee2mqtt:debug 2020-05-29 20:13:53: Non-pingable device '0x00158d0001fa6f1b' was last seen '460237.66' seconds ago.
zigbee2mqtt_1  | zigbee2mqtt:debug 2020-05-29 20:18:53: Non-pingable device '0x00158d0001fa6f1b' was last seen '460537.66' seconds ago.
```

Messages like "device was last seen 458737.656 seconds ago" don't seem to make sense right after sending `'online'` to the availability topics. I figured it might be some internal state issue and tried removing and pairing the device again but no luck.

## What did you expect to happen

I expect it to stay available between updates.

## How to reproduce it (minimal and precise)

Not sure.

## Debug Info
Zigbee2mqtt version: 1.13.1
Adapter hardware: CC2531
Adapter firmware version: 20190619

Koenkk, 2020-06-01:
I've pushed a possible fix to the latest dev branch, can you check if it fixes the issue? https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html

nikolaykasyanov, 2020-06-02:
@Koenkk thanks, I'll switch to a recent dev docker image and see what happens.

nikolaykasyanov, 2020-06-03:
@Koenkk looking good after almost 24 hours, thank you for the quick fix!

closed 2020-06-03 by nikolaykasyanov