Koenkk/zigbee2mqtt #3653: Hue motion sensor: green "motion detected" LED

klaernie, 2020-06-01:
# Feature Request
I would like to have my Philips Hue Motion sensors light their green LED upon motion, like they do in the Hue app when you configure their threshold settings.

Reading #897 it sounds possible, and I noticed that OpenHAB can also trigger the Hue API endpoint that does then places the motion sensor in the right mode. I am well aware that this feature would eat my motion sensor batteries, but while trying to determine the best placement over a couple weeks it would be really good to have.

## Debug Info
Zigbee2mqtt version: 1.13.0
Adapter hardware: CC2531
Adapter firmware version: 20190608

Koenkk, 2020-06-02:
If you could sniff the traffic of the hue bridge when these options are being set this could be added. See this [guide](https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html) on how to sniff traffic.

klaernie, 2020-06-02:
I already tried sniffing it out, but failed miserably as I don't understand enough zigbee to understand the captured information. Doesn't help, that my hie zigbee network still has 4 devices on it and way chattier than the 9 devices on the zigbee2mqtt network. Do you have any hint on what packet type to look for (or favourite Wireshark filter)?

Koenkk, 2020-06-02:
I expect these to be write commands, it helps to immediately stop the sniff after you executed the commands. If you post some screenshots I can probably identify the one for you.

closed 2020-07-26 by stale[bot]

uncle-fed, 2021-01-29:
I've started migrating my hue motion sensor to z2m and ended up here (since my motion sensors' LED blinks now every time it detects motion and this bother me a lot). I am not sure what was the solution here in the end, because my sensors blink after pairing with z2m coordinator and I do not find any z2m option to control this LED behaviour in a similar way like I can do it with Hue bridge and Hue REST API.

From https://developers.meethue.com/develop/hue-api/supported-devices/

> "ledindication" (bool)  Turns device LED during normal operation on or off.  Devices might still indicate exceptional operation (Reset, SW Update, Battery Low).

Philips Hue REST API has a very specific, clearly documented and fully working parameter:  `{"ledindication": false / true}` that you can send to the sensor's `config` object any time and it changes its LED behaviour. As far as I can see, this has not been implemented for z2m, or have I overlooked something?

If so, the "fix" here is not really a fix but a workaround (that does not work in my configuration for some reason). The proper implementation should allow users to configure and control LED's behaviour same way like the Hue API does that - at any time, via configuration parameter / mqtt command, not while pairing only.

I can offer my help with sniffing whatever needs to be sniffed (I have Hue bridge and a CC2531stick here).

Koenkk, 2021-01-29:
@uncle-fed is it a green or red led? Red led indicates the coordinator is not responding.

uncle-fed, 2021-01-29:
Green

Koenkk, 2021-01-29:
Can you sniff the traffic with hue when enabling/disabling this? I also have a hue motion sensor but never seen this green led thingy.

uncle-fed, 2021-01-29:
It is an option that you can easily control via an HTTP request (Official Philips Hue API), if you also have Hue Bridge. You simply send `PUT` request with this payload:
`{"ledindication": true}`
to the URL:
`http://<bridge-ip>/api/<security-string>/sensors/<sensor-id>/config`

To get the list of sensors with their IDs first you can set `GET` request to this URL:
`http://<bridge-ip>/api/<security-string>/sensors`

For me it would be a bit of a hassle now to flash sniffer firmware and set up wireshark, but if you don't have Hue Bridge handy, I can do it, of course.

Koenkk, 2021-01-29:
I have a hue bridge, but would appreciate if you could do this (don't forget to share the network key of your hue network)

uncle-fed, 2021-01-31:
```Frame 7189: 65 bytes on wire (520 bits), 65 bytes captured (520 bits) on interface \\.\pipe\zboss_sniffer, id 0
ZBOSS dump, IN, page 0, channel 15
IEEE 802.15.4 Data, Dst: 0xe751, Src: 0x06b1
    Frame Control Field: 0x8861, Frame Type: Data, Acknowledge Request, PAN ID Compression, Destination Addressing Mode: Short/16-bit, Frame Version: IEEE Std 802.15.4-2003, Source Addressing Mode: Short/16-bit
    Sequence Number: 169
    Destination PAN: 0xa71c
    Destination: 0xe751
    Source: 0x06b1
    [Extended Source: PhilipsL_01:02:8e:13:1c (00:17:88:01:02:8e:13:1c)]
    [Origin: 3]
    TI CC24xx-format metadata: FCS OK
ZigBee Network Layer Data, Dst: 0xe751, Src: 0x755b
    Frame Control Field: 0x0248, Frame Type: Data, Discover Route: Enable, Security Data
        .... .... .... ..00 = Frame Type: Data (0x0)
        .... .... ..00 10.. = Protocol Version: 2
        .... .... 01.. .... = Discover Route: Enable (0x1)
        .... ...0 .... .... = Multicast: False
        .... ..1. .... .... = Security: True
        .... .0.. .... .... = Source Route: False
        .... 0... .... .... = Destination: False
        ...0 .... .... .... = Extended Source: False
        ..0. .... .... .... = End Device Initiator: False
    Destination: 0xe751
    Source: 0x755b
    Radius: 29
    Sequence Number: 122
    [Extended Source: PhilipsL_01:01:c0:2c:c4 (00:17:88:01:01:c0:2c:c4)]
    [Origin: 1]
    ZigBee Security Header
        Security Control Field: 0x28, Key Id: Network Key, Extended Nonce
            ...0 1... = Key Id: Network Key (0x1)
            ..1. .... = Extended Nonce: True
        Frame Counter: 45583718
        Extended Source: PhilipsL_01:02:8e:13:1c (00:17:88:01:02:8e:13:1c)
        Key Sequence Number: 0
        Message Integrity Code: 4fcd07f9
        [Key: XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX]
ZigBee Application Support Layer Data, Dst Endpt: 2, Src Endpt: 64
    Frame Control Field: Data (0x00)
        .... ..00 = Frame Type: Data (0x0)
        .... 00.. = Delivery Mode: Unicast (0x0)
        ..0. .... = Security: False
        .0.. .... = Acknowledgement Request: False
        0... .... = Extended Header: False
    Destination Endpoint: 2
    Cluster: Basic (0x0000)
    Profile: Home Automation (0x0104)
    Source Endpoint: 64
    Counter: 187
ZigBee Cluster Library Frame, Mfr: Philips (0x100b), Command: Write Attributes, Seq: 96
    Frame Control Field: Profile-wide (0x04)
        .... ..00 = Frame Type: Profile-wide (0x0)
        .... .1.. = Manufacturer Specific: True
        .... 0... = Direction: Client to Server
        ...0 .... = Disable Default Response: False
    Manufacturer Code: Philips (0x100b)
    Sequence Number: 96
    Command: Write Attributes (0x02)
    Attribute Field, Boolean: 0x01
        Attribute: 0x0033
        Data Type: Boolean (0x10)
        0000 0001 = Boolean: True
```

uncle-fed, 2021-01-31:
Will this be enough? The only difference between turning LED on or off is the very last "Attribute field". The capture above enables blinking green when motion is detected. `False` value turns it off.

uncle-fed, 2021-01-31:
There is also a 'write response' message, but I guess it is a standard thing, so not really valuable.

```Frame 26318: 62 bytes on wire (496 bits), 62 bytes captured (496 bits) on interface \\.\pipe\zboss_sniffer, id 0
ZBOSS dump, IN, page 0, channel 15
IEEE 802.15.4 Data, Dst: 0x755b, Src: 0x06b1
ZigBee Network Layer Data, Dst: 0x755b, Src: 0xe751
ZigBee Application Support Layer Data, Dst Endpt: 64, Src Endpt: 2
    Frame Control Field: Data (0x00)
        .... ..00 = Frame Type: Data (0x0)
        .... 00.. = Delivery Mode: Unicast (0x0)
        ..0. .... = Security: False
        .0.. .... = Acknowledgement Request: False
        0... .... = Extended Header: False
    Destination Endpoint: 64
    Cluster: Basic (0x0000)
    Profile: Home Automation (0x0104)
    Source Endpoint: 2
    Counter: 121
ZigBee Cluster Library Frame, Mfr: Philips (0x100b), Command: Write Attributes Response, Seq: 53
    Frame Control Field: Profile-wide (0x1c)
        .... ..00 = Frame Type: Profile-wide (0x0)
        .... .1.. = Manufacturer Specific: True
        .... 1... = Direction: Server to Client
        ...1 .... = Disable Default Response: True
    Manufacturer Code: Philips (0x100b)
    Sequence Number: 53
    Command: Write Attributes Response (0x04)
    Status Record
        Status: Success (0x00)
```

uncle-fed, 2021-01-31:
BTW, now after my experiments and re-pairing motion sensor with z2m coordinator I do no longer get green light blinking on motion. Is there some already existing code in z2m that maybe "disables all LEDs" ? :-) I hope that by making this a user controlled option the behaviour can be controlled same way like in the Hue bridge.

Koenkk, 2021-01-31:
I'm not aware of any code to disable this. 

Can you apply the changes from https://github.com/Koenkk/zigbee-herdsman-converters/pull/2186/files and check if it work? You can set it under the exposes tab of this device in the frontend.

uncle-fed, 2021-01-31:
Yes, it works!
Just curious now, whether this is an attribute that is possible to "get", since it belongs together (at least from the Philips API point of view) with other config attributes, such as 'sensitivity' that is already possible to query now, including the web frontend.

uncle-fed, 2021-01-31:
Here is a Report Attributes message that can be observed while sniffing Hue network...

```
Frame 1779: 69 bytes on wire (552 bits), 69 bytes captured (552 bits) on interface \\.\pipe\zboss_sniffer, id 0
ZBOSS dump, IN, page 0, channel 15
IEEE 802.15.4 Data, Dst: 0x0011, Src: 0x2e94
ZigBee Network Layer Data, Dst: 0x755b, Src: 0x2e94
    Frame Control Field: 0x0248, Frame Type: Data, Discover Route: Enable, Security Data
        .... .... .... ..00 = Frame Type: Data (0x0)
        .... .... ..00 10.. = Protocol Version: 2
        .... .... 01.. .... = Discover Route: Enable (0x1)
        .... ...0 .... .... = Multicast: False
        .... ..1. .... .... = Security: True
        .... .0.. .... .... = Source Route: False
        .... 0... .... .... = Destination: False
        ...0 .... .... .... = Extended Source: False
        ..0. .... .... .... = End Device Initiator: False
    Destination: 0x755b
    Source: 0x2e94
    Radius: 30
    Sequence Number: 21
    [Extended Source: PhilipsL_01:02:10:20:7e (00:17:88:01:02:10:20:7e)]
    [Origin: 1222]
    ZigBee Security Header
        Security Control Field: 0x28, Key Id: Network Key, Extended Nonce
            ...0 1... = Key Id: Network Key (0x1)
            ..1. .... = Extended Nonce: True
        Frame Counter: 4678180
        Extended Source: PhilipsL_01:02:10:20:7e (00:17:88:01:02:10:20:7e)
        Key Sequence Number: 0
        Message Integrity Code: 85eec7e5
        [Key: XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX]
ZigBee Application Support Layer Data, Dst Endpt: 65, Src Endpt: 2
    Frame Control Field: Data (0x00)
        .... ..00 = Frame Type: Data (0x0)
        .... 00.. = Delivery Mode: Unicast (0x0)
        ..0. .... = Security: False
        .0.. .... = Acknowledgement Request: False
        0... .... = Extended Header: False
    Destination Endpoint: 65
    Cluster: Basic (0x0000)
    Profile: Home Automation (0x0104)
    Source Endpoint: 2
    Counter: 48
ZigBee Cluster Library Frame, Mfr: Philips (0x100b), Command: Report Attributes, Seq: 148
    Frame Control Field: Profile-wide (0x1c)
        .... ..00 = Frame Type: Profile-wide (0x0)
        .... .1.. = Manufacturer Specific: True
        .... 1... = Direction: Server to Client
        ...1 .... = Disable Default Response: True
    Manufacturer Code: Philips (0x100b)
    Sequence Number: 148
    Command: Report Attributes (0x0a)
    Attribute Field, Boolean: 0x00
        Attribute: 0x0032
        Data Type: Boolean (0x10)
        0000 0000 = Boolean: False
    Attribute Field, Boolean: 0x00
        Attribute: 0x0033
        Data Type: Boolean (0x10)
        0000 0000 = Boolean: False
```

Koenkk, 2021-02-01:
@uncle-fed added it and works correctly with my hue motion sensor.

uncle-fed, 2021-02-01:
Thanks a lot. I'd just propose to capitalize the word "led" in the description text for the frontend.

MattWestb, 2021-02-02:
Yes `led` its something else then `LED` in English.

Koenkk, 2021-02-02:
@uncle-fed done