Koenkk/zigbee2mqtt #3654: Joining a hue bridge to the z2m network

alekslyse, 2020-06-02:
Would it be possible to join the hue bridge to the z2m network to be able to update the firmware of hue devices without having to move the bulbs over.

I see some had some success on deconz, but not sure how it works now

alekslyse, 2020-06-02:
Or it seems like firmwares are now public. Can we use the firmwares on that site for its Philips in z2m

 https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/OTA-Image-Types---Firmware-versions

Koenkk, 2020-06-02:
- The hue bridge cannot join a z2m network
- Updating of Hue bulbs is already supported in the latest dev branch (for the ones we have a firmware for) https://www.zigbee2mqtt.io/information/ota_updates.html

closed 2020-06-02 by Koenkk