Koenkk/zigbee2mqtt #3655: Segmentation fault after updating to 1.13

ReDetection, 2020-06-02:
<details>
Before submitting an issue make sure you've searched for a similar issue and read the documentation: https://www.zigbee2mqtt.io/.
✅

Rules (don't ignore these, your issue will be closed without further notice):
- ✅English only
- ❌Make sure you are on the latest Zigbee2mqtt version - that's the point of the issue
- ✅Provide a clear description of the problem
- ✅DON'T copy logs directly here, post a link to https://hastebin.com/ or https://pastebin.com/.
- ❓Make sure you are running the latest firmware https://github.com/koenkk/z-stack-firmware.

Did you read the FAQ?
- ✅ Yes

Zigbee2mqtt fails to start?
- https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start

✅My case isn't there

</details>

# Bug Report
## What happened

I updated zigbee2mqtt from 1.12.2 to 1.13.1 and now it has Segmentation fault after start. I find this particularly interesting that network becomes online for a brief moment (so Home Assistant retains some history) and then immediately fails. If I go back to 1.12.2, everything works fine.

## What did you expect to happen

No segmentation fault.

## How to reproduce it (minimal and precise)

I am not sure which would repeat the issue for you, but basically all I did was 

```
git checkout 1.13.1
npm i
npm start
```

## Debug Info
Zigbee2mqtt version: 1.13.1 fails, 1.12.2 works fine
Adapter hardware: CC2531
Adapter firmware version: 1.2 without source routing
```json
{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":2018090201}}
```
System version: Debian 10 Buster
I have 32-bit architecture. `Linux bookserver 4.19.0-9-686-pae #1 SMP Debian 4.19.118-2 (2020-04-29) i686 GNU/Linux`
node: v10.19.0
npm: v6.14.5

npm log:
https://pastebin.com/tdUbr3D9

Tail of the `DEBUG=zigbee-herdsman* npm start` output:
https://pastebin.com/ph0TiZ3n

Koenkk, 2020-06-02:
Can you try upgrading your node version to 12?

ReDetection, 2020-06-02:
I would love to, but as far as I know there is no node v12 for 32-bit processors

Koenkk, 2020-06-02:
You can find an unofficial build here: https://unofficial-builds.nodejs.org/download/release/v12.16.3/

ReDetection, 2020-06-02:
Oh thanks, didn't knew that. I will report back the status

ReDetection, 2020-06-03:
Thanks, with node v12 the issue is resolved! so far no segfault

Not sure if I should close this – the installation guide tells 10.x should be fine 
https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html#2-installing

What is your opinion on this?

closed 2020-06-03 by Koenkk