Koenkk/zigbee2mqtt #3656: Device specific calibration isn't working

daha76, 2020-06-02:
# Bug Report
## Device specific calibration isn't working. I have Xiaomi WSDCGQ11LM Temp/Hum/Pressure sensor and pressure reported by the sensor is 33.5 hPa lower. So I found in sensor description possibility to calibrate it manually. My configuration for the sensor looks like this:

devices:
  '0x00158d000449d2df':
    friendly_name: 'THP1'
	pressure_precision: 1
	pressure_calibration: 33

but sensor still uses raw value.
Then I tried global calibration and I added this in my config:

device_options:
  pressure_calibration: 33

This helped and my sensor starts to report correct/calibrated value.

## What did you expect to happen
I would expect that the device specific calibration will be taken into account, because different sensors can report different values for the same attribute like temperature, humidity, pressure, ...

## How to reproduce it (minimal and precise)
Add Xiaomi WSDCGQ11LM and try to calibrate pressure

## Debug Info
Zigbee2mqtt version: 1.13.1 (commit #2cb234c)
Adapter hardware: CC2531
Adapter firmware version: CC2531_DEFAULT_20190608

Koenkk, 2020-06-02:
Make sure the indentation is correct, it should work with

```yaml
devices:
  '0x00158d000449d2df':
    friendly_name: 'THP1'
    pressure_precision: 1
    pressure_calibration: 33
```

daha76, 2020-06-03:
Hmm it was OK in my config. I take a closer look and found that there is mixture of space and tab characters. Could it be that tab and space characters can cause that it wan't work? I changed tab character with spaces for those pressure_xxx lines in device specific config and now it seems that it works. Looks strange to me... but finally it seems to be resolved.

Koenkk, 2020-06-03:
Good, a mix of spaces and tabs indeed doesn't work well in yaml.

closed 2020-06-03 by Koenkk