Koenkk/zigbee2mqtt #3658: zigbee2mqtt not working after migrating from raspberry pi to virtual box

echo4019, 2020-06-02:
Hi I've just moved Hassio over to a virtual machine after running it on a raspberry pi, my snapshot sucsessfully uploaded and everything is working except the zigbee2mqtt add on. 

getting the following error message in the log "zigbee2mqtt:error 2020-06-02 22:53:58: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/ttyACM0'"

Any help appreciated.

tricoos, 2020-06-03:
Well, did you virtualize the physical device - the Zigbee stick? Would this actually work? You need to be able to access the physical hardware here and probably have to change the device path.

echo4019, 2020-06-03:
Yes that's what I thought. I've been looking for a guide to explain how to get the virtual machine to read the ZigBee stick plugged into the usb of the host pc?

echo4019, 2020-06-03:
I installed Hassos on VirtualBox then restored my home assistant from a snapshot. What's the difference in using docker?

closed 2020-07-26 by stale[bot]

bakerboy908, 2021-02-11:
Did you ever solver this issue, im trying to do the same with virtualbox, cant pass the usbthough so im trying to pass just the com port