Koenkk/zigbee2mqtt #3660: ArchLinux  nodejs version update

nottledim, 2020-06-03:
Arch Linux update includes nodejs version 14.  This is not supported yet by zigbee2mqtt.

Node v14 requires a library libicu version 67 whereas node v13 requires libicu v65.  Unfortunately other programs such as ffmpeg have specific version dependencies on libicu as well.  It appears libicu has changed.

I have no idea why zigbe2mqtt does not run with node v14 but it would be very helpful if it could be qualified to do so soon.

Many thanks

Koenkk, 2020-06-03:
Node 14 doesn't seem to directly be the issue, it works fine on MacOS. More logging is needed to investigate this.

```
zigbee2mqtt on  dev [$!] is 📦 v1.13.1-dev via ⬢ v14.4.0 
➜ node --version
v14.4.0

zigbee2mqtt on  dev [$!] is 📦 v1.13.1-dev via ⬢ v14.4.0 
➜ npm start     

> zigbee2mqtt@1.13.1-dev start /Users/Koenkk/Git/zigbee2mqtt
> node index.js

zigbee2mqtt:info  2020-06-03 19:34:14: Logging to console and directory: '/Users/Koenkk/Git/zigbee2mqtt/data/log/2020-06-03.19-34-14' filename: log
.txt
zigbee2mqtt:info  2020-06-03 19:34:14: Starting zigbee2mqtt version 1.13.1-dev (commit #8dc535e)
zigbee2mqtt:info  2020-06-03 19:34:14: Starting zigbee-herdsman...
zigbee2mqtt:info  2020-06-03 19:34:15: zigbee-herdsman started
zigbee2mqtt:info  2020-06-03 19:34:15: Coordinator firmware version: '{"type":"zStack30x","meta":{"transportrev":2,"product":2,"majorrel":2,"minorr
el":7,"maintrel":2,"revision":20190425}}'
zigbee2mqtt:info  2020-06-03 19:34:15: Currently 0 devices are joined:
zigbee2mqtt:warn  2020-06-03 19:34:15: `permit_join` set to  `true` in configuration.yaml.
zigbee2mqtt:warn  2020-06-03 19:34:15: Allowing new devices to join.
zigbee2mqtt:warn  2020-06-03 19:34:15: Set `permit_join` to `false` once you joined all devices.
zigbee2mqtt:info  2020-06-03 19:34:15: Zigbee: allowing new devices to join.
Zigbee started
zigbee2mqtt:info  2020-06-03 19:34:16: Connecting to MQTT server at mqtt://localhost
zigbee2mqtt:info  2020-06-03 19:34:16: Connected to MQTT server
zigbee2mqtt:info  2020-06-03 19:34:16: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
zigbee2mqtt:info  2020-06-03 19:34:16: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.13.1-dev","commit":"8dc535e","coordi
nator":{"type":"zStack30x","meta":{"transportrev":2,"product":2,"majorrel":2,"minorrel":7,"maintrel":2,"revision":20190425}},"log_level":"info","pe
rmit_join":true}'
```

Koenkk, 2020-06-03:
That change is available in the latest dev branch, can this be closed now?

nottledim, 2020-06-03:
yes.  much appreciated.

closed 2020-06-03 by Koenkk