Koenkk/zigbee2mqtt #3662: Xiaomi Plugs\Aqara Sockets announcing themselves toggling on\off

to4ko, 2020-06-04:
# Bug Report
## What happened
Plugs\Sockets some time gets announced themselves toggling on\off. This toggling not traced in logs but physically happening (visually and audio confirmed)

`debug 2020-06-04 14:43:43: Received Zigbee message from 'Plug_Kettle_K', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info  2020-06-04 14:43:43: MQTT publish: topic 'zigbee2mqtt/Plug_Kettle_K', payload '{"power":0.55,"linkquality":89,"last_seen":1591271023202,"state":"ON","voltage":null,"consumption":363.9,"temperature":37}'
.
debug 2020-06-04 14:43:45: Device 'Plug_Kettle_K' announced itself
info  2020-06-04 14:43:45: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"Plug_Kettle_K"}}'
.
debug 2020-06-04 14:43:48: Received Zigbee message from 'Plug_Kettle_K', type 'attributeReport', cluster 'genBasic', data '{"dateCode":"09-29-2017"}' from endpoint 1 with groupID 0
debug 2020-06-04 14:43:49: Received Zigbee message from 'Plug_Kettle_K', type 'attributeReport', cluster 'genOnOff', data '{"61440":50271488,"onOff":1}' from endpoint 1 with groupID 0
info  2020-06-04 14:43:49: MQTT publish: topic 'zigbee2mqtt/Plug_Kettle_K', payload '{"power":0.55,"linkquality":86,"last_seen":1591271029093,"state":"ON","voltage":null,"consumption":363.9,"temperature":37}'
debug 2020-06-04 14:43:49: Received Zigbee message from 'Plug_Kettle_K', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
.
debug 2020-06-04 14:43:50: Received Zigbee message from 'Plug_Kettle_K', type 'attributeReport', cluster 'genBasic', data '{"65281":{"3":25,"5":36,"7":"0x0000000000000000","8":4894,"9":2816,"100":1,"149":363.79205322265625,"152":0,"154":0}}' from endpoint 1 with groupID 0
info  2020-06-04 14:43:50: MQTT publish: topic 'zigbee2mqtt/Plug_Kettle_K', payload '{"power":0,"linkquality":86,"last_seen":1591271030857,"state":"ON","voltage":null,"consumption":363.79,"temperature":25}'
.
debug 2020-06-04 14:43:51: Received Zigbee message from 'Plug_Kettle_K', type 'attributeReport', cluster 'genOnOff', data '{"61440":50271488,"onOff":1}' from endpoint 1 with groupID 0
info  2020-06-04 14:43:51: MQTT publish: topic 'zigbee2mqtt/Plug_Kettle_K', payload '{"power":0,"linkquality":89,"last_seen":1591271031854,"state":"ON","voltage":null,"consumption":363.79,"temperature":25}'
debug 2020-06-04 14:43:52: Received Zigbee message from 'Plug_Kettle_K', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
.
debug 2020-06-04 14:44:01: Received Zigbee message from 'Plug_Kettle_K', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info  2020-06-04 14:44:01: MQTT publish: topic 'zigbee2mqtt/Plug_Kettle_K', payload '{"power":0,"linkquality":89,"last_seen":1591271041195,"state":"ON","voltage":null,"consumption":363.79,"temperature":25}'
`

## What did you expect to happen
expected to not toggle devices

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2mqtt version: 1.13.1 (same on 1.13)
Adapter hardware: CC2538
Adapter firmware version: {'type': 'zStack30x', 'meta': {'transportrev': 2, 'product': 2, 'majorrel': 2, 'minorrel': 7, 'maintrel': 2, 'revision': 20200211}}

Koenkk, 2020-06-06:
Could you provide a sniff of the moment this happens? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

to4ko, 2020-06-07:
> Could you provide a sniff of the moment this happens? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

here you go - it's simply turns computer off :( that never happens with xiaomi gateways.
[image]

[plug_brix_v_sel-announce_toggle_10.14.05.pcapng.gz](https://github.com/Koenkk/zigbee2mqtt/files/4741614/plug_brix_v_sel-announce_toggle_10.14.05.pcapng.gz)

Koenkk, 2020-06-07:
I don't see any commands send to this device leading to the announcement/toggle.

However I noticed one strange thing, on No. 78, 1714, 2812, 3848, 363, 1238 another Zigbee network is communicating on the same channel (not a zigbee2mqtt network). Are you aware of this? Maybe it's causing the trouble?

to4ko, 2020-06-07:
totally forgot about one xiaomi gateway. switched off now. keep looking

PS is there any way to check amount of ram on stick (free \ used)?

Koenkk, 2020-06-08:
I've checked the adapter Z-Stack documentation but couldn't find a command that exposes this information.

highlandr, 2020-06-22:
Hello guys, I have one plug acting like this :) Got no fix to it, must be firmware bug? I've tried to add it to the gateway, check all the configuration and then add it back to the zigbee2mqtt network. No fix...

closed 2020-08-01 by stale[bot]

to4ko, 2020-08-17:
@Koenkk checking thru devices.js file found that lumi.plug (as well as aqara socket) has no fz.ignore_time_read while EU version of lumi plug has it. maybe they are not getting correct time from time to time and once some kind of buffer is full it simply reboots the device?

Koenkk, 2020-08-17:
`fz.ignore_time_read` is just a cosmetic thing. The actual responsible is already done earlier (in zigbee-hersman) and doesn't depend on wether the device has this or not.

It's more likely that in your case (because of your large network) the plug receives/routes too many messages causing it to crash. Obviously this won't happen with the Xiaomi gateway since the number of devices is capped at 64.

to4ko, 2020-08-17:
> `fz.ignore_time_read` is just a cosmetic thing. The actual responsible is already done earlier (in zigbee-hersman) and doesn't depend on wether the device has this or not.
>
> It's more likely that in [...]

so, only option is to split my network and run two networks or simply switch back to xiaomi gw's...

to4ko, 2020-08-17:
Is it possible to change device type from router to end device by changing in database.db? So i might be able to turn plugs to end devices and use couple of 2538 flashed with router firmware

Koenkk, 2020-08-18:
No, that wont change the actual type itself. A device cannot change type without changing the software on it.

to4ko, 2020-08-18:
:(

Wiil plan to get back to xiaomi gw’s

lucas-su, 2020-12-15:
I hope it's okay to jump on an old thread, as I'm experiencing a very similar issue, whereby one channel on an aqara double relay module toggles every time I click an IKEA symfonisk button. These are the only two devices on the network and it does not happen when I rotate the button, which makes me think the amount of messages is not crashing the relay firmware. The button is known for spamming the network a bit, but it is not set up in legacy mode and again if the amount of messages is the problem, I'd expect it too on rotation. 

Considering this, do you think the cause of this behaviour is crashing firmware? 

My log is as follows if this helps:

`Zigbee2MQTT:debug 2020-12-15 21:15:59: Received Zigbee message from 'Symfonisk 1', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0`
`Zigbee2MQTT:info  2020-12-15 21:15:59: MQTT publish: topic 'zigbee2mqtt/Symfonisk 1', payload '{"action":"toggle","brightness":126,"linkquality":102,"rate":195,"update_available":false}'`
`Zigbee2MQTT:info  2020-12-15 21:15:59: MQTT publish: topic 'zigbee2mqtt/Symfonisk 1', payload '{"action":"","brightness":126,"linkquality":102,"rate":195,"update_available":false}'`
`Zigbee2MQTT:info  2020-12-15 21:15:59: MQTT publish: topic 'zigbee2mqtt/Symfonisk 1/action', payload 'toggle'`
`Zigbee2MQTT:debug 2020-12-15 21:15:59: Received Zigbee message from 'Symfonisk 1', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0`
`Zigbee2MQTT:debug 2020-12-15 21:15:59: Received Zigbee message from 'Aqara relay', type 'attributeReport', cluster 'genOnOff', data '{"61440":185810520,"onOff":0}' from endpoint 1 with groupID 0`
`Zigbee2MQTT:info  2020-12-15 21:15:59: MQTT publish: topic 'zigbee2mqtt/Aqara relay', payload '{"consumption":2.96,"energy":2.96,"linkquality":102,"power":0.08,"state":"OFF","state_l1":"OFF","temperature":27,"voltage":229.3}'`
`Zigbee2MQTT:debug 2020-12-15 21:15:59: Received Zigbee message from 'Aqara relay', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0`

buzzcz, 2022-08-11:
I'm experiencing the same thing with Aqara Plug SP-EUC01 toggling it's state when I press my Sonoff SNZB-01. It was happening with certain one, then I removed it and now it's happening with the other one. Couldn't find any connection between those two. Any help would be appreciated. Will try to get sniff later