Koenkk/zigbee2mqtt #3663: [Request] Topic to instruct zigbee2mqtt to rebroadcast?

LukeChannings, 2020-06-04:
Hi, 

This is a general request. Is it possible to have a topic that causes zigbee2mqtt to rebroadcast all of its saved state?

Right now I'm achieving the same effect by re-starting zigbee2mqtt, but if there was simply a topic I could publish to to trigger a rebroadcast that seems a little nicer.

I know zigbee2mqtt is meant to listen on `hass/state` and rebroadcast, and there's also the option to retain messages. But I still think this would be useful generally. Is there already such a topic and I just didn't read [this](https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html) correctly?

Thoughts?

Koenkk, 2020-06-06:
This is currently not possible but could land in the new api #3281

LukeChannings, 2020-06-10:
Ok, thanks @Koenkk !

closed 2020-06-10 by LukeChannings