Koenkk/zigbee2mqtt #3665: zigbee2mqtt on ZZH dongle does not pair anything

jrhbcn, 2020-06-04:
Hi, 

I am trying to setup a new [ZZH](https://electrolama.com/projects/zig-a-zig-ah/#contact) dongle with the latest zigbee2mqtt. I have managed to start zigbee2mqtt by removing the antena (otherwise it complains of used panID or channel). In ay case it seems to start fine. 

However, after I put the antena nothing pairs, I have several xiaomi devices around (temperature, door, water) but none of them manage to finish the pairing process. 

Log does not show anything, running "DEBUG=* npm start" results in the following  just when trying to pair however, this message does not always happens and I am not sure if it is related:

`zigbee-herdsman:controller:log Permit joining +3m
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - mgmtPermitJoinReq - {"addrmode":15,"dstaddr":65532,"duration":254,"tcsignificance":0} +3m
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,5,37,54,15,252,255,254,0,228] +3m
  serialport/stream _write 10 bytes of data +3m
  serialport/binding-abstract write 10 bytes +3m
  serialport/bindings/unixWrite Starting write 10 bytes offset 0 bytesToWrite 10 +3m
  serialport/bindings/unixWrite write returned: wrote 10 bytes +16ms
  serialport/bindings/unixWrite Finished writing 10 bytes +1ms
  serialport/stream binding.write write finished +17ms
  serialport/bindings/poller received "readable" +3m
  serialport/bindings/unixRead Starting read +3m
  serialport/bindings/unixRead Finished read 14 bytes +0ms
  serialport/stream binding.read finished { bytesRead: 14 } +14ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,54,0,82,254,3,69,182,0,0,0,240] +3m
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,54,0,82,254,3,69,182,0,0,0,240] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 54 - [0] - 82 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - mgmtPermitJoinReq - {"status":0} +3m
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,182,0,0,0,240] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 182 - [0,0,0] - 240 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - mgmtPermitJoinRsp - {"srcaddr":0,"status":0} +3m
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":2,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[25,3,2,11,254,0]}} +35ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,26,36,2,2,253,255,0,0,0,0,0,0,242,0,0,242,33,0,2,0,30,6,0,25,3,2,11,254,0,234] +37ms
  serialport/stream _write 31 bytes of data +6ms
  serialport/binding-abstract write 31 bytes +37ms
  serialport/bindings/unixWrite Starting write 31 bytes offset 0 bytesToWrite 31 +20ms
  serialport/stream _read reading { start: 986, toRead: 64550 } +0ms
  serialport/binding-abstract read +1ms
  serialport/bindings/unixRead Starting read +8ms
  serialport/bindings/unixWrite write returned: wrote 31 bytes +2ms
  serialport/bindings/unixWrite Finished writing 31 bytes +0ms
  serialport/stream binding.write write finished +2ms
  serialport/bindings/unixRead read error { [Error: EAGAIN: resource temporarily unavailable, read] errno: -11, code: 'EAGAIN', syscall: 'read' } +1ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +1ms
  serialport/bindings/poller Polling for "readable" +10ms
  serialport/bindings/poller received "readable" +5ms
  serialport/bindings/unixRead Starting read +5ms
  serialport/bindings/unixRead Finished read 6 bytes +0ms
  serialport/stream binding.read finished { bytesRead: 6 } +6ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,2,0,103] +13ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,2,0,103] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 2 - [0] - 103 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequestExt - {"status":0} +14ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  serialport/stream _read reading { start: 992, toRead: 64544 } +2ms
  serialport/binding-abstract read +11ms
  serialport/bindings/unixRead Starting read +4ms
  serialport/bindings/unixRead read error { [Error: EAGAIN: resource temporarily unavailable, read] errno: -11, code: 'EAGAIN', syscall: 'read' } +1ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +0ms
  serialport/bindings/poller Polling for "readable" +5ms
  serialport/bindings/poller received "readable" +5ms
  serialport/bindings/unixRead Starting read +5ms
  serialport/bindings/unixRead Finished read 8 bytes +1ms
  serialport/stream binding.read finished { bytesRead: 8 } +9ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,225,242,2,214] +11ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,225,242,2,214] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [225,242,2] - 214 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":225,"endpoint":242,"transid":2} +24ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  serialport/stream _read reading { start: 1000, toRead: 64536 } +2ms
  serialport/binding-abstract read +9ms
  serialport/bindings/unixRead Starting read +2ms
  serialport/bindings/unixRead read error { [Error: EAGAIN: resource temporarily unavailable, read] errno: -11, code: 'EAGAIN', syscall: 'read' } +1ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +0ms
  serialport/bindings/poller Polling for "readable" +4ms
`
Any ideas what might be happening?

sjorge, 2020-06-04:
Did you set
```
advanced:
  rtscts: false
```

As mentioned on the zzh wiki?

jrhbcn, 2020-06-05:
Thanks @sjorge, I do have that in my config. This is the complete configuration.yaml:

```
homeassistant: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost'
serial:
  port: /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0
advanced:
  rtscts: false
  pan_id: 7023
  channel: 25
  log_level: debug
```

Everything seems OK now starting (no lights on the ZZH dongle which I don't know if it is normal). 

In fact now I see something else in configuration.yaml:
```
devices:
  '0x00158d0002735717':
    friendly_name: '0x00158d0002735717'
```

So I guess the paring did something but did not finish properly :?

sjorge, 2020-06-05:
There is indeed no LED active on ZZH.
Yep it did seem to hav found atleast one device, the entry in database.db should show if the interviewComplete is set to true or false.

jrhbcn, 2020-06-05:
Again, thanks @sjorge! Your comment pointed my to the right direction. After my problems starting zigbee2mqtt (had to remove the antena from the ZZH dongle for it to start or z2m complained of used panID or channel) I tried to pair devices using a USB extension cable to separate the ZZH dongle from the raspberry. That seemed to do the trick. I could pair three devices in a row. 

I'll leave the comment here just to help others if needed. In my case, ZZH seemed to be very sensitive to interference, removing the antena to start z2m and using an extension USB cable for operation solved the issue.

Regards,

closed 2020-06-05 by jrhbcn