Koenkk/zigbee2mqtt #3667: Trådfri E1810 now lists action_group for all actions

sjorge, 2020-06-04:
# Bug Report
## What happened
I paired a E1810 I picked up at IKEA today, wanted to see how different they were from the E1524's.

I noticed all there commands now have 'action_group' set in the mqtt payload, since this device only has one group (random or manually configured) it doesn't seem very useful to include that in every mqtt message.

```
zigbee2mqtt:info  2020-06-04 17:35:26: MQTT publish: topic 'zigbee2mqtt/0xccccccfffe381ebc', payload '{"linkquality":5,"action":"toggle","action_group":22898}
```

## What did you expect to happen
The action_group not be present, as it is always the same for this device.  Having an action_group only makes sense if the device has multiple groups it can send too.

## How to reproduce it (minimal and precise)
Pair a E1810 (or E1524 presubaly) and click a button.

## Debug Info
Zigbee2mqtt version: `1.13.1-dev`
Adapter hardware: CC2531
Adapter firmware version: `{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190619}},`

Koenkk, 2020-06-06:
The used converter is `command_toggle`, as E1524 and E1810 use the same definition (cannot be distinguished), please check if you see the same behaviour for the E1524. If so create a new converter and use it:

```js
E1524_E1810_toggle: {
        cluster: 'genOnOff',
        type: 'commandToggle',
        convert: (model, msg, publish, options, meta) => {
            const payload = {action: getProperty('toggle', msg, model)};
            return payload;
        },
    },
```

sjorge, 2020-06-06:
```zigbee2mqtt:info  2020-06-06 08:02:54: MQTT publish: topic 'zigbee2mqtt/bathroom/lights/remote', payload '{"linkquality":34,"update_available":false,"battery":87,"action":"toggle","action_group":54744}'
zigbee2mqtt:info  2020-06-06 08:03:20: MQTT publish: topic 'zigbee2mqtt/bedroom/lights/remote', payload '{"linkquality":0,"update_available":false,"battery":87,"action":"toggle","action_group":22898}'
```

```
  '0xccccccfffe381ebc':
    friendly_name: bedroom/lights/remote
  '0x90fd9ffffee8e67a':
    friendly_name: bathroom/lights/remote
```

```json
{"id":26,"type":"EndDevice","ieeeAddr":"0xccccccfffe381ebc","nwkAddr":12286,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Battery","modelId":"TRADFRI remote control","epList":[1],"endpoints":{"1":{"profId":49246,"epId":1,"devId":2096,"inClusterList":[0,1,3,9,2821,4096],"outClusterList":[3,4,5,6,8,25,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRI remote control","manufacturerName":"IKEA of Sweden","powerSource":3,"zclVersion":1,"appVersion":32,"stackVersion":87,"hwVersion":1,"dateCode":"20190708","swBuildId":"2.3.014"}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":87}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b001938a7e5","endpointID":1}]}},"appVersion":32,"stackVersion":87,"hwVersion":1,"dateCode":"20190708","swBuildId":"2.3.014","zclVersion":1,"interviewCompleted":true,"meta":{"configured":1,"reporting":1},"lastSeen":1591298962313},
{"id":16,"type":"EndDevice","ieeeAddr":"0x90fd9ffffee8e67a","nwkAddr":53853,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Battery","modelId":"TRADFRI remote control","epList":[1],"endpoints":{"1":{"profId":49246,"epId":1,"devId":2096,"inClusterList":[0,1,3,9,2821,4096],"outClusterList":[3,4,5,6,8,25,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRI remote control","manufacturerName":"IKEA of Sweden","powerSource":3,"zclVersion":1,"appVersion":32,"stackVersion":87,"hwVersion":1,"dateCode":"20190708","swBuildId":"2.3.014"}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":87}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b001938a7e5","endpointID":1}]}},"appVersion":32,"stackVersion":87,"hwVersion":1,"dateCode":"20190708","swBuildId":"2.3.014","zclVersion":1,"interviewCompleted":true,"meta":{"configured":1,"reporting":1},"lastSeen":1591422676043}
```

So yeah both have the same behavior, the bathroom one is an E1524 (shiny buttons, no extra rubber seal) and the bedroom one is an E1810 (matt buttons, extra rubber seal)

sjorge, 2020-06-06:
Looks good with the change

```
zigbee2mqtt:info  2020-06-06 08:09:49: MQTT publish: topic 'zigbee2mqtt/bedroom/lights/remote', payload '{"linkquality":0,"update_available":false,"battery":60,"action":"toggle"}'
```

I'll do a PR against the repo.

sjorge, 2020-06-06:
Fixed with next converters update

closed 2020-06-06 by sjorge