Koenkk/zigbee2mqtt #3670: New device support, Saturn South energy meter

l34rn3d, 2020-06-05:
Trying to add support for an energy meter. 
the values are all non standard,  see pastebin link for manufacture supplied attributes.

have two issue, 
1. 
I dont understand how to bind non standard address in devices.js,
i have tried to steal the code from the meazon din rail meter to no success. 

```
	// saturnsouth
    {
        zigbeeModel: ['SS9007.2.0_5977_2290_SSHA'],
        model: 'SS9007.2.0',
        vendor: 'Saturn South',
        description: 'Din rail energy meter',
        supports: 'volts, Amps, Watts, power factor, temp',
        fromZigbee: [],
        toZigbee: [],
        meta: {configureKey: 2},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(10);
            await bind(endpoint, coordinatorEndpoint, ['seMetering']);
            await configureReporting.onOff(endpoint);
            const options = {manufacturerCode: 4406, disableDefaultResponse: false};
            await endpoint.write('seMetering', {0x1005: {value: 0x063e, type: 25}}, options);
            await configureReporting.onOff(endpoint);
            await endpoint.configureReporting('seMetering', [{
                attribute: {ID: 0x2000, type: 0x29},
                minimumReportInterval: 1,
                maximumReportInterval: repInterval.MINUTES_5,
                reportableChange: 1,
            }], options);
```
2.
as the reported values are non standard, i need to configure 'fromzigbee.js' to understand what value is what. again, attempting to steal from the meazon, but no success, (probably because bind is fail)

If there are no 'configuration' options, the meter will join the network, but it will sit there any do nothing, and report nothing. 

DB entry is also in paste bin. 
i have sniffed the join procedure and data from the meter to brands hub, (attached .pcapng )
[join-bind.pcapng.txt](https://github.com/Koenkk/zigbee2mqtt/files/4734878/join-bind.pcapng.txt)

Attributes and DB entry. 
https://pastebin.com/x23mPbgg

any pointers appreciated.

Zigbee2mqtt version: zigbee2mqtt version 1.13.1-dev (commit #unknown) testing in windows
Adapter hardware: CC2531
Adapter firmware version: CC2531ZNP-Prod

Koenkk, 2020-06-07:
Could you share the network key of your sniff, otherwise it cannot be decrypted.

l34rn3d, 2020-06-07:
5A:69:67:42:65:65:41:6C:6C:69:61:6E:63:65:30:39
01:03:05:07:09:0B:0D:0F:00:02:04:06:08:0A:0C:0D
44:ee:70:2b:e4:2f:5a:78:6c:35:bb:6e:3a:ba:89:af
sorry about that,  not sure which one it was, so i added all three.

Koenkk, 2020-06-07:
Make sure to use the correct endpoints. E.g. the seMetering bind is done against endpoint 8, not 10 (although some configure reportings are indeed done against endpoint 10).

[image]

l34rn3d, 2020-06-10:
if i try to bind to EP8 i get the following, 

```
[zigbee2mqtt:error 2020-06-10 21:35:43: Failed to configure '0x001bc502b0102aa3', attempt 1 (TypeError: Cannot read property 'bind' of undefined
    at bind (C:\dev\zigbee2mqtt-dev\node_modules\zigbee-herdsman-converters\devices.js:40:24)
    at Object.configure (C:\dev\zigbee2mqtt-dev\node_modules\zigbee-herdsman-converters\devices.js:10551:19)
    at Configure.configure (C:\dev\zigbee2mqtt-dev\lib\extension\configure.js:104:45)
    at Configure.onZigbeeStarted (C:\dev\zigbee2mqtt-dev\lib\extension\configure.js:72:28)
    at Controller.callExtensionMethod (C:\dev\zigbee2mqtt-dev\lib\controller.js:352:44))]
```

Binding to EP 10 results in a timeout or 
```
zigbee2mqtt:error 2020-06-10 21:38:25: Failed to configure '0x001bc502b0102aa3', attempt 1 (Error: Bind 0x001bc502b0102aa3/10 seMetering from '0x00124b0018e27cc9/1' failed (Error: AREQ - ZDO - bindRsp after 10000ms)
    at Endpoint.<anonymous> (C:\dev\zigbee2mqtt-dev\node_modules\zigbee-herdsman\dist\controller\model\endpoint.js:246:23)
    at Generator.throw (<anonymous>)
    at rejected (C:\dev\zigbee2mqtt-dev\node_modules\zigbee-herdsman\dist\controller\model\endpoint.js:6:65))
```
or

```
zigbee2mqtt:error 2020-06-10 21:57:34: Failed to configure '0x001bc502b0102aa3', attempt 1 (Error: Write 0x001bc502b0102aa3/10 seMetering({"57610":{"value":1000,"type":21}}, {"timeout":10000,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4278,"transactionSequenceNumber":null}) failed (Error: Write for 'BUFFER' not available)
    at Endpoint.<anonymous> (C:\dev\zigbee2mqtt-dev\node_modules\zigbee-herdsman\dist\controller\model\endpoint.js:159:23)
    at Generator.throw (<anonymous>)
    at rejected (C:\dev\zigbee2mqtt-dev\node_modules\zigbee-herdsman\dist\controller\model\endpoint.js:6:65))
```
is there anything obvious i am missing?

Koenkk, 2020-06-11:
I was wrong, it indeed should be endpoint 10. Maybe the configure reporting has to be done before the bind? Try to make the sequence exactly the same as the sniff (and verify this by sniffing the zigbee2mqtt traffic).

l34rn3d, 2020-06-28:
is there a explanation of the different binding options available in 'devices.JS'.  IE, 'configurekey (1,2,3)' and what is meant by "type:21".
are these integer types, or data types?

Im also assuming as the sniffed data lists all the end points for binding, i will have to replicate that 'devices.js'. is there a 'neater' way of doing that, then to list all the attributes individually?

Koenkk, 2020-06-29:
- `configureKey`: https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L6
- "type:21". > this is the ZCL data type. list of all data types can be found here: https://github.com/Koenkk/zigbee-herdsman/blob/master/src/zcl/definition/dataType.ts
- Im also assuming as the sniffed data lists all the end points for binding > all endpoints are already known after pairing, 
- then to list all the attributes individually > what do you mean with this?

l34rn3d, 2020-07-30:
I have acquired a second CC stick to sniff while trying to join.
The first thing i notice is Z2MQTT is trying to set destination endpoint to 1 rather then EP 8. 
I see this is mentioned in the initial capture with the configure reporting message from coordinator to device. Does this seem correct? 

[image]
(Left cap is Z2MQTT, Right is device HUB)

then to list all the attributes individually > what do you mean with this?

i was meaning that i will need to list all attributes that i want reported in the configuration of device. (as it uses non standard attributes in the x702 cluster)

Koenkk, 2020-07-31:
Can you try with:

```js
{
    zigbeeModel: ['SS9007.2.0_5977_2290_SSHA'],
    model: 'SS9007.2.0',
    vendor: 'Saturn South',
    description: 'Din rail energy meter',
    supports: 'volts, Amps, Watts, power factor, temp',
    fromZigbee: [fz.saturnsouth_meter],
    toZigbee: [],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(8) || device.createEndpoint(8);
        await bind(endpoint, coordinatorEndpoint, ['seMetering']);
        const options = {manufacturerCode: 4278, disableDefaultResponse: true};
        await endpoint.write('seMetering', {0x0702: {value: 0xe10a, type: 21}}, options);
        await endpoint.configureReporting('seMetering', [{
            attribute: {ID: 0xe10a, type: 0x21},
            minimumReportInterval: 1,
            maximumReportInterval: repInterval.MINUTES_5,
            reportableChange: 1,
        }], options);
    },
},	
```

l34rn3d, 2020-07-31:
https://pastebin.com/1dQptKfP
It made quite a long list of errors, previously these have not been generated. 
While trying to rebind once the DB entry has been made i get the following as a zigbee message. 
[image]
[image]

closed 2020-09-08 by github-actions[bot]