Koenkk/zigbee2mqtt #3671: Two existing devices no-longer working

Johnyb62, 2020-06-05:
# Bug Report
System was running well until recent events, now kitchen the two IKEA TRADFRI drivers nolonger respond (ie switch on/off).  Errors in the logs attached

## What happened
Updated Ikea Remote firmware (afterwards don't think it controlled the kitchen lights, but not sure if responsive in hassio).  This remote was only linked to control the kitchen lights
Updated firmware on CC2531 from  CC2531_20190425 to CC2531_DEFAULT_20190608.
Updated Hassio OS from 3.13 to 4.8, wouldn't re-start (no web access or shares).
Power cycled few times eventually booted & had reverted to 3.13 kitchen MQTT devices were not responding (although think I tried kitchen first as closest so assumed others werent)
Restored 1/6/20 backup, still kitchen MQTT devices not working.
Tried writing a clean image on new SDcard and restoring earler image, still kitchen devices not working

## What did you expect to happen
Was expecting the devices to continue to operate after hassio update & Ikea remote firmware update.
Log files: Restarted zigbee2mqtt, attempted to switch on kitchen lights (for failed log entry) and for comparison switch on a working light
https://pastebin.com/izvgeNyt

## Debug Info
Zigbee2mqtt version: 1.13.1
Adapter hardware: CC2531
Adapter firmware version: 20190608

Koenkk, 2020-06-07:
Did you try to remove and repair the drivers? https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove

Johnyb62, 2020-06-15:
Thanks for your suggestion, removing & re-adding solved the issue.  Was curious to understand why they stopped working.   Updating the remote firmware surely wouldnt affect this. Anyway problem solved.

closed 2020-06-15 by Johnyb62