Koenkk/zigbee2mqtt #3672: zigbee2mqtt fails with timer message and does not recover until restarted with USB device unplugged and plugged

jum, 2020-06-05:
# Bug Report
## What happened

I have a small network with 6 Lupus temperature sensors and zigbee2mqtt works reliable for me. Just today I had the issue where it stopped with a timer promise error message and I was unable to recover it. I had to stop the server, unplug the USB port and plug in again and start the server again.

## What did you expect to happen

Should have been able to recover somehow?

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2mqtt version:

 jum@anubis  /usr/local/zigbee2mqtt   master ●  git log -1
commit 2cb234c93473c245d9530f08f1e7a6f6b3495f56 (HEAD -> master, origin/master, origin/HEAD)
Author: Koen Kanters <koenkanters94@gmail.com>
Date:   Sat May 30 16:01:47 2020 +0200

    Update dependencies. #3616
[zb.log](https://github.com/Koenkk/zigbee2mqtt/files/4737017/zb.log)

Adapter hardware: CC2531, CC2530, CC26X2R1 or CC1352P-2

CC2531

Adapter firmware version:

I have no burner yet, I do not know. Bought at IT STUFF Sven Hackstein via Amazon.

Koenkk, 2020-06-07:
The debug logging shows little information. Try to catch the herdsman debug logging before it starts failing.

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

jum, 2020-06-07:
I have now turned on debugging for herdsman, and will wait for it to happen again. But I found the following log entries when the activity stopped:

`
info  2020-06-05 13:17:05: MQTT publish: topic 'zigbee2mqtt/thermometer/Esstisch', payload '{"temperature":24.38,"linkquality":0,"humidity":33.11,"battery":63,"voltage":2700}'
info  2020-06-05 13:17:10: MQTT publish: topic 'zigbee2mqtt/thermometer/Esstisch', payload '{"temperature":24.42,"linkquality":0,"humidity":33.21,"battery":63,"voltage":2700}'
info  2020-06-05 17:12:53: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'
info  2020-06-05 17:12:53: Disconnecting from MQTT server
error 2020-06-05 17:12:59: Failed to stop zigbee
`

And upon attempting to restart:

`info  2020-06-05 17:14:28: Logging to console and directory: '/usr/local/zigbee2mqtt/data/log/2020-06-05.17-14-28' filename: log.txt
info  2020-06-05 17:14:28: Starting zigbee2mqtt version 1.13.1 (commit #2cb234c)
info  2020-06-05 17:14:28: Starting zigbee-herdsman...
error 2020-06-05 17:15:56: Failed to call 'OnEvent' 'stop' (TypeError: Cannot read property 'getEntries' of null
    at Function.loadFromDatabaseIfNecessary (/usr/local/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/device.js:200:55)
    at Function.all (/usr/local/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/device.js:220:16)
    at Controller.getDevices (/usr/local/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/controller.js:223:31)
    at Zigbee.getClients (/usr/local/zigbee2mqtt/lib/zigbee.js:125:30)
    at OnEvent.stop (/usr/local/zigbee2mqtt/lib/extension/onEvent.js:22:42)
    at Controller.callExtensionMethod (/usr/local/zigbee2mqtt/lib/controller.js:350:44)
    at Controller.stop (/usr/local/zigbee2mqtt/lib/controller.js:168:20)
    at process.handleQuit (/usr/local/zigbee2mqtt/index.js:21:20)
    at process.emit (events.js:315:20))
error 2020-06-05 17:15:56: Not connected to MQTT server!
error 2020-06-05 17:15:56: Cannot send message: topic: 'zigbee2mqtt/bridge/state', payload: 'offline
info  2020-06-05 17:15:56: Disconnecting from MQTT server
error 2020-06-05 17:15:59: Error while starting zigbee-herdsman
`

And then after unplugging the USB dongle:

`info  2020-06-05 17:17:07: Logging to console and directory: '/usr/local/zigbee2mqtt/data/log/2020-06-05.17-17-07' filename: log.txt
info  2020-06-05 17:17:07: Starting zigbee2mqtt version 1.13.1 (commit #2cb234c)
info  2020-06-05 17:17:07: Starting zigbee-herdsman...
info  2020-06-05 17:17:09: zigbee-herdsman started
info  2020-06-05 17:17:09: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'
info  2020-06-05 17:17:09: Currently 6 devices are joined:
info  2020-06-05 17:17:09: thermometer/Bett (0x086bd7fffef1be08): TT001ZAV20 - TUYATEC Temperature & humidity sensor (EndDevice)
info  2020-06-05 17:17:09: thermometer/Esstisch (0x086bd7fffef1bdc7): TT001ZAV20 - TUYATEC Temperature & humidity sensor (EndDevice)
info  2020-06-05 17:17:10: thermometer/Couch (0x14b457fffe2ec93d): TT001ZAV20 - TUYATEC Temperature & humidity sensor (EndDevice)
info  2020-06-05 17:17:10: thermometer/Schreibtisch (0x14b457fffe1561d2): TT001ZAV20 - TUYATEC Temperature & humidity sensor (EndDevice)
info  2020-06-05 17:17:10: thermometer/Bad (0x086bd7fffef1c290): TT001ZAV20 - TUYATEC Temperature & humidity sensor (EndDevice)
info  2020-06-05 17:17:10: thermometer/Gaestezimmer (0x14b457fffe2e11e2): TT001ZAV20 - TUYATEC Temperature & humidity sensor (EndDevice)
warn  2020-06-05 17:17:10: `permit_join` set to  `true` in configuration.yaml.
warn  2020-06-05 17:17:10: Allowing new devices to join.
warn  2020-06-05 17:17:10: Set `permit_join` to `false` once you joined all devices.
info  2020-06-05 17:17:10: Zigbee: allowing new devices to join.
info  2020-06-05 17:17:10: Connecting to MQTT server at mqtt://localhost
info  2020-06-05 17:17:10: Connected to MQTT server
info  2020-06-05 17:17:10: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
info  2020-06-05 17:17:10: MQTT publish: topic 'zigbee2mqtt/thermometer/Bett', payload '{"temperature":23,"linkquality":97,"humidity":33.43,"battery":54,"voltage":2700}'
`

closed 2020-07-26 by stale[bot]