Koenkk/zigbee2mqtt #3673: New device support: Meazon 1phase Meter

DatafoxChris, 2020-06-05:
This device is listed in the supported devices page:
https://www.zigbee2mqtt.io/devices/MEAZON_DINRAIL.html

But it seems to be unsupported when i pair it:
"Received message from unsupported device with Zigbee model '102.106.000540'"

{"ieeeAddr":"0x00124b0002cc3bab","type":"Router","networkAddress":9187,"model":"102.106.000540","vendor":"-","description":"-","friendly_name":"0x00124b0002cc3bab","manufacturerID":4406,"manufacturerName":"Meazon S.A","powerSource":"Mains (single phase)","modelID":"102.106.000540","lastSeen":1591390347328}]}

Zigbee2mqtt version: 1.13.1
Adapter hardware: Conbee II

Koenkk, 2020-06-06:
Your model was not listed yet, added it now, should be supported in the latest dev branch in a few hours.

DatafoxChris, 2020-06-06:
Thanks !! 

I also own these additional two: 
102.106.000565
102.106.001148

DatafoxChris, 2020-06-07:
There is an error in the log regarding the Meazon Meter:

Configuring '0x00124b0002cc3bab'
Failed to configure '0x00124b0002cc3bab', attempt 2 (Error: ConfigureReporting 0x00124b0002cc3bab/10 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"timeout":10000,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: Status 'INVALID_VALUE')
at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:329:23)
at Generator.next (<anonymous>)

Koenkk, 2020-06-07:
Can you try changing your `devices.js` to https://gist.github.com/Koenkk/029da75821b1a74c6c1140da11b49c4e  (see https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to get to `devices.js`)

DatafoxChris, 2020-06-07:
Ok thanks, i added the two serialnumbers locally in the devices.js file.
If i download an updated version in the future, do i need to add them again?

DatafoxChris, 2020-06-07:
I will try to convince the manufacturer to start using one ModelID for this device.... not use the serialnumber for this field. That would be great...

DatafoxChris, 2020-06-07:
I switched to dev branch to get support for the Meazon 1phase Meter, and the meter works now. 
But the dev version stops after a few hours, so i reverted to master branch. 

I used devices.js from the dev version in the master version. 

After startup there is one succesful data-packet beeing processed:
 topic 'zigbee2mqtt/Meazon Meter 1', payload '{"linefrequency":49.99,"power":0.4,"voltage":230.51,"current":0,"reactivepower":0,"measureserial":100,"linkquality":49,"state":"OFF","energyconsumed":10,"energyproduced":0,"reactivesummation":0}'

But all following data-packets result in this response:
Received message from unsupported device with Zigbee model '102.106.000540'

Could this be solved easilly, or should i wait for the dev version to become stable ?

DatafoxChris, 2020-06-07:
Ok, downloaded the master again, and edited devices.js from that version to recognize my device serialnumbers. 

Now the Meter works OK with Master branch, the data is beeing processed correctly. The only remaining error (which is not a problem for me) is:

Jun 07 19:56:22 zigbee2mqtt npm[1046]: zigbee2mqtt:info  2020-06-07 19:56:22: Configuring 'Meazon Meter 1'
Jun 07 19:56:23 zigbee2mqtt npm[1046]: zigbee2mqtt:error 2020-06-07 19:56:23: Failed to configure 'Meazon Meter 1', attempt 3 (Error: ConfigureReporting 0x00124b0002cc3bab/10 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"timeout":10000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Status 'INVALID_VALUE')
Jun 07 19:56:23 zigbee2mqtt npm[1046]:     at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:326:23)
Jun 07 19:56:23 zigbee2mqtt npm[1046]:     at Generator.next (<anonymous>)
Jun 07 19:56:23 zigbee2mqtt npm[1046]:     at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:5:58))

Koenkk, 2020-06-08:
Can you also apply the changes from line 91-99 from this file: https://gist.github.com/Koenkk/029da75821b1a74c6c1140da11b49c4e#file-gistfile1-txt-L91-L99 ? Does that fix this problem?

DatafoxChris, 2020-06-08:
Yes, that fixes the error:

Jun 08 14:55:59 zigbee2mqtt npm[1995]: zigbee2mqtt:info  2020-06-08 14:55:59: MQTT publish: topic 'zigbee2mqtt/Meazon Meter 1', payload '{"linefrequency":49.99,"power":0.41,"voltage":229.17,"current":0,"reactivepower":0.04,"measureserial":171,"linkquality":75,"state":"ON","energyconsumed":17,"energyproduced":0,"reactivesummation":0}'
Jun 08 14:56:00 zigbee2mqtt npm[1995]: zigbee2mqtt:info  2020-06-08 14:56:00: Successfully configured 'Meazon Meter 1'
Jun 08 14:56:03 zigbee2mqtt npm[1995]: zigbee2mqtt:info  2020-06-08 14:56:03: MQTT publish: topic 'zigbee2mqtt/Meazon Meter 1', payload '{"linefrequency":49.99,"power":0.41,"voltage":229.17,"current":0,"reactivepower":0.04,"measureserial":171,"linkquality":75,"state":"ON","energyconsumed":17,"energyproduced":0,"reactivesummation":0}'
Jun 08 14:56:08 zigbee2mqtt npm[1995]: zigbee2mqtt:info  2020-06-08 14:56:08: MQTT publish: topic 'zigbee2mqtt/Meazon Meter 1', payload '{"linefrequency":49.99,"power":0.41,"voltage":229.17,"current":0,"reactivepower":0.04,"measureserial":178,"linkquality":75,"state":"ON","energyconsumed":17,"energyproduced":0,"reactivesummation":0}'
Jun 08 14:56:13 zigbee2mqtt npm[1995]: zigbee2mqtt:info  2020-06-08 14:56:13: MQTT publish: topic 'zigbee2mqtt/Meazon Meter 1', payload '{"linefrequency":49.99,"power":0.41,"voltage":229.17,"current":0,"reactivepower":0.04,"measureserial":178,"linkquality":75,"state":"OFF","energyconsumed":17,"energyproduced":0,"reactivesummation":0}'
Jun 08 14:56:13 zigbee2mqtt npm[1995]: zigbee2mqtt:info  2020-06-08 14:56:13: MQTT publish: topic 'zigbee2mqtt/Meazon Meter 1', payload '{"linefrequency":49.99,"power":0.41,"voltage":232.66,"current":0,"reactivepower":0.02,"measureserial":178,"linkquality":75,"state":"OFF","energyconsumed":17,"energyproduced":0,"reactivesummation":0}'
Jun 08 14:56:33 zigbee2mqtt npm[1995]: zigbee2mqtt:info  2020-06-08 14:56:33: MQTT publish: topic 'zigbee2mqtt/Meazon Meter 1', payload '{"linefrequency":49.99,"power":0.41,"voltage":232.66,"current":0,"reactivepower":0.02,"measureserial":178,"linkquality":75,"state":"OFF","energyconsumed":17,"energyproduced":0,"reactivesummation":0}'

Koenkk, 2020-06-08:
Good, can you also try with:

```js
    onOff: async (endpoint) => {
        const payload = [{
            attribute: 'onOff',
            minimumReportInterval: 1,
            maximumReportInterval: repInterval.HOUR,
            reportableChange: 0,
        }];
        await endpoint.configureReporting('genOnOff', payload);
    },
```

And do a reconfigure on this device via https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure , if the log says `Configure succesfully` I will integrate the fix.

DatafoxChris, 2020-06-08:
Reconfigure:

Jun 08 15:12:06 zigbee2mqtt npm[2033]: zigbee2mqtt:info  2020-06-08 15:12:06: Configuring 'Meazon Meter 1'
Jun 08 15:12:08 zigbee2mqtt npm[2033]: zigbee2mqtt:error 2020-06-08 15:12:08: Failed to configure 'Meazon Meter 1', attempt 1 (Error: Write 0x00124b0002cc3bab/10 seMetering({"4101":{"value":1598,"type":25}}, {"timeout":10000,"manufacturerCode":4406,"disableDefaultResponse":false}) failed (Error: Status 'FAILURE')
Jun 08 15:12:08 zigbee2mqtt npm[2033]:     at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:158:23)
Jun 08 15:12:08 zigbee2mqtt npm[2033]:     at Generator.next (<anonymous>)
Jun 08 15:12:08 zigbee2mqtt npm[2033]:     at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:5:58))
Jun 08 15:12:08 zigbee2mqtt npm[2033]: zigbee2mqtt:info  2020-06-08 15:12:08: MQTT publish: topic 'zigbee2mqtt/Meazon Meter 1', payload '{"linefrequency":49.99,"power":0.41,"voltage":232.15,"current":0,"reactivepower":-0.02,"measureserial":179,"linkquality":75,"state":"OFF","energyconsumed":17,"energyproduced":0,"reactivesummation":0}'
Jun 08 15:12:09 zigbee2mqtt npm[2033]: zigbee2mqtt:info  2020-06-08 15:12:09: MQTT publish: topic 'zigbee2mqtt/Meazon Meter 1', payload '{"linefrequency":49.99,"power":0.41,"voltage":232.15,"current":0,"reactivepower":-0.02,"measureserial":179,"linkquality":75,"state":"OFF","energyconsumed":17,"energyproduced":0,"reactivesummation":0}'

DatafoxChris, 2020-06-08:
Anything i can do to help ?

DatafoxChris, 2020-06-09:
I found some details on the onOff group in the manual, added a screenshot to this post. 

[image]

Koenkk, 2020-06-09:
Can you try to remove the device (https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove) and repair it, do you get `Succesfully configured` after pairing it?

DatafoxChris, 2020-06-09:
Leaving the network succeeded.

Joining Succeeded:

Jun 09 19:07:36 zigbee2mqtt npm[4262]: zigbee2mqtt:info  2020-06-09 19:07:36: Successfully interviewed '0x00124b0002cc3bab', device has successfully been paired
Jun 09 19:07:36 zigbee2mqtt npm[4262]: zigbee2mqtt:info  2020-06-09 19:07:36: Device '0x00124b0002cc3bab' is supported, identified as: Meazon DinRail 1-phase meter (MEAZON_DINRAIL)
Jun 09 19:07:36 zigbee2mqtt npm[4262]: zigbee2mqtt:info  2020-06-09 19:07:36: Configuring '0x00124b0002cc3bab'
Jun 09 19:07:36 zigbee2mqtt npm[4262]: zigbee2mqtt:info  2020-06-09 19:07:36: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x00124b0002cc3bab","model":"MEAZON_DINRAIL","vendor":"Meazon","description":"DinRail 1-phase meter","supported":true}}'
Jun 09 19:07:37 zigbee2mqtt npm[4262]: zigbee2mqtt:info  2020-06-09 19:07:37: Successfully configured '0x00124b0002cc3bab'

DatafoxChris, 2020-06-09:
One detail is missing now: linefrequency

Before:
{"linefrequency":49.95,"power":0,"voltage":234.49,"current":0,"reactivepower":0.02,"measureserial":3,"linkquality":0,"state":"ON","energyconsumed":0,"energyproduced":0,"reactivesummation":0}

After:
{"state":"OFF","linkquality":109,"energyproduced":0,"reactivesummation":0,"measureserial":290,"power":0.39,"voltage":228.33,"current":0,"reactivepower":0}

DatafoxChris, 2020-06-09:
Reconfiguring the device (topic: zigbee2mqtt/bridge/configure) results in:

Jun 09 19:23:20 zigbee2mqtt npm[4262]: zigbee2mqtt:info  2020-06-09 19:23:20: Configuring 'Meazon Meter 1'
Jun 09 19:23:21 zigbee2mqtt npm[4262]: zigbee2mqtt:error 2020-06-09 19:23:21: Failed to configure 'Meazon Meter 1', attempt 1 (Error: Write 0x00124b0002cc3bab/10 seMetering({"4101":{"value":1598,"type":25}}, {"timeout":10000,"manufacturerCode":4406,"disableDefaultResponse":false}) failed (Error: Status 'FAILURE')
Jun 09 19:23:21 zigbee2mqtt npm[4262]:     at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:158:23)
Jun 09 19:23:21 zigbee2mqtt npm[4262]:     at Generator.next (<anonymous>)
Jun 09 19:23:21 zigbee2mqtt npm[4262]:     at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:5:58))

DatafoxChris, 2020-06-09:
Would you like me to send a device over to you?

[image]

Koenkk, 2020-06-09:
It seems that the configure succeeds now right after pairing, somehow it doesn't allows itself to configure twice (but this doesn't matter as this only has to succeeds ones). I've integrated this fix in the latest dev branch.

About the missing linefrequency, energyconsumed, probably they are reported after some time. They come from a similar message as e.g. `linefrequency` which you do have.

DatafoxChris, 2020-06-09:
Waited for an hour now, but the missing values do not appear. The EnergyConsumed value is the most important value from an energy meter :) It should be there now... right ?

ccorderod, 2020-06-09:
@DatafoxChris you got the reporting error because there are 2 calls to configure reporting on the device.js
You may want to try this:
    {
        zigbeeModel: ['102.106.000235', '102.106.001111', '102.106.000348', '102.106.000256', '102.106.001242'],
        model: 'MEAZON_DINRAIL',
        vendor: 'Meazon',
        description: 'DinRail 1-phase meter',
        supports: 'on/off, power, energy measurement and temperature',
        fromZigbee: [
            fz.genOnOff_cmdOn, fz.genOnOff_cmdOff, fz.on_off,
            fz.meazon_meter,
        ],
        toZigbee: [tz.on_off],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(10);
            await bind(endpoint, coordinatorEndpoint, ['genOnOff', 'seMetering']);
            await configureReporting.onOff(endpoint);
            const options = {manufacturerCode: 4406, disableDefaultResponse: true};
            await endpoint.write('seMetering', {0x1005: {value: 0x7fff, type: 25}}, options);
        },
    },

insert below the onOff call in line 91 (insert below line 99):
    onOffMeazon: async (endpoint) => {
        const payload = [{
            attribute: 'onOff',
            minimumReportInterval: 1,
            maximumReportInterval: 0xfffe,
            reportableChange: 1,
        }];
        await endpoint.configureReporting('genOnOff', payload);
    },

Please note @Koenkk 's recent commit on dev is much more clean.
DO NOT forget to add you models on zigbeeModel array
The device is also able to measure temperature, but I have been unable to set the reporting for this.
Hope it will help.

ccorderod, 2020-06-09:
@DatafoxChris a couple of questions if you don't mind: 
- where did you gor the documentation for the dinrail? 
- where did you buy meazon stuff? I have not been able to buy those in Spain

DatafoxChris, 2020-06-09:
This results in the following code, is this correct ?

[image]

DatafoxChris, 2020-06-09:
Re-pairing the meter results in:

Jun 09 20:48:18 zigbee2mqtt npm[4262]: zigbee2mqtt:info  2020-06-09 20:48:18: Successfully interviewed '0x00124b0002cc3bab', device has successfully been paired
Jun 09 20:48:18 zigbee2mqtt npm[4262]: zigbee2mqtt:info  2020-06-09 20:48:18: Device '0x00124b0002cc3bab' is supported, identified as: Meazon DinRail 1-phase meter (MEAZON_DINRAIL)
Jun 09 20:48:18 zigbee2mqtt npm[4262]: zigbee2mqtt:info  2020-06-09 20:48:18: Configuring '0x00124b0002cc3bab'
Jun 09 20:48:18 zigbee2mqtt npm[4262]: zigbee2mqtt:info  2020-06-09 20:48:18: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x00124b0002cc3bab","model":"MEAZON_DINRAIL","vendor":"Meazon","description":"DinRail 1-phase meter","supported":true}}'
Jun 09 20:48:21 zigbee2mqtt npm[4262]: zigbee2mqtt:info  2020-06-09 20:48:21: MQTT publish: topic 'zigbee2mqtt/0x00124b0002cc3bab', payload '{"state":"ON","linkquality":111}'
Jun 09 20:48:21 zigbee2mqtt npm[4262]: zigbee2mqtt:info  2020-06-09 20:48:21: MQTT publish: topic 'zigbee2mqtt/0x00124b0002cc3bab', payload '{"state":"ON","linkquality":111,"energyconsumed":29,"energyproduced":0,"reactivesummation":0,"measureserial":301}'
Jun 09 20:48:21 zigbee2mqtt npm[4262]: zigbee2mqtt:error 2020-06-09 20:48:21: Failed to configure '0x00124b0002cc3bab', attempt 3 (Error: Write 0x00124b0002cc3bab/10 seMetering({"4101":{"value":1598,"type":25}}, {"timeout":10000,"manufacturerCode":4406,"disableDefaultResponse":false}) failed (Error: Status 'undefined')
Jun 09 20:48:21 zigbee2mqtt npm[4262]:     at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:158:23)
Jun 09 20:48:21 zigbee2mqtt npm[4262]:     at Generator.next (<anonymous>)
Jun 09 20:48:21 zigbee2mqtt npm[4262]:     at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:5:58))
Jun 09 20:48:26 zigbee2mqtt npm[4262]: zigbee2mqtt:info  2020-06-09 20:48:26: MQTT publish: topic 'zigbee2mqtt/0x00124b0002cc3bab', payload '{"state":"ON","linkquality":111,"energyconsumed":29,"energyproduced":0,"reactivesummation":0,"measureserial":301,"linefrequency":49.99,"power":1.18,"voltage":231.26,"current":0.01,"reactivepower":-0.08}'

DatafoxChris, 2020-06-09:
device has not successfully been paired

But i do get all the values now :)
{"state":"ON","linkquality":112,"energyconsumed":29,"energyproduced":0,"reactivesummation":0,"measureserial":301,"linefrequency":49.99,"power":1.21,"voltage":231.33,"current":0.01,"reactivepower":-0.06}

DatafoxChris, 2020-06-09:
Paired a second meter: 

Jun 09 21:09:17 zigbee2mqtt npm[6455]: zigbee2mqtt:info  2020-06-09 21:09:17: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x00124b0002cc3c34"}}'
Jun 09 21:09:17 zigbee2mqtt npm[6455]: zigbee2mqtt:info  2020-06-09 21:09:17: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00124b0002cc3c34"}}'
Jun 09 21:09:20 zigbee2mqtt npm[6455]: zigbee2mqtt:info  2020-06-09 21:09:20: Successfully interviewed '0x00124b0002cc3c34', device has successfully been paired
Jun 09 21:09:20 zigbee2mqtt npm[6455]: zigbee2mqtt:info  2020-06-09 21:09:20: Device '0x00124b0002cc3c34' is supported, identified as: Meazon DinRail 1-phase meter (MEAZON_DINRAIL)
Jun 09 21:09:20 zigbee2mqtt npm[6455]: zigbee2mqtt:info  2020-06-09 21:09:20: Configuring '0x00124b0002cc3c34'
Jun 09 21:09:20 zigbee2mqtt npm[6455]: zigbee2mqtt:info  2020-06-09 21:09:20: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x00124b0002cc3c34","model":"MEAZON_DINRAIL","vendor":"Meazon","description":"DinRail 1-phase meter","supported":true}}'
Jun 09 21:09:21 zigbee2mqtt npm[6455]: zigbee2mqtt:error 2020-06-09 21:09:21: Failed to configure '0x00124b0002cc3c34', attempt 1 (ReferenceError: payload is not defined
Jun 09 21:09:21 zigbee2mqtt npm[6455]:     at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices.js:8092:59)
Jun 09 21:09:21 zigbee2mqtt npm[6455]:     at async Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.js:104:13))
Jun 09 21:09:25 zigbee2mqtt npm[6455]: zigbee2mqtt:info  2020-06-09 21:09:25: MQTT publish: topic 'zigbee2mqtt/0x00124b0002cc3c34', payload '{"state":"ON","linkquality":112}'
Jun 09 21:09:25 zigbee2mqtt npm[6455]: zigbee2mqtt:info  2020-06-09 21:09:25: Configuring '0x00124b0002cc3c34'
Jun 09 21:09:25 zigbee2mqtt npm[6455]: zigbee2mqtt:error 2020-06-09 21:09:25: Failed to configure '0x00124b0002cc3c34', attempt 2 (Error: Write 0x00124b0002cc3c34/10 seMetering({"4101":{"value":32767,"type":25}}, {"timeout":10000,"manufacturerCode":4406,"disableDefaultResponse":true}) failed (Error: Status 'FAILURE')
Jun 09 21:09:25 zigbee2mqtt npm[6455]:     at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:158:23)
Jun 09 21:09:25 zigbee2mqtt npm[6455]:     at Generator.next (<anonymous>)
Jun 09 21:09:25 zigbee2mqtt npm[6455]:     at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:5:58))
Jun 09 21:09:26 zigbee2mqtt npm[6455]: zigbee2mqtt:info  2020-06-09 21:09:26: MQTT publish: topic 'zigbee2mqtt/0x00124b0002cc3c34', payload '{"state":"ON","linkquality":112}'
Jun 09 21:09:26 zigbee2mqtt npm[6455]: zigbee2mqtt:info  2020-06-09 21:09:26: Configuring '0x00124b0002cc3c34'
Jun 09 21:09:27 zigbee2mqtt npm[6455]: zigbee2mqtt:error 2020-06-09 21:09:27: Failed to configure '0x00124b0002cc3c34', attempt 3 (Error: Write 0x00124b0002cc3c34/10 seMetering({"4101":{"value":32767,"type":25}}, {"timeout":10000,"manufacturerCode":4406,"disableDefaultResponse":true}) failed (Error: Status 'FAILURE')
Jun 09 21:09:27 zigbee2mqtt npm[6455]:     at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:158:23)
Jun 09 21:09:27 zigbee2mqtt npm[6455]:     at Generator.next (<anonymous>)
Jun 09 21:09:27 zigbee2mqtt npm[6455]:     at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:5:58))
Jun 09 21:09:28 zigbee2mqtt npm[6455]: zigbee2mqtt:info  2020-06-09 21:09:28: MQTT publish: topic 'zigbee2mqtt/0x00124b0002cc3c34', payload '{"state":"ON","linkquality":112}'
Jun 09 21:09:29 zigbee2mqtt npm[6455]: zigbee2mqtt:info  2020-06-09 21:09:29: MQTT publish: topic 'zigbee2mqtt/0x00124b0002cc3c34', payload '{"state":"ON","linkquality":112,"energyconsumed":0,"energyproduced":0,"reactivesummation":0,"measureserial":12}'
Jun 09 21:09:34 zigbee2mqtt npm[6455]: zigbee2mqtt:info  2020-06-09 21:09:34: MQTT publish: topic 'zigbee2mqtt/0x00124b0002cc3c34', payload '{"state":"ON","linkquality":112,"energyconsumed":0,"energyproduced":0,"reactivesummation":0,"measureserial":12,"linefrequency":0,"power":0.59,"voltage":233.19,"current":0,"reactivepower":0}'

This device now also sends the correct values.

DatafoxChris, 2020-06-11:
@ccorderod Here is some useful information from Meazon on the temperature question:

Unfortunately, there seems to be a misunderstanding in the device's description on the zigbee2mqtt support so let me describe it a little further. As you can see in the screenshot attached, taken from the 1-phase dinrail user-guide, the device includes "Inlet Temperature" attribute, but this attribute is "Read-Only", not reportable. Also, its unit of measure and formatting cannot be configured, they are there to inform user of how he should read the temperature.

( There is a ZigBee command that can discover all the available atrributes in a cluster along with their capabilites -read/write/report- but zigbee2mqtt does not support it yet )

Please have in mind that this temperature is coming from the metrology chip inside the meter and must be handled as "electrical-cabinet" temperature between components. As you may know, if you direct a thermal camera into an electrical cabinet, its temperature can be much higher than the environment's.

Saying this, the temperature of the device can only be polled from your application, not auto-reported.

[image]

closed 2020-07-26 by stale[bot]

jd1900, 2020-09-11:
> @DatafoxChris a couple of questions if you don't mind:
> 
> * where did you gor the documentation for the dinrail?
> * where did you buy meazon stuff? I have not been able to buy those in Spain

Have you found it in Spain? I'm looking for it as well but I cannot find it here in Spain either...
Or any zigbee DIN rail device with on/off capabilities...

DatafoxChris, 2020-09-12:
Sent an email directly to Meazon, and they sent me the documentation and have been very friendly and helpful. I bought the devices directly from them, they sent the parcel directly to my house.