Koenkk/zigbee2mqtt #3676: Tubular motor Zemismart ZM25TQ pairing but unsupported device message

abaksa1, 2020-06-06:
# Bug Report
## What happened
I attempted to support new device. The device is fully operational, but "NO COMPONENTS Maybe this device is not (yet) supported by Zigbee2Mqtt?" is displayed.

## What did you expect to happen
No error message.

## How to reproduce it (minimal and precise)
I added the following to devices.js
```
    {
        zigbeeModel: ['mcdj3aq', 'mcdj3aq\u0000'],
        model: 'mcdj3aq',
        vendor: 'Zemismart',
        description: 'Tubular motor',
        whiteLabel: [
            {vendor: 'Zemismart', model: 'ZM25TQ'},
        ],
        supports: 'open, close, stop, position',
        fromZigbee: [fz.tuya_curtain, fz.ignore_basic_report],
        toZigbee: [tz.tuya_curtain_control, tz.tuya_curtain_options],
    },
```

Example of mqtt message
`"ieeeAddr":"0xec1bbdfffed3dca0","type":"Router","networkAddress":14981,"model":"mcdj3aq\u0000","vendor":"-","description":"-","friendly_name":"0xec1bbdfffed3dca0","manufacturerID":0,"manufacturerName":"_TYST11_wmcdj3aq","powerSource":"Mains (single phase)","modelID":"mcdj3aq\u0000","hardwareVersion":1,"dateCode":"20180727","lastSeen":1591110598414}]}'
`
## Debug Info
Zigbee2mqtt version: 1.13.1
Adapter hardware: CC2531
Adapter firmware version: 20190608

Koenkk, 2020-06-07:
This no components warning comes from Zigbee2mqttAssistant and it happens when the device has not been added to https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/extension/homeassistant.js#L739 I believe.

abaksa1, 2020-06-07:
Thanks, closing

closed 2020-06-07 by abaksa1

Dylantje, 2020-09-26:
Dear all..
I do have his one working.. Only can use 1 of this motor. The second one will not learn. Is there perhaps a option to repair or set the motor into pair modes? 

`homeassistant: false
permit_join: true
log_level: error
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost'
serial:
  port: /dev/ttyACM1
  disable_led: false
  channel: 11
devices:
  '0x00158d0001a9e2ba':
    friendly_name: 'ZBX: Fietshok pir'
  '0x000d6ffffed65967':
    friendly_name: 'ZBI: Carportcam R'
  '0x00158d00010e4dac':
    friendly_name: 'ZBX: WC Pir'
  '0x00158d0001a4b86d':
    friendly_name: 'ZBXA: Woonkamer'
  '0x00158d00044457a5':
    friendly_name: 'ZBX:  33 BVTHL'
  '0x00158d0001b76f68':
    friendly_name: 'ZBX: Schakelaar over'
  '0x00158d000117b3fd':
    friendly_name: 'ZBX: Ryan Pir'
  '0x00158d000130d60b':
    friendly_name: 'ZBX: Dylan Pir'
  '0x00158d0001ddcd53':
    friendly_name: 'ZBXA: Zolder Overl Pir'
  '0x00158d0001a35d4e':
    friendly_name: 'ZBXA: 22'
  '0x00158d0004102c94':
    friendly_name: 'ZBX: 21 Over'
  '0x00158d000213872d':
    friendly_name: 'ZBXA: Vierkant 1'
  '0x00158d00010dcd46':
    friendly_name: 'ZBX: Rond 1'
  '0xec1bbdfffe47c996':
    friendly_name: 'ZB: Ikea repeater 1 [ Ryan ]'
  '0x00158d0001db35fb':
    friendly_name: 'ZBX: Carport Pir'
  '0x0017880100d7c02a':
    friendly_name: 'ZBHW: Garage R'
  '0x00158d0001f29be2':
    friendly_name: 'ZBX: Bijkeuken Pir'
  '0x00158d0001ddccef':
    friendly_name: 'ZBXA: Keuken Pir'
  '0x00158d00018373ad':
    friendly_name: 'ZBXA: Achterdeur'
  '0x00158d0001a64c71':
    friendly_name: 'ZBXA: Bijkeukendeur'
  '0x00158d0001a8d32e':
    friendly_name: 'ZBX: Gang Pir'
  '0x0017880100ed186c':
    friendly_name: 'ZBHW: Gang R'
  '0x0017880100dbcce7':
    friendly_name: 'ZBHW: Zolder R'
  '0x00178801063405a8':
    friendly_name: 'ZBHR1: Nieuw'
  '0x0017880100e56e0e':
    friendly_name: 'ZBHW: Over'
  '0x0017880100e612b6':
    friendly_name: 'ZBHK: Eettafel 1 R'
  '0x0017880100ffc053':
    friendly_name: 'ZBHK: Eettafel 2 R'
  '0x0017880100fedd0f':
    friendly_name: 'ZBHK: Zitkamer R'
  '0x001788010117afe5':
    friendly_name: 'ZBHG: Kamer TV R'
  '0x0017880100e8dd9d':
    friendly_name: 'ZBHKS: Kachel 1 R'
  '0x0017880100e84bd3':
    friendly_name: 'ZBHKS: Kachel 2 R'
  '0x00158d00014dc9a2':
    friendly_name: 'ZBX: Overloop Pir'
  '0x00158d000117b661':
    friendly_name: 'ZBX: Voordeur'
  '0x00158d000121b9c5':
    friendly_name: 'ZBX: Badkamer T&V'
  '0x00158d00015a9ce3':
    friendly_name: 'ZBX: Garage Pir'
  '0x00158d000149a348':
    friendly_name: 'ZBX: Garagedeur'
  '0x00158d00016da147':
    friendly_name: 'ZBX: Pir over'
  '0x34c69a000bec3df7':
    friendly_name: '0x34c69a000bec3df7[ weet niet? ]'
  '0x00158d00016569cf':
    friendly_name: 'ZBX: PIR over'
  '0x00158d0001537631':
    friendly_name: 'ZBX: Zolder Wastafel'
  '0x00124b00210b624c':
    friendly_name: 'Zgeldop: Douche L'
  '0x00124b00213acb58':
    friendly_name: 'Zgeldop: WC'
  '0x00124b00213b124d':
    friendly_name: 'Zgeldop: Meubel rechts'
  '0x00124b0021382f36':
    friendly_name: 'Zgeldop: Bad R'
  '0x00124b002117b04f':
    friendly_name: 'Zgeldop: Douche R'
  '0x00124b00213ac72f':
    friendly_name: 'Zgeldop: Bad M'
  '0x00124b002117c4a7':
    friendly_name: 'Zgeldop: Meubel Links'
  '0xbc33acfffe363dcc':
    friendly_name: Ryan Voorkant
groups:
  '1':
    friendly_name: 'ZB: Lampen'
    devices:
      - 0x0017880100e612b6/11
      - 0x0017880100ffc053/11
      - 0x0017880100e84bd3/11
      - 0x0017880100fedd0f/11
      - 0x0017880100dbcce7/11
      - 0x001788010117afe5/11
      - 0x0017880100ed186c/11
      - 0x0017880100d7c02a/11
      - 0x00124b0021382f36/11
      - 0x00124b00210b624c/11
      - 0x00124b002117b04f/11
      - 0x00124b00213b124d/11
      - 0x00124b00213acb58/11
  '2':
    friendly_name: Rolgordijnen
    devices:
      - 0xbc33acfffe363dcc/1
`
Everthing working fine. only 1 device...
[image]