Koenkk/zigbee2mqtt #3677: Debounce causes delayed MQTT (tested Symfonisk)

mvjt, 2020-06-06:
# Bug Report
## What happened
When using debounce according to the device docs for Symfonisk, MQTT message for rotate_left/right is delayed and reported together with rotate_stop when rotating stops. So if I rotate for 10s, I get rotate_left and rotate_stop after 10s. Further more, if removing the debounce_ignore: action, it does not send rotate_left at all, only rotate_stop.

## What did you expect to happen
I expect rotate_left to be send when rotating starts (Zigbee message arrives immediately when rotating starts.

## How to reproduce it (minimal and precise)
Using this device config 
`  '0x14b457fffe65dd74':
    friendly_name: symfonisk_svart
    debounce: 0.5
    debounce_ignore:
      - action`

## Debug Info
Zigbee2mqtt version: Latest docker container (1.13.1), also tested dev branch with same issue.

Adapter hardware: CC2531
Adapter firmware version: 20190619 (tested both default and source routing)

Logs:
https://hastebin.com/ozawesudit.http

Koenkk, 2020-06-07:
I think the main issue is that the behaviour of `debounce_ignore` is strange. Based on the name you would expect the action to be immediately published (no debounce is applied for these). Do you agree?

mvjt, 2020-06-07:
There are several things I find strange.

* In general debounce should avoid multiple messages when an input bounces. This is what the debounce method (JS) does. You need to specify whether it should report the message on trailing edge (default) or leading edge. In my view, it should be on leading edge in most cases. As when I´m continously rotating the Symfonisk remote, I expect things to happen immediately and not when i STOP rotating.
* As I have understood debounce_ignore, it will debounce as long as the message is the same. And during the first 10s when I´m rotating the frame is the same so I would not expect debounce_ignore to kick in until I stop rotating after 10s
* When removing debounce_ignore, I do not get rotating_left at all. This must be a bug and not related to debounce_ignore as in this case I´m not using it.
* When reading the Z2M docs on debounce, one example is when using a sensor that sends three messages (one for each metric)  and you want to aggregate them into one message after. I would say this is a rare use case for debounce, but still useful. I´m thinking maybe we need an argument to debounce on how it should behave to support both scenarios.

MPM1107, 2020-06-07:
@Koenkk I'll jump in here because I helped diagnose this in the discord. Debounce ignore is functioning as intended and completely ignoring debounce for those actions would make the debounce totally useless.

The issue is that the remote doesn't send a rotate_start action, it starts off with rotate_left(or right), then continues spamming that message until you stop rotating, at that point it sends a rotate_stop action. Debounce ignore sees the rotate_left and until it sees the action change to rotate_stop, it doesn't send any messages. The desired behavior would be to let the very first rotate_left message be published, then debounce all of the rest, until rotate_stop.

We have figured out that this can be done quite easily, there is a boolean parameter in the debounce function from the  [library we use](https://www.npmjs.com/package/debounce#debouncefn-wait--immediate--false-) called immediate. Changing that to true makes it let the **first** message pass, then debounce the rest, instead of debouncing all messages and then posting the **last** one after the timer runs out.

We don't want to hardcode this parameter to true, because for devices like sensors, you want the latest (last) message to come through. But for devices like this volume controller, you want the first message to come through. So the proposed change would be to add a new config option called debounce_immediate, debounce_risingedge, debounce_fallingedge or something along those lines.

mvjt, 2020-06-07:
> @Koenkk I'll jump in here because I helped diagnose this in the discord. Debounce ignore is functioning as intended and completely ignoring debounce for those actions would make the debounce totally [...]

Also note that adding immediate is not the only change I had to do to get this working. See my code above. But it was one of the changes...Thanks for your help @MPM1107

Koenkk, 2020-06-09:
@mvjt would you mind creating a PR, we can discuss the detail further there.

mvjt, 2020-06-09:
@Koenkk  sure, never done that before so it could be interesting :)

mvjt, 2020-06-20:
This has been fixed in the dev-version...

closed 2020-06-20 by mvjt