Koenkk/zigbee2mqtt #3678: SwannONE Motion Detector - SWO-MOS1PA - Only partial functionality

MattEng0, 2020-06-06:
Recently bought one of these motion detectors and have managed to get the motion functionality sorted with no issues. The details of the device states that it has the ability to detect temperature as well and I have been unsuccessful in pulling the information from the device to get it working. 

Note I'm running on the addon for Home Assistant 

I have used the following code:

```js
	{
		zigbeeModel: ['SWO-MOS1PA'], 
		model: 'SWO-MOS1PA',
		vendor: 'Swann', 
		description: 'Motion and temperature sensor', 
		supports: 'occupancy and temperature',
		fromZigbee: [fz.ias_occupancy_alarm_1, fz.temperature
        ],
        toZigbee: [],
        meta: {configureKey: 1},
		configure: async (device, coordinatorEndpoint) => {
			const endpoint = device.getEndpoint(1);
			await bind(endpoint, coordinatorEndpoint, ['msTemperatureMeasurement']);
			await configureReporting.temperature(endpoint);
		},
	},
```
And I get the following errors:

Upon initial reboot
```
zigbee2mqtt:error 2020-06-06 12:32:09: Failed to configure 'Swann Sensor', attempt 1 (Error: Bind 0x00124b00063c5146/1 msTemperatureMeasurement from '0x00124b00014d96ff/1' failed (Error: AREQ - ZDO - bindRsp after 10000ms)
    at Endpoint.<anonymous> (/zigbee2mqtt-1.13.1/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:244:23)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.13.1/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))
```
And when making sure the device is active after first fail
```
zigbee2mqtt:error 2020-06-06 12:32:52: Failed to configure 'Swann Sensor', attempt 2 (Error: ConfigureReporting 0x00124b00063c5146/1 msTemperatureMeasurement([{"attribute":"measuredValue","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":25}], {"timeout":10000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Timeout - 61243 - 1 - 3 - 1026 - 7 after 10000ms)
    at Endpoint.<anonymous> (/zigbee2mqtt-1.13.1/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:326:23)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.13.1/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))
```
There are no logs giving any indication of a temperature being pushed to the bridge and the motion detection gives the following two debug logs:
```
zigbee2mqtt:debug 2020-06-06 12:35:49: Received Zigbee message from 'Swann Sensor', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"zonestatus":48,"extendedstatus":0}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-06-06 12:35:49: MQTT publish: topic 'zigbee2mqtt/Swann Sensor', payload '{"occupancy":false,"tamper":false,"battery_low":false,"linkquality":39}'
```
And
```
zigbee2mqtt:debug 2020-06-06 12:37:38: Received Zigbee message from 'Swann Sensor', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"zonestatus":49,"extendedstatus":0}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-06-06 12:37:38: MQTT publish: topic 'zigbee2mqtt/Swann Sensor', payload '{"occupancy":true,"tamper":false,"battery_low":false,"linkquality":36}'
```
This is working fine for motion detection.

There are two other issues opened on here relating to the same device which I will link to.

I will continue trying different combinations of the already integrated temperature and motion devices but this is beyond my abilities at the moment.

Thanks for the help.

Koenkk, 2020-06-06:
Probably the temperature is reported via a non-standard way as the device does not supports the temperature measurement cluster (1026). In order to further investigate you probably need to pair it with the original bridge, sniff the traffic and figure out which messages contain the temperature measurements).

MattEng0, 2020-06-06:
OK thanks for the reply.

It's not a crucial feature for me (didn't notice it had it until I had to manually add it to the bridge), I might see if I can get the bridge cheap enough to get it working.

I'll have another shot in the dark with a few settings to see if I hit lucky (I'll try some of the bespoke settings for other sensors). Also thinking of trying a few options using the "toZigbee" incase its a manual request from hub to sensor for temperature.

Apart from that does the debug logging setting give all information reported from the sensor or does it filter it at all?

Koenkk, 2020-06-07:
It depends, in case of an unknown cluster you will only see it in the herdsman debug logging. A better way is to sniff: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

Btw, feel free to already create a pr with the current functionallity.

MattEng0, 2020-07-19:
Closing this issue until I get around to attempting to sniff the traffic properly to implement full functionality.

closed 2020-07-19 by MattEng0

andygriffiths77, 2020-07-22:
> Closing this issue until I get around to attempting to sniff the traffic properly to implement full functionality.

I have quite a few of these sensors and would be great to get the temperature working. I also have an unused Swann Smart Hub, but looks like my ConBee II doesn't work with the Zshark firmware to be able to sniff the traffic. If there is anything I can help with to see how they are working would be glad to pitch in