Koenkk/zigbee2mqtt #3680: Errors after a "clean" install Z2M will not start

Dylantje, 2020-06-06:
Latest version
Odroid N2 with Domoticz latetst beta.

After a brute force some problems with log and therefore with memory. Unstable system and now a bit working again.

But I can no longer get the z2m up and running.

Backup made and everything removed.
Put back what is needed ...
But on a reboot:
`
> zigbee2mqtt@1.13.1 start /opt/zigbee2mqtt
> node index.js

zigbee2mqtt:info  2020-06-06 14:47:07: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-06-06.14-47-07' filename: log.txt
zigbee2mqtt:info  2020-06-06 14:47:07: Starting zigbee2mqtt version 1.13.1 (commit #2cb234c)
zigbee2mqtt:info  2020-06-06 14:47:07: Starting zigbee-herdsman...
zigbee2mqtt:error 2020-06-06 14:47:07: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-06-06 14:47:07: Failed to start zigbee
zigbee2mqtt:error 2020-06-06 14:47:07: Exiting...
zigbee2mqtt:error 2020-06-06 14:47:07: SyntaxError: Unexpected token  in JSON at position 0
    at JSON.parse (<anonymous>)
    at Function.open (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/database.js:17:35)
    at Controller.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/controller.js:91:48)
    at Generator.next (<anonymous>)
    at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/controller.js:5:58)
    at processTicksAndRejections (internal/process/task_queues.js:97:5)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.13.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.13.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-06-06T12_47_07_649Z-debug.log
pi@odroidn2:/opt/zigbee2mqtt$ ^C
pi@odroidn2:/opt/zigbee2mqtt$
`

The log:
[https://hastebin.com/mufahuwiha.rb](url)

Hope you can help me.

Koenkk, 2020-06-07:
I think the database is corrupted, try removing data/database.db

Dylantje, 2020-06-07:
When i remove the DB 
All the devices are gone?

Koenkk, 2020-06-07:
Yes, you can also post it here, maybe it's possible to fix it manually.

Dylantje, 2020-06-07:
mmm try to open mwith sql lite expert.. still get an error..
[database.zip](https://github.com/Koenkk/zigbee2mqtt/files/4741997/database.zip)

Koenkk, 2020-06-07:
It is a text file, so should be able to open in any text editor. However it appears to be corrupted. In case you don't have a backup the only way forward is to repair all devices.

Dylantje, 2020-06-07:
ok. text is opening without any problem..
with notepad i see the fil.
What are my option to check if corrupt?

Dylantje, 2020-06-07:
i see differents between the fist db and this one.
[database 1-6-2020.zip](https://github.com/Koenkk/zigbee2mqtt/files/4742022/database.1-6-2020.zip)

What can i remove in this file? i see a lot data that is not a device?
id 44 as example... looks to my not a device?

Koenkk, 2020-06-07:
Can you try restoring just the `database.db` file from that zip? Looks fine (although there are a lot of duplicated which you can remove without problems, not sure how they ended up there).

Dylantje, 2020-06-07:
Thanks Koenk!!! A live again....

I try to play with the db to get the doubles out of this db...

Offtopic only ...:-)
How can i remove a device?
Try a lot of things..
Remove from the db  and replace that db a option?

Dylantje, 2020-06-08:
hi strange other problem..
when there is a reboot, the system runs for a lot of time... everything looks working fine.
Only then z2m is working not any more, and i see this:
[image]
Perhaps an idea?

Koenkk, 2020-06-09:
For removing see https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove

Can you provide the debug logging when executing the network scan?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

Dylantje, 2020-06-11:
@Koenkk 

Look that my zram is getting full because z2m logging.
There is a logging into the syslog .. when that log is full, my odroid is instable...

I did follow the wiki.. So did not change some settings.

So where it is going wrong...
[syslog.zip](https://github.com/Koenkk/zigbee2mqtt/files/4765761/syslog.zip)

closed 2020-07-26 by stale[bot]