Koenkk/zigbee2mqtt #3681: [request] Publish failed LQI to a topic

vide, 2020-06-06:
This is not a bug report but rather a feature request: would it be feasible to publish the `Failed to execute LQI for 'DEVICE'` error messages to a specific (fixed probably) topic? This way I could create alerts when a device disappears from the network, usually because someone switched it off.

Or is there already a better method to do this that I'm not aware of? (which does not imply parsing the logs and emit a signal, thing that I may end up doing as a workaround?)

Thank you very much, and thanks for the awesome zigbee2mqtt!

dh-harald, 2020-06-06:
Hi,

I've got this error message on network map rescan:
```zigbee2mqtt:debug 2020-06-06 19:45:36: Received MQTT message on 'zigbee2mqtt/bridge/networkmap' with data 'graphviz'
zigbee2mqtt:info  2020-06-06 19:45:36: Starting network scan (includeRoutes 'false')
zigbee2mqtt:debug 2020-06-06 19:45:36: LQI succeeded for 'Coordinator'
zigbee2mqtt:debug 2020-06-06 19:45:36: LQI succeeded for 'light_kids'
zigbee2mqtt:debug 2020-06-06 19:45:36: LQI succeeded for 'repeater_kitchen'
zigbee2mqtt:debug 2020-06-06 19:45:37: LQI succeeded for 'light_corridor'
zigbee2mqtt:debug 2020-06-06 19:45:37: LQI succeeded for 'switch_computer_kids'
zigbee2mqtt:debug 2020-06-06 19:45:37: LQI succeeded for 'light_bedside1'
zigbee2mqtt:debug 2020-06-06 19:45:37: LQI succeeded for 'light_bedside2'
zigbee2mqtt:error 2020-06-06 19:45:37: Failed to execute LQI for 'light_corridor_floor'
zigbee2mqtt:info  2020-06-06 19:45:37: Network scan finished
zigbee2mqtt:error 2020-06-06 19:45:37: Failed to call 'NetworkMap' 'onMQTTMessage' (TypeError: Cannot read property 'map' of undefined
    at fingerprintMatch (/app/node_modules/zigbee-herdsman-converters/index.js:75:42)
    at Object.findByDevice (/app/node_modules/zigbee-herdsman-converters/index.js:50:21)
    at /app/lib/extension/networkMap.js:74:61
    at Array.forEach (<anonymous>)
    at NetworkMap.graphviz (/app/lib/extension/networkMap.js:60:24)
    at NetworkMap.onMQTTMessage (/app/lib/extension/networkMap.js:44:65)
    at async Controller.callExtensionMethod (/app/lib/controller.js:350:21))
```
`light_corridor_floor` is a `Gledopto Zigbee LED controller RGBW (1 ID) (GL-C-007-1ID)`

sjorge, 2020-06-06:
https://www.zigbee2mqtt.io/information/availability.html should do what you want.

vide, 2020-06-06:
I totally missed this when reading the docs. Thank you very much 🙌 and sorry for the noise

closed 2020-06-06 by vide