Koenkk/zigbee2mqtt #3684: Bulbs Flash randomly

netadmindave, 2020-06-06:
i have had many, maybe all, of my cree bulbs randomly flash on and immediately turn off. this happens day or night. in the logs there is an announcement as if it is a new device. this has been happening for some time so it should not be related to a recent update. this has not happened, as far as i am aware, when the bulbs were connected to smart things or with the zigbee function built in to Hassio.

log entry that accompanies the flash. it is the same for any bulb. 
info  2020-06-05 19:31:29: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"Kitchen 1"}}'

current setup:
CC1352P-2 coordinator 
2x CC2531 with external antennas router

mix of bulbs purchased at different times 
2x phillips hue
6x Sylvania LIGHTIFY LED
1x GE led
4x sengled RGB
15x Cree connected

and a few other devices
buttons, dimmer remotes, switches, door locks.

I was having the issue under the production and edge installs

configuration
data_path: /config/zigbee2mqtt
devices: devices.yaml
groups: groups.yaml
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://core-mosquitto'
  user: xxx
  password: xxx
serial:
  port: /dev/ttyACM0
advanced:
  pan_id: xxx
  channel: 11
  network_key:
    -xxx
  availability_blacklist: []
ban: []
whitelist: []
queue: {}
experimental: {}
socat:
  enabled: false
  master: 'pty,raw,echo=0,link=/dev/ttyZ2M,mode=777'
  slave: 'tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5'
  restartdelay: 1
  initialdelay: 1
  options: '-d -d'
  log: false

netadmindave, 2020-06-07:
updating to newest coordinator firmware, will report back.

netadmindave, 2020-06-07:
no change

Koenkk, 2020-06-08:
Can you sniff the traffic at the moment this happens? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

khaeljy, 2020-06-12:
I have the same issue with my Ikea bulbs.

netadmindave, 2020-06-12:
i will try to setup the sniffing this weekend

netadmindave, 2020-06-12:
seems similar to https://github.com/Koenkk/zigbee2mqtt/issues/3662#issue-630764753

to4ko, 2020-07-13:
any success on this one? my issues with plugs are still in place (diff versions of z2m, diff stick firmware, diff channel...all with no success

as i can see - i have options like
1. diff stick (cc2652 or 1352) - maybe lack of ram on stick causing devices to disconnect
2, diff pan_id or ex_pan_id....not sure if this maters
3, switch back to xiaomi gateways or deconz

closed 2020-08-21 by github-actions[bot]