Koenkk/zigbee2mqtt #3687: Sending multiple/duplicate messages in groups of 6

alan-rpi, 2020-06-07:
First of all, thank you so much for creating zigbee2mqtt :)  It was this software that made me choose zigbee.

New installation
Raspberry Pi 4 running Buster
Running Docker 
Downloaded latest zigbee2mqtt image 
Adapter USB CC2531 Sniffer [(https://www.ebay.co.uk/itm/CC2531-CC2540-Sniffer-Protocol-Analyzer-Dongle-BTool-USB-Downloader-for-Zigbee/372750849759)](ebay link)
Adapter firmware version: 0100 CC2531 CC Debugger 05CC 0041
Adapter Sniffer flashed with zigbee2mqtt firmware version 1.13.1   
zigbee2mqtt:info  2020-06-07 19:18:51: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}',

Paired a new Sonoff BASICZBR3 situated 1m from Sniffer - 1st Zigbee device
Pressed the BASICZBR3 push button to turn on  and a few seconds later turned off

**Problem**
Generated 6 On messages:
`zigbee2mqtt:info  2020-06-07 20:16:47: MQTT publish: topic 'zigbee/0x00124b001e728bde', payload '{"state":"ON","linkquality":49}',`
all the same.

zigbee2mqtt was also receiving multiple messages:
`zigbee2mqtt:debug 2020-06-07 20:16:47: Received Zigbee message from '0x00124b001e728bde', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0,`

Generate multiple OFF messages:
`zigbee2mqtt:info  2020-06-07 20:16:51: MQTT publish: topic 'zigbee/0x00124b001e728bde', payload '{"state":"OFF","linkquality":78}'`
all same except linkquality varies: 78, 49, 52, 39, 63 etc 
Off messages continue ad-infinitum.

Long term problem: this will generate large log files in zigbee2mqtt, mosquitto and domoticz, keep the computer from sleeping and possibly make it overheat, make programming rules cumbersome. 

Repeated on/off again.
Again 6 on messages and then continuous off messages:
x6 at 20:52:55 
x6 at 20:56:37
x6 at 21:01:37
x6 at 21:06:37
etc

Log
https://pastebin.com/9RDPGtCp

Sorry if this is a problem with the Sonoff. I'm new to zigbee so I'm unsure if it is the Sonoff sending the multiple messages under it's own volition or zigbee2mqtt that is polling the Sonoff multiple times asking for a status update.  

PS I found issues re multiple messages dated 2018 but they were marked Stale without a solution, so  hence posting this new issue.

Koenkk, 2020-06-08:
Can you try changing the entry in `devices.js` to:

```js
    {
        zigbeeModel: ['BASICZBR3'],
        model: 'BASICZBR3',
        vendor: 'SONOFF',
        description: 'Zigbee smart switch',
        supports: 'on/off',
        fromZigbee: [fz.SA003_on_off],
        toZigbee: [tz.on_off],
    },
```

This should prevent the MQTT publish (not the receive part as we cannot fix that). See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html how to get to `devices.js`.

alan-rpi, 2020-06-08:
Many thanks for a quick response. I did the following

1. followed the Docker instructions at the above link to add herdsman-converters.
2. as Pi root edited devices.js - only 1 line was different:
` fromZigbee: [fz.SA003_on_off],`
3. I amended my compose file under volumes:
`- /opt/zigbee-herdsman-converters:/app/node_modules/zigbee-herdsman-converters`
4. stopped the container and reran the compose
5. checked the zigbee2mqtt log in Portainer and saw errors. See https://pastebin.com/KASP5t1W
The first error is
`internal/modules/cjs/loader.js:969,
   throw err;,
  ^,
 Error: Cannot find module 'https-proxy-agent',`
No other changes have been made to the system. My other Docker containers are running OK.

As an aside, would it be possible to run zigbee2mqtt in Docker as a user by adding:
` --user 1234:1234`

Thank you
Alan

alan-rpi, 2020-06-09:
Hi
Decided on a fresh install. Removed the container and image and re ran the compose file but without the 
`- /opt/zigbee-herdsman-converters:/app/node_modules/zigbee-herdsman-converters	`

Then ran the 
`cd /opt`
`git clone https://github.com/Koenkk/zigbee-herdsman-converters.git`
`cd zigbee-herdsman-converters`
`npm ci`

but noticed that the npm command said nothing other that list the syntax:
`root@raspi4hms:/opt/zigbee-herdsman-converters# npm ci
Usage: npm <command>
where <command> is one of:
    access, ............................ whoami
npm <cmd> -h     quick help on <cmd>
npm -l           display full usage info
npm help <term>  search for help on <term>
npm help npm     involved overview
Specify configs in the ini-formatted file:
    /root/.npmrc
or on the command line via: npm <command> --key value
Config info can be viewed via: npm help config
npm@3.10.10 /usr/lib/node_modules/npm
`

Did it work?

I also listed the Docker /app/node_modules/zigbee-herdsman-converters -l
`-rw-r--r--    1 root     root          1069 Oct 26  1985 LICENSE`
`-rw-r--r--    1 root     root           782 Oct 26  1985 README.md`
`drwxr-xr-x    3 root     root          4096 May 30 15:06 converters`
`-rw-r--r--    1 root     root        382942 Oct 26  1985 devices.js`
`-rw-r--r--    1 root     root          4738 Oct 26  1985 index.js`
`drwxr-xr-x   17 root     root          4096 May 30 15:07 node_modules`
`-rw-r--r--    1 root     root        241030 Oct 26  1985 npm-shrinkwrap.json`
`drwxr-xr-x    2 root     root          4096 May 30 15:06 ota`
`-rw-r--r--    1 root     root          1221 Oct 26  1985 package.json`

and the Pi /opt/zigbee-herdsman-converters -folder
`-rw-r--r-- 1 root root    577 Jun  8 15:49 azure-pipelines.yaml`
`drwxr-xr-x 3 root root   4096 Jun  8 15:49 converters`
`-rwxr-xr-x 1 root root 401658 Jun  8 15:58 devices.js`
`-rw-r--r-- 1 root root   4738 Jun  8 15:49 index.js`
`-rw-r--r-- 1 root root   1069 Jun  8 15:49 LICENSE`
`-rw-r--r-- 1 root root 216130 Jun  8 15:49 npm-shrinkwrap.json`
`drwxr-xr-x 2 root root   4096 Jun  8 15:49 ota`
`-rw-r--r-- 1 root root    981 Jun  8 15:49 package.json`

Does this help?

Alan

Koenkk, 2020-06-09:
Can you try `npm install` instead of `npm ci`? You should get a `/opt/zigbee-herdsman-converters/devices.js`. 

As an alternative, you could also provide the herdsman debug log when receiving the messages. 

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

alan-rpi, 2020-06-09:
Hi. That was better - just a few warnings.

- I edited the devices.js to change the Sonoff BASICZBR3 to fromZigbee: [fz.SA003_on_off],
- I edited the compose file to insert the volumes statement and reran the compose
- Container zigbee2mqtt starts OK and recognises the BASICZBR3
- I switched the BASICZBR3 on manually and then a little later off

MQTT messages have reduced to 1 every 5 minutes rather than 6. Here's a log summary:

`2020-06-09 19:56:45: Received Zigbee message from 'Immersion Switch', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0`
`2020-06-09 19:56:45: **x6** receiving state off, MQTT sent after the 1st:`
`2020-06-09 19:56:45: MQTT publish: topic 'zigbee/Immersion Switch', payload '{"state":"OFF","linkquality":28}`
`2020-06-09 19:56:58: **x6** receiving state on, MQTT sent after the 1st:`
`2020-06-09 19:56:58: MQTT publish: topic 'zigbee/Immersion Switch', payload '{"state":"ON","linkquality":23}'`
`2020-06-09 19:57:03: **x6** receiving state off, MQTT sent after the 1st:`
`2020-06-09 19:57:03: MQTT publish: topic 'zigbee/Immersion Switch', payload '{"state":"OFF","linkquality":49}'`
`2020-06-09 20:00:23: Saving state to file /app/data/state.json`
`2020-06-09 20:01:45: **x6** receiving state off, MQTT sent after the 1st`:
`2020-06-09 20:01:45: MQTT publish: topic 'zigbee/Immersion Switch', payload '{"state":"OFF","linkquality":44}'`
`2020-06-09 20:05:23: Saving state to file /app/data/state.json`
`2020-06-09 20:06:45: **x6** receiving state off, MQTT sent after the 1st`
`2020-06-09 20:06:45: MQTT publish: topic 'zigbee/Immersion Switch', payload '{"state":"OFF","linkquality":44}'`
`**repeats** at 20:11:45, 20:16:45, 20:21:45 etc`

I guess I could live with this but I do think MQTT messages should not repeat when the state has not changed. It may mean I have to check for a none state change in my Domoticz rules.

Couple of questions:

1. Will you make the `fromZigbee: [fz.SA003_on_off]` change permanent in the devices.js? Seems to me other people with a Sonoff BASICZBR3  will have the same problem. Without a permanent change then I have to remember to update the js file if I rerun the compose.
2. Can I run zigbee2mqtt container as a user/group pi by adding this to the compose?
`--user 1234:1234`

Many thanks
Alan

karlhuther, 2024-01-28:
I have read about this issue in several different posts as Im facing something similar. here are my logs from my mqtt server:

1706483447: Received PUBLISH from mqttjs_21b0b174 (d0, q0, r0, m0, 'zigbee2mqtt/device/loft/light/0x5c0272fffe822626', ... (65 bytes))
1706483447: Sending PUBLISH to mqttjs_21b0b174 (d0, q0, r0, m0, 'zigbee2mqtt/device/loft/light/0x5c0272fffe822626', ... (65 bytes))
1706483447: Received PUBLISH from mqttjs_21b0b174 (d0, q0, r0, m0, 'zigbee2mqtt/bridge/logging', ... (169 bytes))
1706483447: Sending PUBLISH to mqttjs_21b0b174 (d0, q0, r0, m0, 'zigbee2mqtt/bridge/logging', ... (169 bytes))
1706483447: Received PUBLISH from mqttjs_21b0b174 (d0, q0, r0, m0, 'zigbee2mqtt/device/loft/light/0x5c0272fffe822626', ... (50 bytes))
1706483447: Sending PUBLISH to mqttjs_21b0b174 (d0, q0, r0, m0, 'zigbee2mqtt/device/loft/light/0x5c0272fffe822626', ... (50 bytes))
1706483447: Received PUBLISH from mqttjs_21b0b174 (d0, q0, r0, m0, 'zigbee2mqtt/bridge/logging', ... (169 bytes))
1706483447: Sending PUBLISH to mqttjs_21b0b174 (d0, q0, r0, m0, 'zigbee2mqtt/bridge/logging', ... (169 bytes))
1706483447: Received PUBLISH from mqttjs_21b0b174 (d0, q0, r0, m0, 'zigbee2mqtt/device/loft/light/0x5c0272fffe822626', ... (50 bytes))
1706483447: Sending PUBLISH to mqttjs_21b0b174 (d0, q0, r0, m0, 'zigbee2mqtt/device/loft/light/0x5c0272fffe822626', ... (50 bytes))

Where can i find a defined solution that limits this to a single message. Im afraid I haven't found a spelled out solution besides using debounce. Id rather not use debounce as it delays the message being sent to listeners and would hinder performance. What is the best solution here?