Koenkk/zigbee2mqtt #3688: Experiences with different CC2530 boards?

JsBergbau, 2020-06-07:
As experienced by two users here https://github.com/Koenkk/zigbee2mqtt/issues/1568#issuecomment-613008695 the CC2530 board gave the maximum range/best signal compared to other devices. The CC2538 was concerning range about the quality of a CC2531 with external antenna. 

So I'd like to test/use a CC2530 board. 
I've found the "E18-MS1PA1-IPX" here https://www.aliexpress.com/item/4000543857571.html
However this is only a module, without anything else. How would I flash this module? 

This device https://www.aliexpress.com/item/4000060995530.html looks very promising CC2530 wutg CC25910 and 19 dBm trasmit power. With external antenna and USB connection https://www.aliexpress.com/item/4000060995530.html
Anybody experiences with it? Seems to be flashable with CC debugger and supplied cable set?

Here https://www.aliexpress.com/item/4000834166422.html? is the device with PCB Antenna und USB connection, 20 dBm transmit power. How can this be flashed?

Here https://www.aliexpress.com/item/4000833648845.html quite the same module but with the possibility to use an external antenna and 27 dBm transmit power.

Hope there are already some experiences with this boards. 

I've searched, but didn't find a similar thread.

MPM1107, 2020-06-08:
Except for the first module you linked, which would require additional components to attach it to a host PC these all seem fine to use to me. There's nothing special about a module that makes it more or less "flashable". All of them have the required pins exposed, all you need is a CC debugger (or arduino/raspberry pi as an alternative) and the right wires to connect to the pins of the module.

JsBergbau, 2020-06-08:
Thanks for the help. I've ordered each and will report back.

closed 2020-07-26 by stale[bot]

JsBergbau, 2020-08-07:
Finally I've got the boards and could test.
Saidly all the other boards don't work.

They can be flashed with CC-Debugger, that works. But that was it.

https://de.aliexpress.com/item/4000833648845.html Isn't recognised at all. No message in dmesg.

Bother other devices fail with

` Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)`

Can't recommend all of them.

```
npm start

> zigbee2mqtt@1.14.2 start /opt/zigbee2mqtt
> node index.js

zigbee2mqtt:info  2020-08-07 14:27:36: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-08-07.14-27-36' filename: log.txt
zigbee2mqtt:info  2020-08-07 14:27:37: Starting zigbee2mqtt version 1.14.2 (commit #b91b456)
zigbee2mqtt:info  2020-08-07 14:27:37: Starting zigbee-herdsman...
zigbee2mqtt:error 2020-08-07 14:27:57: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-08-07 14:27:57: Failed to start zigbee
zigbee2mqtt:error 2020-08-07 14:27:57: Exiting...
zigbee2mqtt:error 2020-08-07 14:27:57: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:87:31)
    at Generator.throw (<anonymous>)
    at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.14.2 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.14.2 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-08-07T13_27_57_405Z-debug.log
```