Koenkk/zigbee2mqtt #3693: Help with mqtt output

Mopele, 2020-06-08:
### Hey, I would like to use a motion sensor with the zigbee2mqtt framework on a raspberry, but I dont quite get it. My goal is:
- To get a python script that can receive a message if a motion is detected.
-Let me emphazise that I am not looking for a home-server solution or a big home-automation software, I want to be able to integrate the zigbee sensors into my own projects! 

#### For now I understood that throug the zigbee2mqtt stick and software I can get the sensor paired to the stick. But do I need an extra Mqtt-broker to communicate with the stick? How do I get the received messages from the stick to my python script?

I know this sounds like a "just google and you will find it" question and I am not even sure if it belongs in the Issues tab, but I really tried to do my own research but I find it hard to even get started with all the stuff, so excuse my clumsy aproach right here!

MPM1107, 2020-06-08:
Yes, zigbee2mqtt needs to connect to a mqtt broker to publish the received messages to that broker. The most popular software for that by far is mosquitto. After that, you can connect to that same broker with your python code. The library I like to use is paho-mqtt, you can easily get it through pip.

Mopele, 2020-06-08:
> Yes, zigbee2mqtt needs to connect to a mqtt broker to publish the received messages to that broker. The most popular software for that by far is mosquitto. After that, you can connect to that same [...]

Thanks, I will do that when I get my USB Gateway!

closed 2020-07-26 by stale[bot]