Koenkk/zigbee2mqtt #3696: Doesn't have "model" attribute, skipped

madradrakie, 2020-06-09:
# Bug Report
## What happened
I added the RobbShop dimmer SKU: ROB_200-014-0 to the zigbee network and is paired and showed in de zigbee2mqtt page in domoticz. 

It is showed as ZG2835RAC ZigBee knob smart dimmer. 

When i select the dimmer normally a table appears below to add the devices to domoticz. For this dimmer it does not work so i can not add the dimmer as device to domoticz. According to the zigbee2mqtt page this dimmer is supported? Am i doing something wrong? the other devices are all added without any problem.

In the debug i have found: doesn't have "model" attribute, skipped.

Thanks in advance
## What did you expect to happen
Select a device in Domoticz

## How to reproduce it (minimal and precise)
`2020-06-09 06:47:11.708 (Zigbee2MQTT) MQTT message: zigbee2mqtt/bridge/log {'type': 'pairing', 'message': 'interview_started', 'meta': {'friendly_name': 'Keuken Inbouwspots'}}
2020-06-09 06:47:11.777 (Zigbee2MQTT) MQTT message: zigbee2mqtt/bridge/config/devices
2020-06-09 06:47:11.778 (Zigbee2MQTT) MQTT message: zigbee2mqtt/bridge/log {'type': 'devices', 'message': [{'ieeeAddr': '0x00124b001949dc57', 'type': 'Coordinator', 'networkAddress': 0, 'friendly_name': 'Coordinator', 'softwareBuildID': 'zStack12', 'dateCode': '20190608', 'lastSeen': 1591678031728}, {'ieeeAddr': '0x00158d00049ff115', 'type': 'EndDevice', 'networkAddress': 17279, 'model': 'MCCGQ11LM', 'vendor': 'Xiaomi', 'description': 'Aqara door & window contact sensor', 'friendly_name': 'Deursensor Tussendeur', 'manufacturerID': 4151, 'manufacturerName': 'LUMI', 'powerSource': 'Battery', 'modelID': 'lumi.sensor_magnet.aq2', 'hardwareVersion': 2, 'softwareBuildID': '3000-0001', 'dateCode': '20161128', 'lastSeen': 1591676191952}, {'ieeeAddr': '0x00158d00046668e7', 'type': 'EndDevice', 'networkAddress': 48946, 'model': 'RTCGQ11LM', 'vendor': 'Xiaomi', 'description': 'Aqara human body movement and illuminance sensor', 'friendly_name': 'PIR Kastenkamer', 'manufacturerID': 4151, 'manufacturerName': 'LUMI', 'powerSource': 'Battery', 'modelID': 'lumi.sensor_motion.aq2', 'lastSeen': 1591677696998}, {'ieeeAddr': '0x00158d00039dc48f', 'type': 'EndDevice', 'networkAddress': 47540, 'model': 'DJT11LM', 'vendor': 'Xiaomi', 'description': 'Aqara vibration sensor', 'friendly_name': 'Vibration Kattenluik', 'manufacturerID': 4151, 'manufacturerName': 'LUMI', 'modelID': 'lumi.vibration.aq1', 'softwareBuildID': '3000-0001', 'lastSeen': 1591675018747}, {'ieeeAddr': '0x00158d000486551f', 'type': 'EndDevice', 'networkAddress': 13321, 'model': 'RTCGQ11LM', 'vendor': 'Xiaomi', 'description': 'Aqara human body movement and illuminance sensor', 'friendly_name': 'PIR Studeerkamer', 'manufacturerID': 4151, 'manufacturerName': 'LUMI', 'powerSource': 'Battery', 'modelID': 'lumi.sensor_motion.aq2', 'hardwareVersion': 1, 'softwareBuildID': '3000-0001', 'dateCode': '20170627', 'lastSeen': 1591676925276}, {'ieeeAddr': '0x00158d00044d193d', 'type': 'EndDevice', 'networkAddress': 27233, 'model': 'WSDCGQ11LM', 'vendor': 'Xiaomi', 'description': 'Aqara temperature, humidity and pressure sensor', 'friendly_name': 'Temperatuursensor Slaapkamer', 'manufacturerID': 4151, 'manufacturerName': 'LUMI', 'powerSource': 'Battery', 'modelID': 'lumi.weather', 'hardwareVersion': 30, 'softwareBuildID': '3000-0001', 'dateCode': '20191205', 'lastSeen': 1591675947170}, {'ieeeAddr': '0x00158d00047d61f6', 'type': 'EndDevice', 'networkAddress': 26015, 'model': 'MCCGQ11LM', 'vendor': 'Xiaomi', 'description': 'Aqara door & window contact sensor', 'friendly_name': 'Deursensor Voordeur', 'manufacturerID': 4151, 'manufacturerName': 'LUMI', 'powerSource': 'Battery', 'modelID': 'lumi.sensor_magnet.aq2', 'hardwareVersion': 2, 'softwareBuildID': '3000-0001', 'dateCode': '20161128', 'lastSeen': 1591676069521}, {'ieeeAddr': '0xec1bbdfffe93808c', 'type': 'Router', 'networkAddress': 49838, 'model': 'LED1836G9', 'vendor': 'IKEA', 'description': 'TRADFRI LED bulb E26/E27 806 lumen, dimmable, warm white', 'friendly_name': 'Washok', 'manufacturerID': 4476, 'manufacturerName': 'IKEA of Sweden', 'powerSource': 'Mains (single phase)', 'modelID': 'TRADFRI bulb E27 WW 806lm', 'hardwareVersion': 2, 'softwareBuildID': '2.1.022', 'dateCode': '20181203', 'lastSeen': 1591636383306}, {'ieeeAddr': '0xec1bbdfffe90fdeb', 'type': 'Router', 'networkAddress': 39762, 'model': 'LED1836G9', 'vendor': 'IKEA', 'description': 'TRADFRI LED bulb E26/E27 806 lumen, dimmable, warm white', 'friendly_name': 'Trapkast', 'manufacturerID': 4476, 'manufacturerName': 'IKEA of Sweden', 'powerSource': 'Mains (single phase)', 'modelID': 'TRADFRI bulb E27 WW 806lm', 'hardwareVersion': 2, 'softwareBuildID': '2.1.022', 'dateCode': '20181203', 'lastSeen': 1591636980778}, {'ieeeAddr': '0xec1bbdfffe7f3b28', 'networkAddress': 16704, 'vendor': '-', 'description': '-', 'friendly_name': 'Keuken Inbouwspots', 'lastSeen': 1591678031682}]}
2020-06-09 06:47:11.779 (Zigbee2MQTT) Received available devices list from bridge
2020-06-09 06:47:11.779 (Zigbee2MQTT) Coordinator address is 0x00124b001949dc57
2020-06-09 06:47:11.779 (Zigbee2MQTT) Device MCCGQ11LM 0x00158d00049ff115 (Deursensor Tussendeur)
2020-06-09 06:47:11.779 (Zigbee2MQTT) Device RTCGQ11LM 0x00158d00046668e7 (PIR Kastenkamer)
2020-06-09 06:47:11.780 (Zigbee2MQTT) Device DJT11LM 0x00158d00039dc48f (Vibration Kattenluik)
2020-06-09 06:47:11.780 (Zigbee2MQTT) Device RTCGQ11LM 0x00158d000486551f (PIR Studeerkamer)
2020-06-09 06:47:11.780 (Zigbee2MQTT) Device WSDCGQ11LM 0x00158d00044d193d (Temperatuursensor Slaapkamer)
2020-06-09 06:47:11.781 (Zigbee2MQTT) Device MCCGQ11LM 0x00158d00047d61f6 (Deursensor Voordeur)
2020-06-09 06:47:11.781 (Zigbee2MQTT) Device LED1836G9 0xec1bbdfffe93808c (Washok)
2020-06-09 06:47:11.782 (Zigbee2MQTT) Device LED1836G9 0xec1bbdfffe90fdeb (Trapkast)
2020-06-09 06:47:11.782 (Zigbee2MQTT) Device 0xec1bbdfffe7f3b28 (Keuken Inbouwspots) doesn't have "model" attribute, skipped
2020-06-09 06:47:11.833 (Zigbee2MQTT) MQTT message: zigbee2mqtt/bridge/log {'type': 'device_announced', 'message': 'announce', 'meta': {'friendly_name': 'Keuken Inbouwspots'}}`

## Debug Info
Zigbee2mqtt version:
v.1.13.0 (zStack12 20190608)

Adapter hardware: CC2531

Adapter firmware version: latest

Koenkk, 2020-06-09:
@stas-demydiuk is this related to the domoticz plugin?

stas-demydiuk, 2020-06-09:
Yes, plugin uses `model` attribute to create appropriate logical devices in Domoticz and shows `... doesn't have "model" attribute, skipped` message when this attribute is missed. 

I noticed that it happens time to time that model attribute doesn't available immediately after device pairing and appears in about an hour or so

madradrakie, 2020-06-09:
is is working now after some changes in devices.js file. The model which was recognized was ZG2835RAC. In de devices.js file this model number was assigned to 

`{
      zigbeeModel: ['Micro Smart Dimmer'],
      model: 'ZG2835RAC',
      vendor: 'Sunricher',
      description: 'ZigBee knob smart dimmer',
      extend: generic.light_onoff_brightness,
 },`

i know changed this one to:

`    {
        zigbeeModel: ['Micro Smart Dimmer'],
        model: 'ZG2835RAC',
        vendor: 'ROBB',
        description: 'ZigBee AC phase-cut rotary dimmer',
        extend: generic.light_onoff_brightness,
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
            await configureReporting.onOff(endpoint);
        },
    },
`
I also added the following in __init__.py file:

`'ZG2835RAC': DimmableBulbAdapter,   # ROBB ZigBee AC phase-cut rotary dimmer
`

Now the dimmer is also recognized in Domoticz so i can add the dimmer as a device.

closed 2020-06-09 by madradrakie