Koenkk/zigbee2mqtt #3697: Other device model after paired

tabletka-ao, 2020-06-09:
# Bug Report
## What happened
Recently I bought wall switches X711A, X712A but after paired  device model is TS0601.

{"id":2,"type":"Router","ieeeAddr":"0xec1bbdfffe76c09f","nwkAddr":54292,"manufId":4098,"manufName":"_TZE200_dhdstcqc","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_dhdstcqc","powerSource":1,"zclVersion":3,"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[]}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1591649605846}

debug: Received Zigbee message from '0xec1bbdfffe76c09f', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0

info : MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"devices","message":[{"ieeeAddr":"0x00124b001cd47e50","type":"Coordinator","networkAddress":0,"friendly_name":"Coordinator","softwareBuildID":"zStack12","dateCode":"20180507","lastSeen":1591677360069},{"ieeeAddr":"0xec1bbdfffe76c09f","type":"Router","networkAddress":54292,"model":"TS0601","vendor":"TuYa","description":"Curtain motor","friendly_name":"0xec1bbdfffe76c09f","manufacturerID":4098,"manufacturerName":"_TZE200_dhdstcqc","powerSource":"Mains (single phase)","modelID":"TS0601","hardwareVersion":1,"dateCode":"","lastSeen":1591677352233}]}'

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2mqtt version: v1.7.0+
Adapter hardware: CC2531 (Model M0802010007)
Adapter firmware version: CC2531ZNP-PROD

closed 2020-06-09 by tabletka-ao

tabletka-ao, 2020-06-17:
@Machaeon 
At the moment only solution #2778

galzil, 2020-08-09:
I just noticed that myself as the switches still work, but the identification is wrong.
The only problem with that is that the devices are treated as EndDevice and not as Router.
This is the database.db dump for these devices
`{"id":3,"type":"EndDevice","ieeeAddr":"0x680ae2fffe71b08f","nwkAddr":30197,"manufId":4098,"manufName":"_TYZB01_mtlhqn48","powerSource":"Battery","modelId":"TS0012","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,4,5,6],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65506":19,"modelId":"TS0012","manufacturerName":"_TYZB01_mtlhqn48","powerSource":3,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0014d98784","endpointID":1}]},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[4,5,6],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0014d98784","endpointID":1}]}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1596966023782}
{"id":4,"type":"EndDevice","ieeeAddr":"0x680ae2fffe71ae65","nwkAddr":43898,"manufId":4098,"manufName":"_TYZB01_xfpdrwvc","powerSource":"Battery","modelId":"TS0011","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,4,5,6],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"TS0011","manufacturerName":"_TYZB01_xfpdrwvc","powerSource":3,"zclVersion":3,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[]}},"appVersion":255,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1596969830458}`
They are Mains powered [X711A](https://www.zigbee2mqtt.io/devices/X711A.html) and should be Routers as well, as you can see on the first post in this thread.