Koenkk/zigbee2mqtt #3698: Hive motion sensors MOT003 are now supported but no lux or temperature is being reported

dnedeltchev, 2020-06-09:
# Bug Report
## What happened
I have also reported this under Daniel's hassio-zigbee2mqtt thread. I am not sure 
MOT003 Hive motion sensors are supported according to the zigbee2mqtt supported devices page and indeed they do pair successfully and I can see all sensors but... the only sensor data that I can get is motion, battery, tamper and link quality. The temperature and lux sensors return no data at all.

I sometimes see this in the logs but I have not figured out how to reproduce this error. I have 3 hive sensors and they all behave the same way so it's unlikely it is a hardware issue. The hive sensors were connected to a hive hub and working as expected before they were connected to the zigbee stick.

zigbee2mqtt:info 2020-06-08 18:15:31: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00124b001312b72c"}}'
zigbee2mqtt:error 2020-06-08 18:15:41: Failed to configure '0x00124b001312b72c', attempt 3 (Error: Bind 0x00124b001312b72c/6 msTemperatureMeasurement from '0x00124b0018e322e2/1' failed (Error: AREQ - ZDO - bindRsp after 10000ms)
at Endpoint. (/zigbee2mqtt-1.13.1/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:244:23)
at Generator.throw ()
at rejected (/zigbee2mqtt-1.13.1/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))

## What did you expect to happen
As per the supported devices note, I would have expected to get lux and temperature data.
## How to reproduce it (minimal and precise)
Pair the hive sensor and then look at the sensor data.
## Debug Info
Zigbee2mqtt version: latest 1.13.1
Adapter hardware: CC2531
Adapter firmware version:
zigbee2mqtt:info  2020-06-09 12:24:00: MQTT publish: topic 'zigbee2mqtt/bridge/config/devices', payload '[{"ieeeAddr":"0x00124b0018e322e2","type":"Coordinator","networkAddress":0,"friendly_name":"Coordinator","softwareBuildID":"zStack12","dateCode":"20190608","lastSeen":1591701840038},

Koenkk, 2020-06-09:
The reason it's not reported is probably the configure error (`Failed to configure`). Try to configure it manually via https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure , but right before executing this command make sure to wake the device up by pressing a button on it.

Please also close the issue on danielswelch repo, as it's no related to this.

dnedeltchev, 2020-06-10:
Thank you Koen,
This is quite a bit over my head :) I have no idea where to start with that :)
Is it likely that this will be fixed in the future or this is not fixable due to hardware/other limitations, etc.?

hunterdrayman, 2020-11-17:
I have successfully paired the MOT003 with a CC2531 although by comparison with my Sonoff and Xiaomi sensors it is reading about 2 degrees high and not reacting quickly to temperature changes. The motion sensor works well.

`
Zigbee2MQTT:info  2020-11-17 08:22:04: Successfully configured 'temperature-occupancy/unit1'
Zigbee2MQTT:info  2020-11-17 08:22:05: MQTT publish: topic 'zigbee2mqtt/temperature-occupancy/unit1', payload '{"battery":88,"battery_low":false,"linkquality":36,"occupancy":true,"tamper":false,"temperature":27}'
`