Koenkk/zigbee2mqtt #3702: Danalock V3 not reporting battery status

elbueno222, 2020-06-09:
# Bug Report
## What happened

According to https://www.zigbee2mqtt.io/devices/V3-BTZB.html, the lock Danalock V3 should support lock/unlock and battery status, but after pairing the device in Home Assistant using zigbee2mqtt just two entities are created (lock and linkquality). Battery status is missing.

## What did you expect to happen

I would expect a battery entity being created and the lock reporting the battery level.

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2mqtt version: 1.13.1.1
Adapter hardware: CC2531
Adapter firmware version: Source_Routing_20190619

Koenkk, 2020-06-09:
Fixed in latest dev branch.

closed 2020-06-09 by Koenkk

elbueno222, 2020-06-09:
Wow, that was quick!! I can confirm the fix is working, thank you very much