Koenkk/zigbee2mqtt #3703: Set 'colortemp_move' on IKEA lightbulbs

knieschm, 2020-06-09:
# Bug Report
## What happened
If setting 'colortemp_move' on Ikea LED1537R6 bulb or Ikea L1527 panel color temperature is not changing. There are no error messages for zigbee2mqtt or zigbee-herdsman.
## What did you expect to happen
The color temperature of lightbulb is changing until 'stop'-message is received. I expected the same behavior as for 'brightness_move'.
## How to reproduce it (minimal and precise)
Set 'colortemp_move' to e.g. '-20'.
## Debug Info
Zigbee2mqtt version: 1.13.1
Adapter hardware: CC1352P-2
Adapter firmware version: 20200417

Koenkk, 2020-06-09:
It seems that the structure is a bit different, try with e.g. `{"colortemp_move": "up", "rate": 50}`. Not sure why this was done, need to investigate that further.

knieschm, 2020-06-09:
Yes, if using this structure the lightbulb is reacting on the commands. However it stops at the hardcoded limits.
https://github.com/Koenkk/zigbee-herdsman-converters/blob/6246d62fd20fec41a6fc5d46adacc67563e947be/converters/toZigbee.js#L313

knieschm, 2020-06-11:
I tested you changes and did some investigations but could not find a solution.
The lightbulb is always lowering the color temperature, no matter what rate is send. This is influenced by setting the limits to zero. 
If setting the limits to `const payload = {minimum: 153, maximum: 500};` I can control the color temperature correctly. 
However the movement stop at a minimum of 250 and a maximum of 454. This may be a limitation by IKEA since I can not see the color temperature changing if setting the color temperature to values outside of this range.
Furthermore it is not possible to send the message `topic: "zigbee2mqtt/light/set/color_temp_move"
payload: 20`. This will result in `typeof value === 'number')` being false (value has the type string) which leads to using the old implementation.

Koenkk, 2020-06-11:
Can you apply the changes from https://github.com/Koenkk/zigbee-herdsman-converters/pull/1306/files and see if it fixes the 2 issues.

250 and 454 are probably the min/max color temp values so seems logic that it stops there.