Koenkk/zigbee2mqtt #3704: Network lags after enabling report and check_availabilty

Hacker-CB, 2020-06-09:
# Bug Report
## What happened
I have network with 96 devices (around 60+ IKEA bulbs from total). 
After enabling `report: true `and `check_availabilty: 60` network had a lot of delays and fails.
After some time I have disabled report and check_availabilty features, but I still see a lot of messages in logs like:
`Received Zigbee message from 'kitchen_track_lamp_north_03', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":212}' from endpoint 1 with groupID 0`

I tried to reboot coordinator, all bulbs but nothing was changed.
Network become much less stable and lot of devices sometimes does not work.

So, seems that it impossible to disable report after enabling it.

## What did you expect to happen

Network should be still stable

## How to reproduce it (minimal and precise)

Enable report feature and disable it later.

## Debug Info
Zigbee2mqtt version: 1.13.1
Adapter hardware: CC2538+CC2592
Adapter firmware version: https://github.com/jethome-ru/zigbee-firmware/blob/master/coordinator/cc2538_cc2592/MODKAMRU_V3_USB_80.hex (re-compiled TI firmware with MODCAM pactches and 80 direct children, instead of 100)

Koenkk, 2020-06-09:
Disabling the report is indeed not supported yet. But for IKEA you can fix it by repowering the bulbs (as IKEA bulbs forget their reporting setup after losing power).

Hacker-CB, 2020-06-09:
So, for your mind is reporting is a a high load for such large network?

Koenkk, 2020-06-09:
I don't have experience with such large networks and reporting, but it indeed seems to be the case.

way2many, 2020-06-17:
Just to add a bit of info, I have around 45 devices and check_availability add a huge stress if enabled with reporting. Currently, I only have reporting with Zstack 1.2 on a CC2531 and network is stable.

closed 2020-07-26 by stale[bot]