Koenkk/zigbee2mqtt #3709: Failed to enable unit: "multi-user.targe" is not a valid unit name.

Dylantje, 2020-06-10:
Last beta Domoticz
Odroid N2 with latest armbian

Everthing looks running fine.
Until i will set:
` sudo systemctl enable zigbee2mqtt.service
`
I get this strange error?
`Failed to enable unit: "multi-user.targe" is not a valid unit name.
`
Did use google.
No thing found for this strange thing
Pi user does have sudo options
Only error with this command

z2m is running..

Koenkk, 2020-06-10:
Probably you forgot to copy the last character a **t** (multi-user.target). See https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html#5-optional-running-as-a-daemon-with-systemctl

closed 2020-06-10 by Koenkk

hctru68, 2021-01-14:
Thank you so much,
That link is useful for me. I fixed this bug by that link.