Koenkk/zigbee2mqtt #3715: Unsupported device - TERNCY Wall Switch

nox1982, 2020-06-11:
Hello,

Just to check if there is any plan to add this wall switch from Terncy?
https://www.xiaoyan.io/en-us/wallswitch-withneutral/

I tried to add but log shown it is unsupported device at the moment. I do not have the technical expertise to add.

Unsupported device?
- https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

-->

# Bug Report
## What happened
Device paired
Logs indicate device is not supported - instructions provided

## What did you expect to happen
Device to be supported

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2mqtt version: latest
Adapter hardware: CC2531
Adapter firmware version:

nox1982, 2020-07-19:
So I edited the devices.js but I really don't know how to proceed with converter part.

Can anyone rendered the support on below how to proceed further?

zigbee2mqtt:debug 2020-07-18 01:59:37: Received Zigbee message from '0x000d6f0014463374', type 'attributeReport', cluster 'genOta', data '{"currentFileVersion":95}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-07-18 01:59:37: Received Zigbee message from '0x000d6f0014463374', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-07-18 01:59:37: MQTT publish: topic 'zigbee2mqtt/0x000d6f0014463374', payload '{"state":"OFF","linkquality":36,"state_left":"OFF","state_right":"ON"}'
zigbee2mqtt:debug 2020-07-18 01:59:37: Received Zigbee message from '0x000d6f0014463374', type 'attributeReport', cluster 'manuSpecificClusterAduroSmart', data '{"25":1}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-07-18 01:59:37: No converter available for 'TERNCY-WS01-S2' with cluster 'manuSpecificClusterAduroSmart' and type 'attributeReport' and data '{"25":1}'
zigbee2mqtt:debug 2020-07-18 01:59:38: Received Zigbee message from '0x000d6f0014463374', type 'attributeReport', cluster 'manuSpecificClusterAduroSmart', data '{"32":0}' from endpoint 1 with groupID 0

closed 2020-07-22 by nox1982

mygithandle, 2022-02-05:
I've just moved into a newly renovated house chock-full of Terncy wall switches, lights and other devices. I really like the look and feel of the products, and would like to see it gain more following and support, esp in the HA community. 

BUT, after first few days with the vendor's hub and app, and an unsatisfying first trial of an HA integration found on GH,  I am ready to take the leap into getting these switches into zigbee2mqtt.

The OP here was trying to integrate a TERNCY-WS01-S2 wall switch, whereas mine are  WS01-D1, D2, D3, and D4 (different multi-gang count). I presume mine are more recently launched versions; don't know whether that means additional functionality. I notice that a few other Terncy devices now have converters: https://zigbee.blakadder.com/vendors.html

It will probably be 2-3 weeks before I get a zigbee USB adaptor and other stuff ready. And haven't yet decided whether to put zigbee2mqtt on my existing HA rPi, or move the whole mess to a proper server, and run z2m in a docker. 

In any case, hope this comment at this time might attract other folks with Terncy wall switch hardware to check in here, and maybe pitch in to help when I start trying to sort out a converter.

Maozhenlei, 2023-02-04:
@mygithandle 
Hello, have you successfully add the switch to HA? I also have a room full of Xiaoyan equipment, and I am troubled by it

mygithandle, 2023-02-20:
got frustrated and too busy with other things, and set this project aside for a year or so.  Just starting over. I am thinking of trying to develop my own converter files for my XiaoYan devices. I think they use the same chipset as many Xiaomi products, so will try using those products converter files as reference. Expect this will be a slow project, weeks before success.

will report any progress here

Helloookimi, 2023-04-26:
have you successfully add the switch to Z2M? I also have this switch, I tried add to Z2M but failed.