Koenkk/zigbee2mqtt #3723: Reset track length for ZNCLDJ12LM

arjanvanhoogdalem, 2020-06-12:
Please include the option to reset track length for the curtain motor ZNCLDJ12LM.
Currently my motor thinks it's on a very small track, and doesn't fully close anymore.

MPM1107, 2020-06-12:
Can you try the process in the video linked in this thread? [https://community.smartthings.com/t/xiaomi-aqara-curtain-motor-integration/125961/22](https://community.smartthings.com/t/xiaomi-aqara-curtain-motor-integration/125961/22) I'm not sure if it's possible to trigger this via zigbee, if it is somebody has to sniff the traffic that gets sent from the original hub first, the feature can't be added without that info.

arjanvanhoogdalem, 2020-06-12:
Thanks. I found the same video, but it's for the older model of the motor. I have the B1 motor, and there it doesn't work. 
It is possible to reset it via the app I think, but I would need a hub to do so.

arjanvanhoogdalem, 2020-06-13:
I found another way to reset the track length. Pressing the button 10+ times. So this ticket is not needed anymore.

closed 2020-06-13 by arjanvanhoogdalem